meshtastic/firmware #5053: Reboot loop on Heltec v3, powered from USB only

bradisbell, 2024-10-14:
I have a couple Heltec v3 modules that like to get stuck in a boot loop.  Here's a log from one of them, with no battery connected.  Only USB.  The USB power is solid... same problem occurs with any type of USB power source, from a proper host like a laptop, to passive power bricks and batteries.

When the board does run, it detects the voltage around 3.7V, until some seconds later and then it sees it as 4.9V.  This is after I set the multiplier to `6`.

I've tried a couple different firmware versions... no luck.

It seems to want to reboot on this step:

```
INFO  | ??:??:?? 4 [PowerFSM] Initialise the NimBLE bluetooth module
```

Is there anything that can be done to convince the ESP32 to not reboot?  Can we configure it to ignore whatever voltage it thinks it has and keep running?  In particular, can this sort of setting be made *before* we start powering up the 2.4 GHz radio?

Here's the full log...

```
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xf (BROWNOUT_RST),boot:0x28 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x4bc
load:0x403c9700,len:0xbd8
load:0x403cc700,len:0x2a0c
entry 0x403c98d0
E (343) esp_core_dump_flash: No core dump partition found!
E (343) esp_core_dump_flash: No core dump partition found!
▒▒@INFO  | ??:??:?? 0

//\ E S H T /\ S T / C

INFO  | ??:??:?? 0 Booted, wake cause 0 (boot count 1), reset_reason=brownout
DEBUG | ??:??:?? 0 Filesystem files (24576/1048576 Bytes):
DEBUG | ??:??:?? 0  /prefs/channels.proto (57 Bytes)
DEBUG | ??:??:?? 0  /prefs/config.proto (162 Bytes)
DEBUG | ??:??:?? 0  /prefs/db.proto (371 Bytes)
DEBUG | ??:??:?? 0  /static/.gitkeep (0 Bytes)
[   460][I][esp32-hal-i2c.c:75] i2cInit(): Initialising I2C Master: sda=41 scl=42 freq=100000
[   473][I][esp32-hal-i2c.c:75] i2cInit(): Initialising I2C Master: sda=17 scl=18 freq=100000
DEBUG | ??:??:?? 0 Power::lipoInit lipo sensor is not ready yet
DEBUG | ??:??:?? 0 Using analog input 1 for battery level
INFO  | ??:??:?? 0 ADCmod: ADC Characterization based on Two Point values and fitting curve coefficients stored in eFuse
INFO  | ??:??:?? 0 Scanning for i2c devices...
[   508][W][Wire.cpp:301] begin(): Bus already started in Master Mode.
DEBUG | ??:??:?? 0 Scanning for I2C devices on port 2
[   538][W][Wire.cpp:301] begin(): Bus already started in Master Mode.
DEBUG | ??:??:?? 0 Scanning for I2C devices on port 1
DEBUG | ??:??:?? 0 I2C device found at address 0x3c
INFO  | ??:??:?? 0 ssd1306 display found
INFO  | ??:??:?? 0 ssd1306 display found
DEBUG | ??:??:?? 0 0x3 subtype probed in 2 tries
INFO  | ??:??:?? 0 1 I2C devices found
DEBUG | ??:??:?? 0 acc_info = 0
INFO  | ??:??:?? 0 S:B:43,2.5.5.e182ae7
INFO  | ??:??:?? 0 Build timestamp: 1728125067
DEBUG | ??:??:?? 0 Total heap: 291376
DEBUG | ??:??:?? 0 Free heap: 254448
DEBUG | ??:??:?? 0 Total PSRAM: 0
DEBUG | ??:??:?? 0 Free PSRAM: 0
DEBUG | ??:??:?? 0 NVS: UsedEntries 78, FreeEntries 552, AllEntries 630, NameSpaces 4
DEBUG | ??:??:?? 0 Setup Preferences in Flash Storage
DEBUG | ??:??:?? 0 Number of Device Reboots: 25
INFO  | ??:??:?? 0 OTA firmware version 0.3.0.87f0f2f
INFO  | ??:??:?? 0 Initializing NodeDB
INFO  | ??:??:?? 0 Loading /prefs/db.proto
INFO  | ??:??:?? 0 Loaded /prefs/db.proto successfully
INFO  | ??:??:?? 0 Loaded saved devicestate version 23, with nodecount: 1
INFO  | ??:??:?? 0 Loading /prefs/config.proto
INFO  | ??:??:?? 0 Loaded /prefs/config.proto successfully
[... 176 lines cut ...]
INFO  | ??:??:?? 1 Bandwidth set to 250.000000
INFO  | ??:??:?? 1 Power output set to 22
DEBUG | ??:??:?? 1 Current limit set to 140.000000
DEBUG | ??:??:?? 1 Current limit set result 0
DEBUG | ??:??:?? 1 Set DIO2 as RF switch, result: 0
DEBUG | ??:??:?? 1 SX126X_RXEN not defined, defaulting to RADIOLIB_NC
DEBUG | ??:??:?? 1 SX126X_TXEN not defined, defaulting to RADIOLIB_NC
DEBUG | ??:??:?? 1 Using MCU pin -1 as RXEN and pin -1 as TXEN to control RF switching
INFO  | ??:??:?? 1 Set RX gain to boosted mode; result: 0
INFO  | ??:??:?? 1 SX1262 Radio init succeeded, using SX1262 radio
INFO  | ??:??:?? 1 Not using WIFI
DEBUG | ??:??:?? 1 LoRA bitrate = 118.500000 bytes / sec
INFO  | ??:??:?? 1 PowerFSM init, USB power=1
DEBUG | ??:??:?? 1 Enter state: BOOT
[  1212][D][esp32-hal-cpu.c:244] setCpuFrequencyMhz(): PLL: 480 / 6 = 80 Mhz, APB: 80000000 Hz
DEBUG | ??:??:?? 1 [Power] Battery: usbPower=0, isCharging=0, batMv=3100, batPct=0
DEBUG | ??:??:?? 1 [Screen] Screen: Started...
DEBUG | ??:??:?? 1 [Screen] haveGlyphs=1
INFO  | ??:??:?? 1 [RangeTestModule] Range Test Module - Disabled
DEBUG | ??:??:?? 2 [Screen] haveGlyphs=1
WARN  | ??:??:?? 2 [RadioIf] recv - lora rx disabled because RegionCode_Unset
DEBUG | ??:??:?? 2 [RadioIf] Packet received (noise?) : 428ms
DEBUG | ??:??:?? 3 [Screen] haveGlyphs=1
...
```

caveman99, 2024-10-14:
try reflashing with wipe.

bradisbell, 2024-10-14:
> try reflashing with wipe.

I've tried that several times.

thebentern, 2024-10-14:
Browning out on USB is almost definitely indicating a hardware setup issue. Try a different USB power supply. Seems unlikely that two separate devices would fail if it was the boards' DC converters. More likely the power supply.

RightHandMan1, 2024-10-14:
Change the ADC value to 5.11 and check. voltage of 4.9 is WAY high, show show around 4.2 when fully charged on a V3. 4.9 is probly rebooting from false overvoltage.

thebentern, 2024-10-14:
ADC override shouldn't be factoring into that brownout message though. That's generated by the ESP32 itself.

garthvh, 2024-10-14:
> Change the ADC value to 5.11 and check. voltage of 4.9 is WAY high, show show around 4.2 when fully charged on a V3. 4.9 is probly rebooting from false overvoltage.

There is not a battery, so the adc pin does nothing.

bradisbell, 2024-10-14:
> Browning out on USB is almost definitely indicating a hardware setup issue. Try a different USB power supply. Seems unlikely that two separate devices would fail if it was the boards' DC converters. More likely the power supply.

I've tried all sorts of different power supplies, no luck.  Probably a bad batch of boards then?  They're fresh out of the box with the only thing connected being the antenna.

closed 2024-10-14 by thebentern

HeedfulCrayon, 2024-10-23:
I have also been experiencing this.  The only thing that I have found to consistently fix the issue is actually touching my finger to the USB shielding and the wifi/bt antenna (the little spring looking antenna).  I am thinking there is a bad ground somewhere with the wifi or bt, and by bridging this gap, it allows it to boot

bradisbell, 2024-10-23:
> I have also been experiencing this. The only thing that I have found to consistently fix the issue is actually touching my finger to the USB shielding and the wifi/bt antenna (the little spring looking antenna)

Interesting!  That's worked for me by accident... thought it was a fluke and didn't think anything of it, but sure enough that's working on the board on my desk right now.

Ccz119, 2025-02-23:
same here..why is this happening??

BlitzdUK, 2025-06-11:
> I have also been experiencing this. The only thing that I have found to consistently fix the issue is actually touching my finger to the USB shielding and the wifi/bt antenna (the little spring [...]

I was having the exact same issue but this worked for me!
I wonder what the logical reason behind this is?