meshtastic/firmware #5056: [Bug]: Setting position.gps_mode to DISABLED Does Not Skip GPS Detection

D4rk4, 2024-10-14:
### Category

Other

### Hardware

Not Applicable

### Firmware Version

2.5.6.d55c08d

### Description

When executing the command meshtastic --set position.gps_mode DISABLED, the expectation is that the device should skip any GPS detection and related processes. However, the device continues to probe for GPS modules, as evidenced by the debug logs.

Expected Behavior
- The device should not attempt to detect or communicate with any GPS modules.
- No GPS-related debug logs should appear.
- The device should skip any GPS initialization processes.

Actual Behavior
- The device continues to probe for GPS modules.
- GPS detection attempts are logged, trying various GPS commands.
- Ultimately, it logs a warning about failing to find a GNSS module.

### Relevant log output

```Shell
DEBUG | ??:??:?? 20 [GPS] Probing for GPS at 38400 
DEBUG | ??:??:?? 21 [GPS] Trying $PDTINFO (UC6580)...
DEBUG | ??:??:?? 21 [GPS] Trying $PDTINFO (UM600)...
DEBUG | ??:??:?? 22 [GPS] Trying $PCAS06,1*1A (ATGM336H)...
DEBUG | ??:??:?? 22 [GPS] Trying $PCAS06,1*1A (ATGM332D)...
DEBUG | ??:??:?? 23 [GPS] Trying $PAIR021*39 (AG3335)...
DEBUG | ??:??:?? 23 [GPS] Trying $PAIR021*39 (AG3352)...
DEBUG | ??:??:?? 24 [GPS] Trying $PQTMVERNO*58 (LC86)...
DEBUG | ??:??:?? 24 [GPS] Trying $PCAS06,0*1B (L76K)...
DEBUG | ??:??:?? 25 [GPS] Trying $PMTK605*31 (L76B)...
WARN  | ??:??:?? 26 [GPS] Failed to find GNSS Module (baudrate 38400)
```

thebentern, 2024-10-14:
Use mode `NOT_PRESENT`. `DISABLED` mode does not turn off detection, it just prevents the GPS from populating position packets
https://meshtastic.org/docs/configuration/radio/position/#gps-mode

closed 2024-10-14 by thebentern

D4rk4, 2024-10-14:
> Use mode `NOT_PRESENT`. `DISABLED` mode does not turn off detection, it just prevents the GPS from populating position packets https://meshtastic.org/docs/configuration/radio/position/#gps-mode

Maybe we can add skipping GPS detection because it spends some CPU cycles and eats power?

fifieldt, 2024-10-17:
@D4rk4 , DISABLED involves power management, and we can't do that properly unless we know what GPS it is.