meshtastic/firmware #5066: [Bug]: changing Device Config -> Role to ROUTER causes node to freeze after reboot + screen stays off

mtsdawson, 2024-10-14:
### Contact Details

skelzer

### What happened?

Tested this on 3 Heltecs.  After a full erase and install of 2.5.6, then immediately changing role to Router, the node tries to reboot and fails.  Reset doesn't fix the issue.  Only way to recover the node is to force another full erase and install after replugging USB and perfectly timing a reset by clicking the Erase Flash and Install button on the web flasher (it takes a few retries for timing to work and succeed with flash).

### App Version

2.5.1

### Phone

OnePlus 8T / Android 14

### Device

Heltex v3 with no GPS or sensor modules.

### Firmware

2.5.6 beta

### Relevant log output

```shell
INFO  | ??:??:?? 436 getFromRadio=STATE_SEND_FILEMANIFEST
INFO  | ??:??:?? 436 getFromRadio=STATE_SEND_COMPLETE_ID
INFO  | ??:??:?? 436 getFromRadio=STATE_SEND_PACKETS
INFO  | ??:??:?? 436 To Radio onwrite
DEBUG | ??:??:?? 436 New ToRadio packet
DEBUG | ??:??:?? 436 PACKET FROM PHONE (id=0xae2fa03c fr=0x00 to=0xac, WantAck=1, HopLim=3 Ch=0x0 Portnum=6 priority=70)
DEBUG | ??:??:?? 436 Enqueued local (id=0xae2fa03c fr=0x00 to=0xac, WantAck=1, HopLim=3 Ch=0x0 Portnum=6 priority=70)
INFO  | ??:??:?? 436 Telling client we have new packets 6
INFO  | ??:??:?? 436 BLE notify fromNum
DEBUG | ??:??:?? 436 [Router] Add packet record (id=0xae2fa03c fr=0x00 to=0xac, WantAck=1, HopLim=3 Ch=0x0 Portnum=6 priority=70)
DEBUG | ??:??:?? 436 [Router] handleReceived(REMOTE) (id=0xae2fa03c fr=0x00 to=0xac, WantAck=1, HopLim=3 Ch=0x0 Portnum=6 priority=70)
DEBUG | ??:??:?? 436 [Router] Module 'Admin' wantsPacket=1
INFO  | ??:??:?? 436 [Router] Received Admin from=0x0, id=0xae2fa03c, portnum=6, payloadlen=6
INFO  | ??:??:?? 436 [Router] Handling admin payload 43
INFO  | ??:??:?? 436 [Router] Client is receiving a set_time_only command.
INFO  | ??:??:?? 436 [Router] getFromRadio=STATE_SEND_PACKETS
DEBUG | ??:??:?? 436 [Router] Upgrading time to quality NTP
DEBUG | 17:27:22 436 [Router] Read RTC time as 1728926842
DEBUG | 17:27:22 436 [Router] Module 'Admin' considered
DEBUG | 17:27:22 436 [Router] Module 'routing' wantsPacket=1
INFO  | 17:27:22 436 [Router] Received routing from=0x0, id=0xae2fa03c, portnum=6, payloadlen=6
DEBUG | 17:27:22 436 [Router] Routing sniffing (id=0xae2fa03c fr=0x00 to=0xac, WantAck=1, HopLim=3 Ch=0x0 Portnum=6 priority=70)
DEBUG | 17:27:22 436 [Router] Partially randomized packet id 2104132125
DEBUG | 17:27:22 436 [Router] Enqueued local (id=0x7d6a821d fr=0xac to=0xac, WantAck=0, HopLim=3 Ch=0x0 Portnum=5 requestId=ae2fa03c rxtime=1728926842 priority=120)
INFO  | 17:27:22 436 [Router] getFromRadio=STATE_SEND_PACKETS
DEBUG | 17:27:22 437 [Router] Module 'routing' considered
DEBUG | 17:27:22 437 [Router] Rx someone rebroadcasting for us (id=0x7d6a821d fr=0xac to=0xac, WantAck=0, HopLim=3 Ch=0x0 Portnum=5 requestId=ae2fa03c rxtime=1728926842 priority=120)
DEBUG | 17:27:22 437 [Router] didn't find pending packet
DEBUG | 17:27:22 437 [Router] Add packet record (id=0x7d6a821d fr=0xac to=0xac, WantAck=0, HopLim=3 Ch=0x0 Portnum=5 requestId=ae2fa03c rxtime=1728926842 priority=120)
DEBUG | 17:27:22 437 [Router] handleReceived(REMOTE) (id=0x7d6a821d fr=0xac to=0xac, WantAck=0, HopLim=3 Ch=0x0 Portnum=5 requestId=ae2fa03c rxtime=1728926842 priority=120)
DEBUG | 17:27:22 437 [Router] Module 'routing' wantsPacket=1
INFO  | 17:27:22 437 [Router] Received routing from=0xda5475ac, id=0x7d6a821d, portnum=5, payloadlen=2
INFO  | 17:27:22 437 [Router] getFromRadio=STATE_SEND_PACKETS
DEBUG | 17:27:22 437 [Router] Routing sniffing (id=0x7d6a821d fr=0xac to=0xac, WantAck=0, HopLim=3 Ch=0x0 Portnum=5 requestId=ae2fa03c rxtime=1728926842 priority=120)
DEBUG | 17:27:22 437 [Router] Received a ACK for 0xae2fa03c, stopping retransmissions
DEBUG | 17:27:22 437 [Router] Delivering rx packet (id=0x7d6a821d fr=0xac to=0xac, WantAck=0, HopLim=3 Ch=0x0 Portnum=5 requestId=ae2fa03c rxtime=1728926842 priority=120)
DEBUG | 17:27:22 437 [Router] Update DB node 0xda5475ac, rx_time=1728926842
DEBUG | 17:27:22 437 [Router] Forwarding to phone (id=0x7d6a821d fr=0xac to=0xac, WantAck=0, HopLim=3 Ch=0x0 Portnum=5 requestId=ae2fa03c rxtime=1728926842 priority=120)
DEBUG | 17:27:22 437 [Router] Module 'routing' considered
INFO  | 17:27:22 437 Telling client we have new packets 7
INFO  | 17:27:22 437 BLE notify fromNum
INFO  | 17:27:22 437 getFromRadio=STATE_SEND_PACKETS
DEBUG | 17:27:22 437 phone downloaded packet (id=0x6e64261b fr=0xac to=0xff, WantAck=0, HopLim=3 Ch=0x0 Portnum=67 priority=10)
[... 91 lines cut ...]
DEBUG | 17:27:33 448 [Router] Add packet record (id=0x90c8aa1f fr=0xac to=0xac, WantAck=0, HopLim=3 Ch=0x0 Portnum=5 requestId=ae2fa03e rxtime=1728926853 priority=120)
DEBUG | 17:27:33 448 [Router] handleReceived(REMOTE) (id=0x90c8aa1f fr=0xac to=0xac, WantAck=0, HopLim=3 Ch=0x0 Portnum=5 requestId=ae2fa03e rxtime=1728926853 priority=120)
DEBUG | 17:27:33 448 [Router] Module 'routing' wantsPacket=1
INFO  | 17:27:33 448 [Router] Received routing from=0xda5475ac, id=0x90c8aa1f, portnum=5, payloadlen=2
DEBUG | 17:27:33 448 [Router] Routing sniffing (id=0x90c8aa1f fr=0xac to=0xac, WantAck=0, HopLim=3 Ch=0x0 Portnum=5 requestId=ae2fa03e rxtime=1728926853 priority=120)
DEBUG | 17:27:33 448 [Router] Received a ACK for 0xae2fa03e, stopping retransmissions
DEBUG | 17:27:33 448 [Router] Delivering rx packet (id=0x90c8aa1f fr=0xac to=0xac, WantAck=0, HopLim=3 Ch=0x0 Portnum=5 requestId=ae2fa03e rxtime=1728926853 priority=120)
DEBUG | 17:27:34 448 [Router] Update DB node 0xda5475ac, rx_time=1728926853
DEBUG | 17:27:34 448 [Router] Forwarding to phone (id=0x90c8aa1f fr=0xac to=0xac, WantAck=0, HopLim=3 Ch=0x0 Portnum=5 requestId=ae2fa03e rxtime=1728926853 priority=120)
DEBUG | 17:27:34 448 [Router] Module 'routing' considered
INFO  | 17:27:34 448 [PowerFSM] Loss of power in Powered
DEBUG | 17:27:34 448 [PowerFSM] Enter state: ON
INFO  | 17:27:41 455 Rebooting
��ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x4213c90e
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:
```

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Talie5in, 2024-10-16:
I've encountered this on 2.5.5 on a Rak4631 as well, setting device to Router and the device is in a un-usable state but could still BLE connect to it from Android 2.5.0/2.5.1 (reboot, nodedb reset, factory reset commands do not in fact action just stay at 0%)

Was able to rescue this device running a DFU OTA over the top with same firmware which then has the device work as normal with existing config in-tact (including staying in router mode) (effectively a reboot)

It's like it goes to reboot, but never fully reboots.
No logs as this device is up a tower on a mountain :)

HarukiToreda, 2024-10-16:
according to the documentation Router role dissables the screen, and puts the node on power save mode, meaning that the ble and wifi will be off if there's no traffic detected, it will wake once it a packet is received to re-route it on ESP32.

https://meshtastic.org/docs/configuration/radio/device/#user-content-fn-2

[image]

Talie5in, 2024-10-16:
Not the case on non-esp32, Bluetooth stays enabled, however, wakes with packet, tried that, send it a DM or send a traceroute through it from 30m away didn't change it.

mtsdawson, 2024-10-16:
"Reset doesn't fix the issue. Only way to recover the node is to force another full erase and install after replugging USB and perfectly timing a reset by clicking the Erase Flash and Install button on the web flasher (it takes a few retries for timing to work and succeed with flash)."

HarukiToreda, 2024-10-16:
yes, that is normal on Router mode, it's meant to quickly go back to sleep to conserve power and retransmit. You're not meant to connect to it with a phone via Blutooth. This is why we recommend to only use Router role on top of mountains and such. It is working as intended. You don't want people grabbing that node and having access to your private mesh.

If you want to control that node and take it out of router role, you need to use an admin node to change the role via the Mesh and then you can get it on client role. or as you mentioned re-flashing.

closed 2024-10-16 by garthvh