meshtastic/firmware #5068: [Bug]: Import configuration fails at 74-78%

mtsdawson, 2024-10-14:
### Contact Details

skelzer

### What happened?

I flashed new Heltec v3 to 2.5.6 beta with "Full Erase and Install" - Node A. Then scanned QR code from web client to import channels. Then adjusted long/short names, enabled GPS, redefined pins to 48/47, changed LoRa freq slot to 20, changed BT fixed PIN to custom #, added admin key 1/3, and enabled legacy admin channel.  I then exported config and saved locally on phone.

I then flashed another Heltec v3 to same firmware - Node B.  When I try to import the config from Node A, it always stops at 74% on the first try and auto-reboots.  After reboot, it fails at 78% when I try again.  After another reboot, it fails at 74% when I try a third time.  Long/short names do not import, nor do channels.

I've tried this one 6+ different nodes, this is always the case.  There seems to be no scenario where the full config will import via Android BT, requiring the user to complete configuration via web client QR and manual entry.

I reflashed the node with each previous firmware and it only succeeded at 100% with build 2.4.1 (or might be 2.4.2?).  The error persists with builds 2.4.1/2+.

Thanks for amazing work on encryption features introduced with 2.5+!

### App Version

2.5.1

### Phone

OnePlus 8T / Android 14

### Device

Heltec v3 with GPS module (pins 48 & 47)

### Firmware

2.5.6 beta

### Relevant log output

```shell
None of the debug log entries seem relevant but happy to post anything if helpful.
```

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

andrekir, 2024-10-14:
https://github.com/meshtastic/firmware/pull/4679 added a `disableBluetooth()` call to admin commands preventing clients from receiving confirmation receipts.

https://github.com/meshtastic/firmware/pull/4898 made some changes to address this import/export issue but looks like still not 100%.

GMart, 2024-10-18:
Would like to add my findings as well. Heltec Tracker V1.1, Trying to import all settings on 2.5.7, and I get 83% before the node reboots. It then goes into a bootloop, but that's another issue.

madeofstown, 2024-10-18:
Must be a similar issue on Linux Native. Settings appear to upload successfully but the Short Name and Long Name settings don't get set. I haven't noticed if any other settings aren't being set but I mostly use defaults.

ctmalan, 2024-12-15:
Import configuration fails at 74-78% is a firmware update bug unrelated to the Android app.

This should be moved to the firmware upload part of the web client and this bug closed here.

https://github.com/meshtastic/Meshtastic-Android/issues/1462 should be reopended as applicable to Android.

c-:

closed 2025-02-21 by github-actions[bot]