meshtastic/firmware #5080: [Bug]: Cannot flash Heltec V3 using any method

va7wv, 2024-10-16:
### Category

Hardware Compatibility

### Hardware

Heltec V3

### Firmware Version

None

### Description

Brand new Heltec V3 off of Amazon.  Tried the web flasher initially without success.  Python esptool.py can see the device plugged in without any issue.  Using Linux.

I can read that it is an ESP S3 right off the chip.  I am trying Heltec V3 firmware files across all the releases, nothing works.

install bash script output:
```
./device-install.sh firmware-heltec-v3-2.5.6.d55c08d.bin
Trying to flash firmware-heltec-v3-2.5.6.d55c08d.bin, but first erasing and writing system information
esptool.py v4.8.1
Found 33 serial ports
Serial port /dev/ttyUSB0
Connecting.........
Detecting chip type... Unsupported detection protocol, switching and trying again...
Connecting.........
Detecting chip type... ESP32
Chip is ESP32-D0WD (revision v1.0)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: 0c:b8:15:2e:da:98
Uploading stub...
Running stub...
Stub running...
Erasing flash (this may take a while)...
Chip erase completed successfully in 0.8s
Hard resetting via RTS pin...
esptool.py v4.8.1
Found 33 serial ports
Serial port /dev/ttyUSB0
Connecting...........
Detecting chip type... Unsupported detection protocol, switching and trying again...
Connecting..........
Detecting chip type... ESP32
Chip is ESP32-D0WD (revision v1.0)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: 0c:b8:15:2e:da:98
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Unexpected chip id in image. Expected 0 but value was 9. Is this image for a different chip model?
```

esptool.py output:
```esptool.py v4.8.1
Found 33 serial ports
Serial port /dev/ttyUSB0
Connecting...........
Detecting chip type... Unsupported detection protocol, switching and trying again...
Connecting...........
Detecting chip type... ESP32
Chip is ESP32-D0WD (revision v1.0)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: 0c:b8:15:2e:da:98
Uploading stub...
Running stub...
Stub running...
Warning: ESP32 has no Chip ID. Reading MAC instead.
MAC: 0c:b8:15:2e:da:98
Hard resetting via RTS pin...
```

lsusb output:
```
Bus 002 Device 031: ID 1a86:7523 QinHeng Electronics CH340 serial converter
```

caveman99, 2024-10-16:
Try a different USB cable. If that does not work, try another USB cable. repeat till you succeeed. Ferrite loops may also help.

closed 2024-10-16 by caveman99

va7wv, 2024-10-17:
For anyone in the future, it was the USB hub that was the issue.  The quality of the USB cable may be an issue as stated.  I used an Anker brand USB-A to USB-C cable.  I was able to see the Heltec V3 using "lsusb", but this wasn't enough. 

Under Linux, use "sudo dmesg" to ensure that you can see the Heltec V3 being enumerated properly, something like the following.  If you don't get this, try going straight to a USB port with a decent quality cable.  I get the drift that these Heltecs are sensitive.

```
 5757.709210] usb 1-1: new full-speed USB device number 8 using xhci_hcd
[ 5757.866146] usb 1-1: New USB device found, idVendor=10c4, idProduct=ea60, bcdDevice= 1.00
[ 5757.866152] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 5757.866154] usb 1-1: Product: CP2102 USB to UART Bridge Controller
[ 5757.866156] usb 1-1: Manufacturer: Silicon Labs
[ 5757.866158] usb 1-1: SerialNumber: 0001
[ 5757.902231] usbcore: registered new interface driver cp210x
[ 5757.902254] usbserial: USB Serial support registered for cp210x
[ 5757.902293] cp210x 1-1:1.0: cp210x converter detected
[ 5757.904839] usb 1-1: cp210x converter now attached to ttyUSB0
```