meshtastic/firmware #5082: [Enhancement]: Macro-exclude neighbor info module from mainline firmware

thebentern, 2024-10-16:
### Platform

Cross-Platform

### Description

The current state of Neighbor info module:

1. Serves no real world use outside of topology visualizations in third party maps
2. Not supported in any of the official apps or clients
3. Can easily create a significant amount of traffic with fairly large packets on larger meshes

andrekir, 2024-10-17:
excluding makes sense to me. it's mostly misused and often does more harm than good, especially when left enabled all the time.

GUVWAF, 2024-10-17:
True, but I would find it a shame to remove it while it clearly is still used, and there are ways to mitigate the misuse. 

Due to the introduction of `hop_start`, there is currently no need to transmit them via LoRa, unless you want to obtain the info from remote nodes. We’ve already introduced that nodes do not rebroadcast those packets when the module is not enabled, and the [minimum broadcast interval](https://github.com/meshtastic/firmware/blob/fbb6778415dac8d6e7e33d8c79510f77ecdb42fa/src/mesh/Default.h#L20) has been increased to two hours, but we’ll only see the effect of this when people update their firmware.

GUVWAF, 2024-10-17:
I think people mostly want to see who their direct neighbors are, and not the number. 

Another thing we could do is add the array of neighbors to the MQTT MapReport, and not broadcast it over LoRa (by default).

andrekir, 2024-10-17:
> Due to the introduction of `hop_start`, there is currently no need to transmit them via LoRa

that means moving NeighborInfo use primarily over MQTT? the official server drops NeighborInfo packets.

IMO it's too complicated for such a narrow use case vs. the limited value of the information it provides.

thebentern, 2024-10-17:
> that means moving NeighborInfo use primarily over MQTT? the official server drops NeighborInfo packets.
>
> IMO it's too complicated for such a narrow use case vs. the limited value of the information [...]

We could potentially relax that constraint if we allow NeighborInfo to be a MQTT-only transport packet

garthvh, 2024-10-17:
Agree with Andre, creates a lot of confusion and traffic for near zero benefit.

thebentern, 2024-10-17:
This is actually a good opportunity to finally implement the no-lora broadcast nodenum destination. This would essentially be MQTT for now, but also potentially BLE mesh or other adapters in the future.

valentintintin, 2024-10-19:
Send Neighbor only on MQTT is a great idea.
What could also be good is to send it over LoRa (for mountain nodes without Internet) but with 0 hop instead of the hop limit.
My gateway is at home and I receive the mountain directly so no need to rebroadcast this packet.
Is it something you already think about?

closed 2024-10-19 by thebentern

GiuStel, 2024-10-29:
> True, but I would find it a shame to remove it while it clearly is still used, and there are ways to mitigate the misuse.
>
> Due to the introduction of `hop_start`, there is currently no need to [...]

We make extensive use of mountain nodes that are not connected to the internet, and neighborinfo allows us to monitor the radio links and record any performance degradation that may indicate issues such as water infiltration, antenna damage, etc.

What alternative solution do you offer?

macvenez, 2024-11-01:
> We make extensive use of mountain nodes that are not connected to the internet, and neighborinfo allows us to monitor the radio links and record any performance degradation that may indicate issues such as water infiltration, antenna damage, etc.
> 
> What alternative solution do you offer?

I agree with you, that's maybe one of the most useful info for remote nodes on mountains which can take days between planning and hikes to reach.
Moving them to MQTT only will cut them away enirely from that feature.
Why just not giving the user (with a switch like I've seen doing for almost anything) the possibility to enable it and leaving it default to off?
I understand that this feature may not be used very widely (we actually use it always) but that's the only info we have about the real state of each node of the mesh which is the center of the project...

garthvh, 2024-11-01:
Uses way too much airtime, ruins the mesh for everyone.

iondulgheru, 2024-11-03:
This change is a slap in the face also for responsible users.  I would like to be able to monitor the neigbours link quality for my remote nodes without BLE or WIFI. Also it seems to introduce a bug, after I upgraded to 2.5.8 my nodes with RAK4631 started to be non-responsive for hours. The problem was fixed after I disabled Neighbour info. This problem still persists on 2.5.10, if I enable Neighbour info the nodes become unresponsive for hours, they recover only for a short a while. From my tests, if I send a Traceroute request to one of these nodes and Neighbour info is activated, the node does not respond anymore. On a node with 2.5.7 I don't have this problem. Also I downgraded to 2.5.7 a node with 2.5.8 this problem was not present anymore.

IZ1IVA, 2024-11-03:
I don't understand the hate this option is getting in here. Indeed it is an _option_, not a default and it's pretty valuable as a radio path debugging tool.

Perhaps a bigger default update interval might help?

In our 100+ node network we found that the best way to save airtime was switching to MediumFast. We sacrificed very little in range, but gained a lot more in all other aspects.

garthvh, 2024-11-03:
> I don't understand the hate this option is getting in here. Indeed it is an _option_, not a default and it's pretty valuable as a radio path debugging tool.
>
> Perhaps a bigger default update interval [...]

Nothing helped other than turning it completely off over Lora.

IZ1IVA, 2024-11-04:
> Nothing helped other than turning it completely off over Lora.

I didn't know you're participating in our Italian public network, how could I miss that?

fifieldt, 2024-11-04:
> I agree with you, that's maybe one of the most useful info for remote nodes on mountains which can take days between planning and hikes to reach (which of course, due to the lack of internet, can't connect to MQTT).

How about the ability to request it to a DM, using the admin key (with heavy limiting) for occasional "debugging"? Would that work? I'm assuming you wouldn't be checking the remote nodes that often, yeah? Like just when a new link was going up, or if a regular link stopped working?

GiuStel, 2024-11-04:
> Uses way too much airtime, ruins the mesh for everyone. Users had the ability to control this for two years and constantly misused it.

Our network operates almost exclusively via radio, and nearly all nodes use neighborinfo. Despite this, we have never encountered congestion issues; on the contrary, the airtime remains low. I can understand that in longFast mode congestion might be more likely, but in MediumFast mode, the problem does not exist.

I invite you to reconsider your decision and allow us to continue benefiting from a tool that helps us understand the network topology, monitor the quality of radio links, and engage in the meaningful work of improving them over time.

iondulgheru, 2024-11-04:
The Neighbour Info is useful for my remote nodes. What should I do if I still want this Neigbhour Info? Add a phone with mobile connection and BLE? Users should have the option to see the neighbour info data for the remote nodes. More restrictions could be applied, but the option should exist.
Also when I upgraded to 2.5.8 (which incroporated this change) multiple nodes with NRF52 started beeing unresponsive and worked properly only when I disabled the Neighbour info module.

marcoSanti, 2024-11-04:
> Nothing helped other than turning it completely off over Lora.

Perhaps i missed something , but do you have any proof that supports your statement? In italy there is a network of several hundred nodes that works, with the neighbors modules enabled, without having any more issues.

ponzano, 2024-11-04:
> Nothing helped other than turning it completely off over Lora.

I agree with the others, in my city we have about 100 nodes online over Lora and 160 including Mqtt and don’t have any issues using neighbor module (enabled on every device!). It is very VERY useful because enables us to understand how to optimize nodes locations to get better coverage or choose new places to connect remote users <ins>without</ins> needing to use mqtt. This module needs to be left optional (no, optional doesn’t mean that the users have the option to make a fork and modify the firmware themselves).

garthvh, 2024-11-04:
It uses at least a third d and often more than half the airtime everywhere it is turned on. Like more in an Italian mesh under a duty cycle.  We have a ton of data via the public MQTT, it is not really possible to have it on for a 200 node mesh and not have a major impact on airtime.

GUVWAF, 2024-11-04:
From when is that data? It’s true that previously the NeighborInfo module created too much traffic when the default interval was only 15 min. However, first already it was introduced that nodes that didn’t have it enabled would not rebroadcast it, and afterwards the default interval had been increased to 6 hours, and the maximum was set to 2 hours. This was only released on August 19, so indeed there may have been people using the 15 min. until quite recently (or still do). However, I think broadcasting it every 6 hours is not really an issue considering DeviceTelemetry, Position and EnvironmentTelemetry (if enabled) is sent every hour and NodeInfo every 3 hours.

I still think an option to transmit over LoRa (default to false) would be fine.

IZ1IVA, 2024-11-04:
> It uses at least a third and often more than half the airtime everywhere it is turned on. Likely more in an Italian mesh under a duty cycle. We have a ton of data via the public MQTT, it is not really possible to have it on for a 200 node mesh and not have a major impact on airtime.

You're perhaps assuming that all nodes have the NeighborInfo module turned on and that those who did turn it on are leaving the (too short?) default update interval untouched. This is not the case. Is it an edge case? I don't think so.

We maintain that MediumFast is probably a better default preset and solves multiple issues related to AirTime, as well as longer default NeighborInfo/telemetry updated intervals. In our mesh, we set them to 3600 or 7200 according to the coverage of a node and the traffic it needs to support.

ponzano, 2024-11-04:
> It uses at least a third and often more than half the airtime everywhere it is turned on. Likely more in an Italian mesh under a duty cycle.  We have a ton of data via the public MQTT, it is not really possible to have it on for a 200 node mesh and not have a major impact on airtime.

Are there really so many people turning it on? It's not even available in the apple app :/ 
What about putting a big disclaimer with examples on how much it impacts air time when a user tries to enable it? 
Something like: check this to allow enabling
- [ ] I understand that enabling neighbor module will have an impact on the network 

I think that even having the minimum time set to 12 hours would still allow us to get very valuable information while limiting traffic a lot. 
Another option could be giving the possibility to enable this module only to router nodes (even without mqtt).
Just please give us the option to choose!

GiuStel, 2024-11-04:
Anyone can check by himself here:
https://map.loraitalia.it/

[image]

Check each node airtime info by clicking respective icon...

[image]

At the moment nodes are sending neighborinfo packets mainly on an hour basis.

All the nodes you see on the map (except 5 operating in Longfast) are part of the same mesh and can reach each other.

ponzano, 2024-11-04:
> Anyone can check by himself here:
> 
> https://map.loraitalia.it/
> 
> 
> 
> [image]
> 
> 
> 
> Check each node airtime info by clicking respective icon...
> 
> 
> 
> [image]
> 
> 
> 
> Currently, nodes are mainly sending neighborinfo packets on an hourly basis.
> 
> 
> 
> All the nodes you see on the map (except for 5 operating in LongFast mode) are part of the same mesh and can reach each other.

I'll add with another example, counting 141 nodes online at the time of screenshot on our network (Lora mediumfast + LoraItalia mqtt)
[image]

[image]

garthvh, 2024-11-04:
Several million of these packets get blocked on mqtt every week, these are huge packets and tons of people turn it on from Android without any understanding of what it does. Meshtastic is far too big now for these honor system settings.

GUVWAF, 2024-11-04:
This might have been true previously, but we should really look at the impact of the new defaults and not make decisions based on the 15 min. interval of the past.
Depending on the number of neighbors, it’s not a big packet (12 bytes fixed and 8 bytes per neighbor) and it is now only allowed to be sent every 2 hours, and by default every 6 hours. This is the lowest default broadcast interval we have. Someone enabling EnvironmentTelemetry will burden the mesh more than this module.

ponzano, 2024-11-04:
> This might have been true previously, but we should really look at the impact of the new defaults and not make decisions based on the 15 min. interval of the past.
>
> Depending on the number of [...]

Agreed

garthvh, 2024-11-04:
From what I can see it is still a huge traffic monster, certainly can not be enabled on long fast, and MQTT only still seems like the best option.  Environment telemetry has a lot more utility.

GiuStel, 2024-11-04:
> From what I can see it is still a huge traffic monster, certainly can not be enabled on long fast, and MQTT only still seems like the best option. Environment telemetry has a lot more utility.

We've been using this 'monster,' set to send every 60 minutes, for a year on a network that will soon reach 200 nodes, and our airtime usage is minimal. We even manage to handle telemetry and exchange friendly messages without saturating anything. Could it be that the LongFast default is actually the source of the issues?

GUVWAF, 2024-11-04:
> From what I can see it is still a huge traffic monster, certainly can not be enabled on long fast, and MQTT only still seems like the best option. Environment telemetry has a lot more utility.

The data lags behind, because not everyone updates their nodes immediately. Utility is subjective, one might not be so interested in the temperature and humidity of a node nearby, but would like to know which nodes it can hear directly for node placement.

ponzano, 2024-11-06:
Is there any progress regarding this?

garthvh, 2024-11-07:
> > From what I can see it is still a huge traffic monster, certainly can not be enabled on long fast, and MQTT only still seems like the best option.  Environment telemetry has a lot more utility.
> 
> Could you show us the data about how much of a monster it is? We showed you ours and it isn't

I don't see much data supporting the idea that it does not impact the network a picture of a map does not prove much.  With the new two way trace route it is just not necessary over lora anymore.

I am seeing 50K packets a day just on long fast still for neighbor info, 5 million packets in the last two weeks.

macvenez, 2024-11-07:
> 
> I don't see much data supporting the idea that it does not impact the network a picture of a map does not prove much. With the new two way trace route and hops away it is just not necessary over lora anymore.
> 
> I am seeing 50K packets a day just on long fast still for neighbor info, 5 million packets in the last two weeks.

What's the total amount of packets we are talking about? That number doesn't mean anything without a reference....

BTW that [map](https://map.loraitalia.it) just reflects every data you will get through the app, you have like a control panel with every node in the mesh.

Please have a look at it and check airtime for nodes, even in most crowded zones it never reaches 10% (you can also have historical data).

I just can't understand why do we have to give up on a feature we always use just because someone is misusing it. Maybe another idea would be to limit to mqtt only on lower tier radio presets (like all long* ones).
We have data supporting it and anyone can check it themselves, that feature is definitely what helps us the most on making the mesh better by optimising it. If we can't have infos on remote nodes we won't be able to troubleshoot them.

ponzano, 2024-11-07:
> What's the total amount of packets we are talking about? That number doesn't mean anything without a reference....
>
>
>
> BTW that [map](https://map.loraitalia.it) just reflects every data you will get [...]

Totally agree, @garthvh we need more context to evaluate those numbers.
I once again suggest you take a look yourself at map.loraitalia.it 
You can see airtime and channel utilization for every node, just click on their name.

thebentern, 2024-11-07:
NeighborInfo exists only in a community mapping sense which is almost always tightly coupled to MQTT. Without support in any of the major apps, there is not going to be any traction with attempts to resurrect it.

iondulgheru, 2024-11-07:
@thebentern , @garthvh some people want to have a big picture of the network, which include device count in an area and signal strenght. Is there any way to collect the neighbour info data from remote nodes that don't have MQTT? It doesn't need to be so granular, it can be once per hour, or a few times a day.

GiuStel, 2024-11-07:
I'd like to continue discussing it, so I opened a thread here:

[https://github.com/orgs/meshtastic/discussions/48](https://github.com/orgs/meshtastic/discussions/48)

garthvh, 2024-11-07:
@ponzano i saw a whole bunch of 15 minute neighbor packets in your log. This feature is just not necessary over Lora, you can make the exact same map without the extra packets.

thebentern, 2024-11-07:
> Also, @thebentern, because you brought the term community into play, taking away an option that is not enabled by default and that is wanted by the community is a little bit exagerated. Until now Meshtastic was made popular by these communities, but in this case you act like you know better and what the users want and know doesn't matter. And acting this way, removing silently a feature, in a rush, and then say that there needs to be some effort to ressurect it, also looks exagerated. It doesn't mean I don't apreciate you and your work, is just this single case where I think a mistake was made.

There was nothing silent about this. It was not only discussed multiple times on the official Discord server where, but this Github Issue's existence alone is documentation that it was not "silently" removed and in fact there was a back and forth discussion of what the removal looks like. 

No one in the project is saying we know "better" either. The official firmware features are first and foremost driven by what the official apps support. The neighbor info module's conception was originally intended for the Meshtastic Network Management Client, which has sadly been without any maintainer for quite some time now. Additionally, none the core contributors of the official apps have taken enough interest to add support for NeighborInfo topology visualizations into their respective platforms for its entire existence. This feature unfortunately has been a _cart before the horse_ from the beginning, which is why this decision was made. 

> I'd like to continue discussing it, so I opened a thread here:
> 
> https://github.com/orgs/meshtastic/discussions/48

Not sure why this needs another thread?

GiuStel, 2024-11-07:
> > I'd like to continue discussing it, so I opened a thread here:
> > https://github.com/orgs/meshtastic/discussions/48
> 
> Not sure why this needs another thread?

Because we're talking on a closed thread and because I want to know what other people think about this.
I'd also want to use, enjoy and contribute the new Github Discussions and I carefully checked the recomendation "I have done a search for similar discussions*", and found nothing about Neighborinfo Module.
So please, gentleman, undo delete my post.

garthvh, 2024-11-07:
There is no need to create extra discussions, all the relevant core contributors and info is here.

GiuStel, 2024-11-07:
https://meshtastic.org/docs/legal/conduct/

Our Standards
Examples of behavior that contributes to a positive environment for our community include:

- Demonstrating empathy and kindness toward other people
- Being respectful of differing opinions, viewpoints, and experiences
- Giving and gracefully accepting constructive feedback
- Accepting responsibility and apologizing to those affected by our mistakes, and learning from the experience
- Focusing on what is best not just for us as individuals, but for the overall community

thebentern, 2024-11-07:
> https://meshtastic.org/docs/legal/conduct/

I'm not sure why you feel anyone here has violated the code of conduct... As state above, we have taken a direction here which aligns the firmware with the official apps. There is no ill will. We have heard yours and other's feedback. I am sorry this is disappointing to you, but the core contributors are responsible for maintaining this project are not interested in continuing development of it. Attempting to coerce them to carry a feature they don't believe aligns with the vision for the project is not fair.

GiuStel, 2024-11-07:
Listen, maybe you haven't followed the discussion closely, or perhaps I haven't been clear enough: no one here is trying to force anyone to do anything. You have a very clear idea, but it doesn’t reflect the reality of our use case. We showed you, but you demonstrated no interest whatsoever.

Since not everything happens on Discord (otherwise you wouldn’t be promoting the use of GitHub Discussions), I wrote a polite, constructive, non-accusatory post that didn’t violate any forum rules. Writing and translating it into your language took an hour of my time. **If something here is unfair and disrespectful, it’s erasing it with a stroke.**

I suppose we’re free to discuss the Neighborinfo module even if you’ve decided to remove it, right? Or is that forbidden? Can we use the forum to hear the opinions of other Meshtastic users, or is that also not allowed? Do we want to continue with another 50 posts under this issue? I don’t see the point.

From you, Thebentern, and from Garthvh, I didn’t find any empathy, nor any intention to listen to the reasoning of those who participated with differing opinions and experiences. I didn’t find openness to constructive feedback or willingness to try to find a solution that could work for everyone.

The essence of this discussion is ‘we talked about it on Discord, we’ve decided, and there’s nothing you can say to change our minds.’ 

Not bad!

GUVWAF, 2024-11-07:
> the core contributors are responsible for maintaining this project are not interested in continuing development of it 

I’m willing to add a “transmit over LoRa” setting, defaulting to false, while keeping the restriction of a minimum 2-hour interval, and default of 6 hours. Then in the documentation (and apps if they get the setting) there could be a warning to be cautious about enabling it. Would this PR be accepted?

iondulgheru, 2024-11-07:
> I’m willing to add a “transmit over LoRa” setting, defaulting to false, while keeping the restriction of a minimum 2-hour interval, and default of 6 hours. Then in the documentation (and apps if [...]

Sounds excellent. Thank you!

thebentern, 2024-11-07:
> I’m willing to add a “transmit over LoRa” setting, defaulting to false, while keeping the restriction of a minimum 2-hour interval, and default of 6 hours. Then in the documentation (and apps if [...]

If you want to take ownership of this module and ensure that the minimum is enforced, I'm okay with that. Personally I'd like to see the absolute minimum be 4 hours.

garthvh, 2024-11-07:
> I’m willing to add a “transmit over LoRa” setting, defaulting to false, while keeping the restriction of a minimum 2-hour interval, and default of 6 hours. Then in the documentation (and apps if [...]

I really don't think it should be available on long fast, maybe 6 hours.

ponzano, 2024-11-07:
> @ponzano i saw a whole bunch of 15 minute neighbor packets in your log. This feature is just not necessary over Lora, you can make the exact same map without the extra packets.

Oh I didn't know! So that means that we have very good air time and channel util even if some people have set the neighbor module to transmit data every 15 minutes! That's great!

GUVWAF, 2024-11-07:
> If you want to take ownership of this module and ensure that the minimum is enforced, I'm okay with that. Personally I'd like to see the absolute minimum be 4 hours.

Okay, I will open a PR to add the setting and change the absolute minimum to 4 hours.

thebentern, 2024-11-07:
> 
> Okay, I will open a PR to add the setting and change the absolute minimum to 4 hours.

@GUVWAF another thought for curbing improper usage we can consider is gating the LoRA transport to the Router role only. We shouldn't be interested in the transient topology changes of a Client node anyway

GUVWAF, 2024-11-07:
@thebentern Hmm, but not all static nodes should be assigned as Router. This limitation should not become an incentive to use the Router role.

thebentern, 2024-11-07:
> @thebentern Hmm, but not all static nodes should be assigned as Router. This limitation should not become an incentive to use the Router role.

You're right. It could create a perverse incentive to use Router role, which we want to avoid.

tekstrand, 2024-11-10:
> discussed multiple times on the official Discord server

I came here wanting to read more about this change, but couldn't find any links to those conversations in Discord. For me, that would be helpful. I also want to remind us all that Discord is a synchronous communication tool, while GitHub is asynchronous. We've already identified this fact evidenced by our practice of opening issues to document changes and allow discussion.

I believe we have an opportunity to be more explicit in our communication about changes that remove functionality, or significantly alters operation. Which brings me to the question that came to mind after all of this. How might we do this?

A few questions/ideas:
- How would we determine if a change is "significant"?
- If a significant change has been discussed in Discord, and the team decides to open an issue/PR, we should link to the Discord conversations that led to it.
- Maybe PRs or issues can be marked with a specific label that can be followed or auto posted to Discord, similar to #announcements 
- If a change is identified as significant, can we require the issue or PR be open in GitHub for a minimum amount of time, allowing the community an opportunity for async input?

PR #5087 (2024-10-19): Initial NODENUM_BROADCAST_NO_LORA implementation with NeighborInfo module
Closes #5082