meshtastic/firmware #5096: [Bug]: Not Getting Too Large Error from messages that are too large

garthvh, 2024-10-19:
### Category

Other

### Hardware

T-Lora v2 1.6

### Firmware Version

2.5.5

### Description

When sending messages larger than 255 bytes I never get back the expected routing error and am instead seeing errors.

### Relevant log output

```Shell
2024-10-19 11:28:06.025759-0700 Meshtastic[1387:215064] [📟 Radio] 💥 Can't decode protobuf reason='bytes overflow' pb_msgdesc 0x3f41dfe0
2024-10-19 11:28:06.027724-0700 Meshtastic[1387:215064] [📟 Radio] 💥 Error: ignoring malformed toradio
```

garthvh, 2024-10-19:
I am getting packet too large messages from 222 Bytes to 238 bytes

thebentern, 2024-11-28:
Closed with update to protobuf length restriction

closed 2024-11-28 by thebentern