meshtastic/firmware #5105: [Bug]: PKI DMs to remote MQTT connected nodes fail

ufergus, 2024-10-20:
### Category

Other

### Hardware

T-Beam S3

### Firmware Version

2.5.6.d55c08d

### Description

For our local club mesh we have a private mqtt broker to help bridge gaps in our coverage.  I was testing with another club member when we realized some DMs with new 2.5.6 nodes were not going through/or being received like expected.  Let me describe the test setup.

node_A <--LoRa--> node_B <---MQTT---> node_C <--LoRa--> node_D

In this setup we have four nodes, one LoRa connected node each behind two MQTT connected nodes.  All nodes are running firmware 2.5.6. 

DM from A to/from B works
DM from D to/from C works
DM from A to/from D works
DM from B to/from C works
DM from A to/from C fails
DM from D to/from B fails  

For the cases that pass, we get the Person Check-mark icon.
For the cases that fail, we the Cloud Check-mark icon.

In our testing, sending a DM from behind a MQTT connected node to a remote MQTT connected node fails to work.  Help?  Thank you!

GUVWAF, 2024-10-20:
Are you using the public MQTT server? There packets are zero-hopped (meaning they don't get forwarded over the mesh after going through an MQTT node). Using your own MQTT server, this should work if the nodes don't have "ignore MQTT" enabled (and your hop limit allows it).

I wouldn't know why it would work for A to D and not for A to C.

caveman99, 2024-10-20:
Are you running on a region that is rate limited? I wouldn't know how to explain this any other way. if it's filtering A->D would not work.

ufergus, 2024-10-20:
Did some additional testing today and it appears to be working now.  Sorry for the false report.

closed 2024-10-20 by ufergus