meshtastic/firmware #5106: [Bug]: Routing of messages of unknown channels in LOCAL_ONLY mode.

CamFlyerCH, 2024-10-20:
### Category

Other

### Hardware

Heltec V3

### Firmware Version

2.5.8.4575352d8

### Description

While monitoring my node in client or router mode with CLI "meshtastic --noproto", i found that messages from other channels don't get dropped besides the rebroadcast mode is set to LOCAL_ONLY. I think this should not be done and it is a bug. Example below.

### Relevant log output

```Shell
DEBUG | 11:02:01 510 [RadioIf] Lora RX (id=0x27ce8c95 fr=0xa8 to=0xff, WantAck=0, HopLim=2 Ch=0x2a encrypted rxSNR=-10 rxRSSI=-116 hopStart=3)
DEBUG | 11:02:01 510 [RadioIf] Packet received : 542ms
DEBUG | 11:02:01 510 [Router] Add packet record (id=0x27ce8c95 fr=0xa8 to=0xff, WantAck=0, HopLim=2 Ch=0x2a encrypted rxSNR=-10 rxRSSI=-116 hopStart=3)
WARN  | 11:02:01 510 [Router] No suitable channel found for decoding, hash was 0x2a!
DEBUG | 11:02:01 510 [Router] packet decoding failed or skipped (no PSK?) (id=0x27ce8c95 fr=0xa8 to=0xff, WantAck=0, HopLim=2 Ch=0x2a encrypted rxtime=1729422121 rxSNR=-10 rxRSSI=-116 hopS
DEBUG | 11:02:01 510 [Router] Module 'routing' wantsPacket=1
INFO  | 11:02:01 510 [Router] Received routing from=0xda585fa8, id=0x27ce8c95, portnum=-1054474215, payloadlen=62999
DEBUG | 11:02:01 510 [Router] Routing sniffing (id=0x27ce8c95 fr=0xa8 to=0xff, WantAck=0, HopLim=2 Ch=0x2a encrypted rxtime=1729422121 rxSNR=-10 rxRSSI=-116 hopStart=3)
INFO  | 11:02:01 510 [Router] Rebroadcasting received floodmsg
DEBUG | 11:02:01 511 [Router] enqueuing for send (id=0x27ce8c95 fr=0xa8 to=0xff, WantAck=0, HopLim=1 Ch=0x2a encrypted rxtime=1729422121 rxSNR=-10 rxRSSI=-116 hopStart=3 priority=64)
DEBUG | 11:02:01 511 [Router] txGood=19,txRelay=15,rxGood=53,rxBad=11
DEBUG | 11:02:01 511 [Router] rx_snr found. hop_limit:1 rx_snr:-10.000000
DEBUG | 11:02:01 511 [Router] rx_snr found in packet. As a router, setting tx delay:385
DEBUG | 11:02:01 511 [Router] Module 'routing' considered
WARN  | 11:02:01 511 [RadioIf] Can not send yet, busyRx
DEBUG | 11:02:01 511 [RadioIf] rx_snr found. hop_limit:1 rx_snr:-10.000000
DEBUG | 11:02:02 511 [RadioIf] rx_snr found in packet. As a router, setting tx delay:231
ERROR | 11:02:02 511 [RadioIf] ignoring received packet due to error=-7
DEBUG | 11:02:02 511 [RadioIf] Packet received (noise?) : 542ms
DEBUG | 11:02:02 511 [RadioIf] Starting low level send (id=0x27ce8c95 fr=0xa8 to=0xff, WantAck=0, HopLim=1 Ch=0x2a encrypted rxtime=1729422121 rxSNR=-10 rxRSSI=-116 hopStart=3 priority=64)
DEBUG | 11:02:02 511 [RadioIf] Packet transmitted : 542ms
DEBUG | 11:02:02 512 [RadioIf] Completed sending (id=0x27ce8c95 fr=0xa8 to=0xff, WantAck=0, HopLim=1 Ch=0x2a encrypted rxtime=1729422121 rxSNR=-10 rxRSSI=-116 hopStart=3 priority=64)
DEBUG | 11:02:03 512 [RadioIf] Lora RX (id=0x27ce8c95 fr=0xa8 to=0xff, WantAck=0, HopLim=0 Ch=0x2a encrypted rxSNR=5.25 rxRSSI=-84 hopStart=3)
DEBUG | 11:02:03 512 [RadioIf] Packet received : 542ms
DEBUG | 11:02:03 512 [Router] Found existing packet record for fr=0xda585fa8,to=0xffffffff,id=0x27ce8c95
DEBUG | 11:02:03 512 [Router] Found existing packet record for fr=0xda585fa8,to=0xffffffff,id=0x27ce8c95
DEBUG | 11:02:03 512 [Router] Add packet record (id=0x27ce8c95 fr=0xa8 to=0xff, WantAck=0, HopLim=0 Ch=0x2a encrypted rxSNR=5.25 rxRSSI=-84 hopStart=3)
DEBUG | 11:02:03 512 [Router] Ignoring dupe incoming msg (id=0x27ce8c95 fr=0xa8 to=0xff, WantAck=0, HopLim=0 Ch=0x2a encrypted rxSNR=5.25 rxRSSI=-84 hopStart=3)
DEBUG | 11:02:03 512 [Router] Incoming msg was filtered from 0xda585fa8
```

GUVWAF, 2024-10-20:
Thanks for reporting, this was working before, but broke with 2.5. It should work again after #5109.

closed 2024-10-20 by thebentern

CamFlyerCH, 2024-10-21:
I tested it and it is working !

PR #5109 (2024-10-20): Fix rebroadcasting encrypted packets when `KNOWN_ONLY`/`LOCAL_ONLY` is used
Fixes #5106. 

This slipped in in 6eabbaf4321a3694fb48b4a55331c53d2ebb5d63. Already during sniffing, the FloodingRouter is rebroadcasting, so it should return before this.