meshtastic/firmware #5107: [Bug]: PRIVATE_HW build parameter being ignored in RAK4631 platformio.ini

jbuck2005, 2024-10-20:
### Category

Other

### Hardware

Rak4631

### Firmware Version

2.5.8.2ba72c15

### Description

When trying to compile customized firmware (using a fresh clone of the master branch) for a device _based_ on the RAK4631, the firmware completely ignores the PRIVATE_HW flag. This seems to be a new issue as of a few days ago; I was previously able to compile my modified code base (version unknown) and flash the device with the PRIVATE_HW.

The VSC (PlatformIO) build environment is set to rak4631 ( Ctl-Shift-P -> PlatformIO: Pick Project Environment -> env:rak4631 )

Editing the file:   ./variants/rak4631/platformio.ini

changing line 6 from:
`build_flags = ${nrf52840_base.build_flags} -Ivariants/rak4631 -D RAK_4631`
to:
`build_flags = ${nrf52840_base.build_flags} -I variants/rak4631 -D PRIVATE_HW`

has absolutely NO effect on the device's reported hardware in the Meshtastic app (Android) nor on the MQTT mapping platforms.

Further, setting the build flag to something which **should** fail to compile also has no effect:
`build_flags = ${nrf52840_base.build_flags} -I variants/rak4631 -D DOES_NOT_EXIST`

I tried building firmware using the PIO build task, as well as the VSC build action (with the modified rak4631 platformio.ini file selected) and the result is identical.

### Relevant log output

```Shell
*  Executing task: C:\Users\User\.platformio\penv\Scripts\platformio.exe run --environment rak4631 

Processing rak4631 (board: wiscore_rak4631; platform: platformio/nordicnrf52@^10.5.0; framework: arduino)
---------------------------------------------------------------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/nordicnrf52/wiscore_rak4631.html
PLATFORM: Nordic nRF52 (10.6.0) > WisCore RAK4631 Board   
HARDWARE: NRF52840 64MHz, 243KB RAM, 796KB Flash
DEBUG: Current (stlink) External (jlink, stlink)
PACKAGES:
 - framework-arduinoadafruitnrf52 @ 1.10601.0+sha.4f591d0 
 - framework-cmsis @ 2.50700.210515 (5.7.0)
 - tool-adafruit-nrfutil @ 1.503.0 (5.3)
 - tool-sreccat @ 1.164.0 (1.64)
 - toolchain-gccarmnoneeabi @ 1.70201.0 (7.2.1)
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
[nanopb] No generation needed.
Found 71 compatible libraries
Scanning dependencies...
Dependency Graph
|-- RadioLib @ 7.0.2
|-- ESP8266 and ESP32 OLED driver for SSD1306 displays @ 4.4.1+sha.e16cee1
|-- OneButton @ 2.6.1
|-- arduino-fsm @ 2.2.0+sha.7db3702
|-- TinyGPSPlus @ 1.0.3+sha.71a82db
|-- Thread @ 0.0.0+sha.1ae8778
|-- Nanopb @ 0.4.9
|-- ErriezCRC32 @ 1.0.1
|-- NonBlockingRTTTL @ 1.3.0
|-- SparkFun ATECCX08a Arduino Library @ 1.3.1+sha.5cf62b3
|-- Crypto @ 0.4.0
|-- Adafruit BusIO @ 1.16.1
|-- Adafruit Unified Sensor @ 1.1.14
|-- Adafruit BMP280 Library @ 2.6.8
|-- Adafruit BMP085 Library @ 1.2.4
[... 683 lines cut ...]
Compiling .pio\build\rak4631\FrameworkArduino\wiring.c.o
Compiling .pio\build\rak4631\FrameworkArduino\wiring_analog.cpp.o
Compiling .pio\build\rak4631\FrameworkArduino\wiring_analog_nRF52.c.o
Compiling .pio\build\rak4631\FrameworkArduino\wiring_digital.c.o
Compiling .pio\build\rak4631\FrameworkArduino\wiring_private.c.o
Compiling .pio\build\rak4631\FrameworkArduino\wiring_shift.c.o
Archiving .pio\build\rak4631\libFrameworkArduino.a
Linking .pio\build\rak4631\firmware.elf
Checking size .pio\build\rak4631\firmware.elf
Building .pio\build\rak4631\firmware.hex
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"Generating UF2 file

RAM:   [==        ]  21.5% (used 53600 bytes from 248832 bytes)
Flash: [==========]  97.3% (used 793136 bytes from 815104 bytes)
Converting to uf2, output size: 1586688, start address: 0x26000
Wrote 1586688 bytes to C:\Users\User\Documents\GitHub\meshtastic\.pio\build\rak4631/firmware.uf2
Building .pio\build\rak4631\firmware.zip
Zip created at .pio\build\rak4631\firmware.zip
========================================================================= [SUCCESS] Took 78.42 seconds =========================================================================

Environment    Status    Duration
-------------  --------  ------------
rak4631        SUCCESS   00:01:18.420
========================================================================== 1 succeeded in 00:01:18.420 ========================================================================== *  Terminal will be reused by tasks, press any key to close it.
```

todd-herbert, 2024-10-20:
Just a quick observation: I notice that `RAK_4631` is also defined in the variant.h file for the RAK4631 platform, which seems a little bit non-standard (?).

Apparently it's been defined in there for a year or so, so I'm not sure what is unmasking the problem just now.

caveman99, 2024-10-20:
RAK_4631 especially is tied to a lot of things inside the source code that makes the module work as expected. You cann add another define to act upon in macro guards but please never replace it. If you want to signal a'private hw' you can add your macro to the architecture.h for nrf52 platform.

closed 2024-10-20 by caveman99

jbuck2005, 2024-10-20:
> RAK_4631 especially is tied to a lot of things inside the source code that makes the module work as expected. You can add another define to act upon in macro guards but please never replace it. If you want to signal a'private hw' you can add your macro to the architecture.h for nrf52 platform.

Why would this have worked previously though? (changing from RAK_4631 to PRIVATE_HW in the platformio.ini)

Further, what you are suggestion actually is contrary to what is posted as part of the development documentaion.
[https://meshtastic.org/docs/development/firmware/build/](https://meshtastic.org/docs/development/firmware/build/)