meshtastic/firmware #5108: [Feature Request]: Exclude unsupportable modules from firmware and apps

garthvh, 2024-10-20:
### Platform

NRF52, ESP32, RP2040, Linux Native, Cross-Platform

### Description

As a first step towards building customized firmware it would be good if we could start by disabling modules in the firmware and apps that are unsupportable on specific hardware.

The heltec wireless paper for example does not have pins that would make most GPIO related functionality work.

BrianG61UK, 2024-10-20:
You might as well forget the Heltec V3 altogether.
It's unusable.
There are an utterly ludicrously large number of bugs in the Heltec firmware and in client.mestastic.org on Chrome.

Talie5in, 2024-10-21:
@garthvh Outside of Apps (eg, iOS/Android Apps), a lot defines are already available for excluding modules at compile time - assume this would be code changes to signify to the phone apps what modules are/are not compiled in when connecting to the node and gathering module configs.

https://github.com/meshtastic/firmware/blob/master/src/configuration.h#L289L325

thebentern, 2024-11-20:
Closed with #5300
There are probably some stragglers, but we can fix those as they become evident.

closed 2024-11-20 by thebentern