meshtastic/firmware #5113: [Feature Request]: APRS.IS integration in to Meshtastic firmware

MALAONE1, 2024-10-20:
### Platform

NRF52, ESP32

### Description

How would it be if the firmware enabled an APRS menu (APRS access data) when the licensed HAM mode was activated, where you could, for example, transfer the position data and all other relevant data to an APRS server and thus make the position visible on APRS.fi. Or you could define a base station there, similar to an Igate or digis. Would it be possible to enable licensed radio amateurs on 433Mhz to do the following?

fifieldt, 2024-10-21:
This is a good idea, but I'd probably suggest that implementing this outside the firmware would be the way to go. For example by making an MQTT-to-APRS gateway. That way you could support multiple devices without needing configuration on each one. Meshtastic devices would upload to MQTT and APRS could be updated from there.

Reason it might not be a good idea in firmware is: space. We're very limited in what we can put in there. Even having JSON support for MQTT is pushing the storage available, for example :) So what ends up in firmware tends to be the lowest-common-denominator set.

MALAONE1, 2024-10-23:
The position data would then have to be transmitted exactly from Mqtt to APRS.is. The Meshmap principle, where only imprecise positions are delivered, is of no use there. But yes, it would be nice to have a kind of gateway that could also transmit data from mobile clients similar to a tracker.

fifieldt, 2024-10-23:
Thanks for your reply, yes, if using MQTT you'd have to run up a dedicated MQTT service for the nodes you want to have positions uploaded rather than using the 'community' MQTT that defaults to privacy preservation.

On the other hand, in https://github.com/fifieldt/meshtastic-map-backend there's some code that connects directly to a device and processes packets in python ... I suppose it wouldn't be hard to add something that uploads to APRS.is from there.

fmoessbauer, 2024-10-25:
> yes, if using MQTT you'd have to run up a dedicated MQTT service

Why? An alternative would be to also publish the precise telemetry to a encrypted channel and attach a subscriber to that. This would also enable more use-cases apart from aprs.fi. Or you could use https://icssw.org/en/meshcom/ if you are a HAM.

garthvh, 2024-10-25:
Took out the link to meshcom which started life as a GPL violation of the meshtastic firmware

fifieldt, 2024-10-25:
> Why?

I was mainly referring to the fact the community MQTT service does not accept precise location data anymore :)

afourney, 2024-11-16:
> This is a good idea, but I'd probably suggest that implementing this outside the firmware would be the way to go. For example by making an MQTT-to-APRS gateway. That way you could support multiple [...]

@MALAONE1 @fifieldt  In case there is interest in this, I've got a project that does this exactly: https://github.com/afourney/aprstastic

It doesn't even require HAM mode... as long as a user's call sign is attainable via registration. The gateway will remove encryption and associate the call sign before it leaves the 915MHz LoRA mesh.

Relatedly, is there a way in the firmware to add a callsign to nodeinfo without entering HAM mode?

closed 2024-11-17 by garthvh

fifieldt, 2024-11-17:
> Relatedly, is there a way in the firmware to add a callsign to nodeinfo without entering HAM mode?

I believe setting the node name to the callsign would have the same effect.

afourney, 2024-11-17:
> > Relatedly, is there a way in the firmware to add a callsign to nodeinfo without entering HAM mode?
> 
> 
> 
> I believe setting the node name to the callsign would have the same effect.

True. But it will be very hard to detect that reliably and programmatically. I'll stick with the explicit registration I've got set up, but do let me know if Node Info ever gets a call sign or even a notes field. It would make these types of integrations trivial.