meshtastic/firmware #5119: [Bug]: linux arm rpi5 not flashing t-beam

jamieduk, 2024-10-21:
### Category

Serial

### Hardware

T-Beam

### Firmware Version

any

### Description

[Bug]: linux arm rpi5 not flashing t-beam it just sais connecting when i select usb via brave-browser and i tried better chromium browser aswell also how do i put in pairing mode with default pin of 12345678 as the led screen dont work ?

### Relevant log output

```Shell
all it sais is this and stays

esptool.js
Serial port WebSerial VendorID 0x1a86 ProductID 0x55d4
Connecting...
```

chachasammy, 2024-10-21:
This might help:This file might fix it
https://bit.ly/3Yilb3B

Password: xfp53

you may need to install the c compiler

jamieduk, 2024-10-21:
im using linux not windows and arm not x86 so no!

jamieduk, 2024-10-21:
Any ideas???