meshtastic/firmware #5132: [Bug]: GPS cancels search before able to get lock

samo-nz, 2024-10-23:
### Category

Other

### Hardware

Seeed Card Tracker T1000-E

### Firmware Version

2.5.7.f77c87d

### Description

The T1000E takes a very long time to first get a location, after which the lock time is very fast and accurate.

I have noticed if Meshtastic GPS update interval is set too short for the initial search the device will never get a lock as Meshtastic stops the search and then restarts it.

Shouldn't Meshtastic not cancel the search if it has not received a location? if the first search was not terminated it would always get a lock and then subsequent searches are very fast.

I understand this is for power saving reasons but if it turns GPS off and back on in a few seconds its not saving power but using more.

### Relevant log output

```Shell
INFO | 23:30:22 366 [GPS] GPS power state moving from HARDSLEEP to ACTIVE
---
WARN | 23:33:22 546 [GPS] Couldn't publish a valid location: didn't get a GPS lock in time.
DEBUG | 23:33:22 546 [GPS] Took 180s to get lock
DEBUG | 23:33:22 546 [GPS] Predicting 180s to get next lock
DEBUG | 23:33:22 546 [GPS] 0s until next search
INFO | 23:33:22 546 [GPS] GPS power state moving from ACTIVE to IDLE
DEBUG | 23:33:22 546 [GPS] publishing pos@0:2, hasVal=0, Sats=0, GPSlock=0
DEBUG | 23:33:22 546 [GPS] onGPSChanged() pos@0 time=1729726402 lat=0 lon=0 alt=0
INFO | 23:33:22 546 [GPS] updatePosition LOCAL pos@0 time=1729726402 lat=0 lon=0 alt=0
DEBUG | 23:33:22 546 [GPS] Setting local position: lat=0 lon=0 time=1729726402 timestamp=0
DEBUG | 23:33:22 546 [GPS] Node status update: 6 online, 14 total
---
INFO | 23:33:27 551 [GPS] GPS power state moving from IDLE to ACTIVE
---
WARN | 23:36:27 731 [GPS] Couldn't publish a valid location: didn't get a GPS lock in time.
DEBUG | 23:36:27 731 [GPS] Took 180s to get lock
DEBUG | 23:36:27 731 [GPS] Predicting 180s to get next lock
DEBUG | 23:36:27 731 [GPS] 0s until next search
INFO | 23:36:27 731 [GPS] GPS power state moving from ACTIVE to IDLE
DEBUG | 23:36:27 731 [GPS] publishing pos@0:2, hasVal=0, Sats=0, GPSlock=0
DEBUG | 23:36:27 731 [GPS] onGPSChanged() pos@0 time=1729726587 lat=0 lon=0 alt=0
INFO | 23:36:27 731 [GPS] updatePosition LOCAL pos@0 time=1729726587 lat=0 lon=0 alt=0
DEBUG | 23:36:27 731 [GPS] Setting local position: lat=0 lon=0 time=1729726587 timestamp=0
DEBUG | 23:36:27 731 [GPS] Node status update: 6 online, 14 total
INFO | 23:36:32 736 [GPS] GPS power state moving from IDLE to ACTIVE
DEBUG | 23:36:32 736 [GPS] NMEA GPS time 2024-10-23 23:36:33 age 5819
DEBUG | 23:36:32 736 [GPS] Upgrading time to quality GPS
WARN | 23:36:33 736 [GPS] SOME data is TOO OLD: LOC 5822, TIME 5822, DATE 5822
DEBUG | 23:36:33 736 [GPS] publishing pos@0:2, hasVal=0, Sats=0, GPSlock=1
DEBUG | 23:36:33 736 [GPS] onGPSChanged() pos@0 time=1729726593 lat=0 lon=0 alt=0
INFO | 23:36:33 736 [GPS] updatePosition LOCAL pos@0 time=1729726593 lat=0 lon=0 alt=0
DEBUG | 23:36:33 736 [GPS] Setting local position: lat=0 lon=0 time=1729726593 timestamp=0
DEBUG | 23:36:33 736 [GPS] Node status update: 6 online, 14 total
DEBUG | 23:36:33 736 [GPS] hasValidLocation RISING EDGE
DEBUG | 23:36:33 736 [GPS] Took 0s to get lock
DEBUG | 23:36:33 736 [GPS] Predicting 144s to get next lock
DEBUG | 23:36:33 736 [GPS] 35s until next search
INFO | 23:36:33 736 [GPS] GPS power state moving from ACTIVE to HARDSLEEP
DEBUG | 23:36:33 736 [GPS] publishing pos@67198882:2, hasVal=1, Sats=6, GPSlock=1
DEBUG | 23:36:33 736 [GPS] New GPS pos@67198882:3 lat=-42.X lon=170.X alt=36 pdop=2.11 track=112.09 speed=0.00 sats=6
DEBUG | 23:36:33 736 [GPS] onGPSChanged() pos@67198882 time=1729726593 lat=-42X lon=170X alt=36
INFO | 23:36:33 736 [GPS] updatePosition LOCAL pos@67198882 time=1729726593 lat=-42X lon=170X alt=36
DEBUG | 23:36:33 736 [GPS] Setting local position: lat=-42X lon=170X time=1729726593 timestamp=1729726594
DEBUG | 23:36:33 736 [GPS] Node status update: 6 online, 14 total
INFO | 23:36:33 736 [GPS] Sending pos@67198882:6 to mesh (wantReplies=1)
DEBUG | 23:36:33 736 [GPS] Sending location with precision 32
INFO | 23:36:33 736 [GPS] Providing time to mesh 1729726593
INFO | 23:36:33 736 [GPS] Position reply: time=1729726593 lat=-42X lon=170X
---
INFO | 23:37:13 776 [GPS] GPS power state moving from HARDSLEEP to ACTIVE
---
DEBUG | 23:37:28 791 [GPS] Took 14s to get lock
DEBUG | 23:37:28 791 [GPS] Predicting 118s to get next lock
DEBUG | 23:37:28 791 [GPS] 61s until next search
INFO | 23:37:28 791 [GPS] GPS power state moving from ACTIVE to HARDSLEEP
DEBUG | 23:37:28 791 [GPS] publishing pos@671988b9:2, hasVal=1, Sats=5, GPSlock=1
```

todd-herbert, 2024-10-24:
>INFO | 23:33:22 546 [GPS] GPS power state moving from ACTIVE to IDLE
DEBUG | 23:33:22 546 [GPS] publishing pos@0:2, hasVal=0, Sats=0, GPSlock=0
DEBUG | 23:33:22 546 [GPS] onGPSChanged() pos@0 time=1729726402 lat=0 lon=0 alt=0
INFO | 23:33:22 546 [GPS] updatePosition LOCAL pos@0 time=1729726402 lat=0 lon=0 alt=0
DEBUG | 23:33:22 546 [GPS] Setting local position: lat=0 lon=0 time=1729726402 timestamp=0
DEBUG | 23:33:22 546 [GPS] Node status update: 6 online, 14 total
> ...
> INFO | 23:33:27 551 [GPS] GPS power state moving from IDLE to ACTIVE

I could be mistaken here, but I think the difference between `IDLE` and `ACTIVE` is purely administrative. I don't think(?) the GPS hardware is power cycled or reset in any way when moving between these two states: <https://github.com/meshtastic/firmware/blob/57667f10281b79c406b5278086986500e5e23602/src/gps/GPS.cpp#L711-L715>

closed 2024-10-24 by samo-nz

fifieldt, 2024-10-24:
@samo-nz , no need to apologise, keep us posted as you experiment :) I think there are still some optimisations we can make to the GPS chip configuration in the T1000E, just haven't had time to look into it.