meshtastic/firmware #5144: [Feature Request]: Add no-ack mode for SerialModule

lesykm, 2024-10-25:
### Platform

Cross-Platform

### Description

Sometimes it is more useful to send packets as fast as possible and not wait for acks (for example, when higher-level protocols handle retransmits). In that case overhead and delays of acks in undesirable.

lesykm, 2024-10-25:
Will add PR in this repo after protobufs lands.

garthvh, 2024-10-25:
For the text message mode? If you want this you need to use the proto mode where you can set want ack on each packet. Flooding the mesh in text mode seems like a really problematic addition.

closed 2024-10-25 by thebentern

lesykm, 2024-10-29:
@garthvh @thebentern
This was made for higher level protocols (think xmodem, ppp, etc) over [SIMPLE](https://github.com/meshtastic/firmware/blob/e12fd27b498a67d68fd3daeccdf6166f5488056f/src/modules/SerialModule.cpp#L338) mode (just bytestream). Using protobufs for this would require a lot of changes for decades established software, including Linux kernel.
I tested with my PR changes and it works much faster and reliable than with ACKs.

@garthvh, regarding "Flooding the mesh in text mode seems like a really problematic addition."
2. Actually it is vice-versa.
With existing code - on each data packet we have 2X radio packets in ether - sent + ack.
With ACKs turned off we have 1X radio packets in ether - just sent.

Considering speed you closed such simple issue and attached PRs - looks like I missed something in contribution guidelines.
What is the correct way to contribute to Meshtastic? 
Do I need firstly discuss this with eng team on Discord or something?

thebentern, 2024-10-29:
@lesykm perhaps a new TEXT_NO_ACK serial mode would be acceptable, but as I mentioned earlier, my preferred route would be that if you want to more advanced control over the packets sent out that you use the existing protobuf mode, as it gives you full control. I believe adding an additional boolean config that only applies to one mode is not good housekeeping for the serial config section.

garthvh, 2024-10-29:
The simple and text modes are explicitly meant to not require a bunch of configuration, the protos mode allows for full control. Opening an issue first is usually best if you don't want to be frustrated that the feature is not in line with how a module is designed.

lesykm, 2024-10-29:
> perhaps a new TEXT_NO_ACK serial mode would be acceptable, but as I mentioned earlier, my preferred route would be that if you want to more advanced control over the packets sent out that you use [...]

Got it. SIMPLE_NO_ACK (TEXT is not an option as it is not plain bytestream) will work for me :). If I implement it, is there a chance it will be accepted?

I completely understand great features of protobufs mode (I develop binary protocol applications (though Thrift, not Protobufs) on a daily basis), but it will require development of extra pieces of software on sender and receiver parts (and it is not always possible to do that in embedded system).
While SIMPLE_NO_ACK mode will allow to pass bytes without any changes to sender/receiver software. Current SIMPLE mode also allows this but waiting (and sometimes losing) ACKs brings a lot of overhead, delays and reduces reliability.

lesykm, 2024-10-29:
> The simple and text modes are explicitly meant to not require a bunch of configuration, the protos mode allows for full control. Opening an issue first is usually best if you don't want to be frustrated that the feature is not in line with how a module is designed.

I get it, will it be OK if I will add new mode as @thebentern suggested above?

> Opening an issue first is usually best if you don't want to be frustrated that the feature is not in line with how a module is designed.

Done! Can this one be reopened please? :)
I promise to add myself block time period - to not to file PRs after opening a new issue for a week to not frustrate people :)