meshtastic/firmware #5150: [Bug]: T-Deck locks up on builds 2.5.7 and 2.5.8

N1obu, 2024-10-26:
### Category

Hardware Compatibility

### Hardware

T-Deck

### Firmware Version

2.5.7

### Description

Upgraded from 2.5.6 to 2.5.8 and the T-Deck locks up.  Both keyboard and serial interface won't work.  Tried to downgrade to 2.5.7 with the same results.  Downgraded back to 2.5.6 and everything works fine.

caveman99, 2024-10-26:
is this a T-Deck or a T-Deck Plus?

N1obu, 2024-10-26:
It's just a regular T-Deck

agajdosi, 2024-10-30:
I have a similar issue. My non-Plus T-Deck gets stuck and then reboots quite randomly. Happens only when I touch the screen/buttons, when using it via Android app it is stable.

N1obu, 2024-11-16:
I just tried 2.5.12        Still locks up.     Downgraded back to 2.5.6

brad28b, 2024-11-25:
Go into your "Power" settings, make sure the option to conserve power if on a limited source ie solar, is turned OFF.

thebentern, 2024-11-25:
Closed with #5422

closed 2024-11-25 by thebentern