meshtastic/firmware #5167: [Bug]: display shutting down when external notification set to pwm, vibrate and buzzer

adamstern2000, 2024-10-28:
### Category

Other

### Hardware

T-Beam S3

### Firmware Version

2.5.6

### Description

Affects 2.4.3, 2.5.4, 2.5.6 and probably many more.
I am using a t-beam s3 core with 5v buzzer and 3.3v vibrator.

external notification settings:
Enabled true
Alert bell true
Alert message true
Use pwm true
Active true
Output pin 3
Output duration unset
Nag unset
Alert buzzer on bell true
Alert vibra on bell true
Alert buzzer on message true
Alert vibra on message true
Output pin buzzer pin 3
Output pin vibra 46

### Relevant log output

```Shell
-
```

garthvh, 2024-10-28:
Pretty likely you have a GPIO pin conflict with the buzzer.

todd-herbert, 2024-10-28:
Are you certain that your device is the T-Beam S3 Core? If your device is instead a T-Beam S3 Supreme, the pin conflict would be with IO3, which is used by the LoRa radio.

adamstern2000, 2024-10-28:
Thanks Todd.  That is interesting- I’ll have to fix that.  But I’ve also tested on ip38, and it does the exact same thing.

and you are right- it’s the supreme.  I loaded the firmware for s3 core.  Maybe that’s the issue.  Which firmware for t-beam supreme?

adamstern2000, 2024-10-28:
When I disable the pwm, the issue goes away.  But the beep becomes a chirp and the vibration doesn’t work.  

The vibration has always been on the same.pin.  The buzzer has been on 38 and 3 - depending on my build, and it happens on both configs.