meshtastic/firmware #5171: [Bug]: Station G2 Freeze with wifi and Bluetooth on

Madmesh, 2024-10-28:
### Category

Hardware Compatibility, BLE, WiFi

### Hardware

Station G2

### Firmware Version

2.5.7

### Description

The configuration allows for both the wifi and blue tooth to be on at the same time. This causes the wifi to loose connection and the unit to freeze after about 20 hours. Turning off either wifi or Bluetooth seems to correct the issue.  With the 2.4 firmware, I think it turned off bluetooth after a while? Just was not an issue before the 2.5. For me anyways. I can not provide a log file as the unit seems to freeze and stop logging.

garthvh, 2024-10-28:
They are not on at the same time, they use the same radio

fifieldt, 2024-10-28:
Confirming what @garthvh wrote - this device does not support wifi and Bluetooth on at the same time.

closed 2024-10-28 by fifieldt