meshtastic/firmware #5175: [Bug]: firmware 2.5.x: no plaintext private messages anymore

dl9sau, 2024-10-28:
### Category

Other

### Hardware

T-Beam

### Firmware Version

2.5.6.d55c08d

### Description

First let me say: crypto by default is good.

As licensed amateur radio operators we experiment on our frequencies with meshtastic.
Amateur radio does not allow (by law) encryption. This is good, because the sense of amateur radio is that it is "open": everyone is be able to listen.
We operate on non-encrypted ("channels" (I mean the 7 message board slots)).
Direct contact (in plaintext) was possible in firmware 2.4.x.
Since firmware 2.5, unencrypted messaging does not work.

Fix: old unencrypted behavior by honoring the switch  "Licensed amateur radio" in "User" config.

garthvh, 2024-10-28:
Should work between two nodes in licensed user mode, it won't work to node with PKC and no license.

mertensm72, 2024-10-28:
+1 for hams indispensable

dl9sau, 2024-10-28:
Update:
we testet a bit more.

I run v2.5.6.
I can write to a v2.4.x user, and vice versa.

If I write to a v2.5.x user, I immediately get the crossed-out cloud symbol.
I suppose nothing is sent on rf.
The app shows in the "debug log":

Packet28.10.24 19:28:04from: 2053897304 (!7a6bfc58) (!7a6bfc58)
to: 2053897304
decoded {
  portnum: ROUTING_APP
  payload: "\030\""
  request_id: 3921883325
}
id: 351920580
rx_time: 1730140084
hop_limit: 7
priority: ACK

Packet28.10.24 19:28:03to: 1129910 (!00000000)792
decoded {
  portnum: TEXT_MESSAGE_APP
  payload: "Test6"
}
id: 3921883325
want_ack: true
public_key: "\\\033[\206\314I-<\300o;:[i\a\333u\256\260\237V\317F{1q\213\347AK\352j"
pki_encrypted: true

The routing_app messages follows each message. I tried to write to several v2.5 users, same effect.
The routing_app message refers to the IDof the message. "From" (number and hex-number, which is the same value) and "To" (same number) seems a message from the tracker to the app. I assume for indicating that something went wrong.
The "payload" "\030\"" is  part of a protocol and not obviously - perhaps it says "I'm the cloud and nothing has been sent".

I'm not the only v2.5 user with this problem (actually their problems was the reason I dig into); I encouraged them to test what I did and to look what's seen in their debug log.

Btw, who is responsible for the crypto, the app or the tracker? I ask for better understanding. I see the unencrypted plaintext message going to the tracker, along with the public_key, andthe tag "pki_encrypted: true".

garthvh, 2024-10-28:
Those users have encryption, which is unavailable in licensed mode. The new crypto is obvious not available in licensed user mode.

garthvh, 2024-10-28:
> +1 for hams indispensable
Hams can't use the new dm encryption.

garthvh, 2024-10-28:
Using the --set-ham command from the cli should set up licensed user mode properly.

iondulgheru, 2024-10-29:
Before 2.5 users that had "Licensed amateur radio" enabled could send a direct message to another user that had "Licensed amateur radio" also enabled. After 2.5 is not possible anymore. To be in line with the law requests, why can't we delete the encryption keys so we can still write a direct message that is unencrypted?

dl9sau, 2024-10-29:
To v2.4 users, the meshtastic-app debug log shows this:

Packet29.10.24 07:28:32from: 2053897304 (!7a6bfc58) (!7a6bfc58)
to: 2053897304
decoded {
  portnum: ROUTING_APP
  payload: "\030\000"
  request_id: 3921883388
}
id: 76012911
rx_time: 1730183312
hop_limit: 7
priority: ACK

Packet29.10.24 07:28:22to: 1384546678 (!52868176)
decoded {
  portnum: TEXT_MESSAGE_APP
  payload: "Test"
}
id: 3921883388
want_ack: true

=> Without "public_key: " and "pki_encrypted: true".

What do you think, it may more propably a bug in the meshtastic-app than in the firmware?
The meshtastic-app is configured correctly (licensed ham). In Settings->Security she shows my public/private keys (which have been autogenerated). There's no option to remove the keys. My own symbol has a green key symbol (like every other v2.5 user heard on rf).
On the other hand, the concept flaw must not be only in the app (and the tracker should know about the crypto state), because else, if the app is not connected, the tracker should change the key symbol in it's node beacon to a yellow unlocked key - and this would make no sense.

iondulgheru, 2024-10-29:
> Using the --set-ham command from the cli should set up licensed user mode properly.

@garthvh , I used the --set-ham command from the cli and direct messaging still doesn't work. I even tried nodedb reset and deleting older chats. In the end I also tried factory reset, reapplying the configuration manually and using again the --set-ham command from the cli.
To test this I am using 2 ESP32 nodes that are on my desk, with firmware 2.5.7, meshtastic cli 2.5.3 and Meshtastic Android app v.2.5.2.

dl9sau, 2024-10-29:
It looks like meshtastic-firmware and meshtastic-app are incompatible
to each other.
 
1) App
If the app (in ham mode) on receiption of "is_ham" taged node-packets
wouldn't mark them with a green closed key but with a yellow open
key, and if it would use the plaintext-Message format (which it
uses towards v2.4 users), messaging would be possible again.
 
2) firmware
If in the own node info broadcast, the tracker would mark no-crypto
(I don't know the protocol encoding in detail), which means that
the beacon looks like that of a firmwate v2.4 user, others would
see my node with a yellow open key and messaging would work.
Or: if, in ham-mode, the firmware would not indicate to the app
on received nodes that crypto is available, then messaging would work.
 
Perhaps it could be fixed on both sides, in the app and in the firmware.
Better safe than sorry ;)
 
Which approach is the best one is up to the developers of the
firmware and the app.
 
Happy fixing

dl9sau, 2024-11-11:
App version 2.5.7 from this weekend fixes the problem (tested with android app), with this patch:
https://github.com/meshtastic/Meshtastic-Android/commit/414e0bb5449c44226ff991265673f94431abc367

At least mostly. Because:
while with my device contacts are displayed with yellow opened lock, someone in our region with the same app version complained he sees some users with a red, crossed, lock - which should not occur.
I did not dig into how crypto keys are exchanged (node info broadcasrs?). Anyway, he deleted the list of known nodes and will look if that problem has gone.

I'll cross-reference this tickte (which supposedly may closed) to android-app-tree.

fifieldt, 2024-11-16:
Thanks for your patience and proactively following up. Closing now due to the noted fix in the Android app.

closed 2024-11-16 by fifieldt