meshtastic/firmware #5185: [Bug]: MQTT Json error on RAK WisBlock 4631 with firmware 2.5.8.

Helbigmedia, 2024-10-29:
### Category

Hardware Compatibility

### Hardware

Rak4631

### Firmware Version

2.5.8.6485f03

### Description

I use my own mqtt mosquitto server. The server works correctly with Heltec devices. With the RAK WisBlock 4631 with firmware 2.5.8.6485f03 and earlier, no correct JSON arrives at the mqtt server. There are unreadable characters in between.
Only the sent text and the channel name are correct. In the example, the message "OK" is sent on the "Regatta" channel.

### Relevant log output

```Shell
"
p_:ÿÿÿÿ"Ok5c =y× gHXdx Regatta	!3a12345
```

madeofstown, 2024-10-29:
Since when does the RAK4631 support WiFi networks?

caveman99, 2024-10-29:
the 4631 does only support json with a special firmware. Either on the Ethernet module or thorugh the App realy (not sure about that one, never used it)

closed 2024-10-29 by garthvh

Helbigmedia, 2024-10-29:
The RAK WisBlock 4631 is connected to the mobile phone. There you can set the mobile phone as a proxy client in the meshtastic app. The mqqt messages are then sent to the server via the mobile internet.

caveman99, 2024-10-29:
yes, what you are seeing is protobuf mode, not json. the rak does not have the capability to generate json topics.

Helbigmedia, 2024-10-29:
@caveman99 Many thanks for the info. It is a pity that RAK does not support Json. I quickly solved it like this. I am now converting the protbuf to Json in Nodered and then sending it to the MQTT server again.