meshtastic/firmware #5188: [Bug]: meshtasticd installation fails on Ubuntu 24.04

j75, 2024-10-29:
### Category

Other

### Hardware

Not Applicable

### Firmware Version

2.5.7.f77c87d

### Description

Trying to install [meshtasticd_2.5.7.f77c87d_amd64.deb](https://github.com/meshtastic/firmware/releases/download/v2.5.7.f77c87d/meshtasticd_2.5.7.f77c87d_amd64.deb) package on Ubuntu 24.04.1 fails withe the following message:
```
The following NEW packages will be installed:
  liborcania2.3{a} libulfius2.7t64{a} libyder2.0t64{a} meshtasticd{b} 
0 packages upgraded, 4 newly installed, 0 to remove and 0 not upgraded.
Need to get 45,1 MB of archives. After unpacking 71,7 MB will be used.
The following packages have unmet dependencies:
 meshtasticd : Depends: libyaml-cpp0.7 which is a virtual package and is not provided by any available package
```

On what OS the package should be installed? Are there any building instructions for Debian-based systems?

j75, 2024-10-29:
Maybe this issue is related to https://github.com/meshtastic/firmware/issues/4329 ?

caveman99, 2024-10-29:
the files are not compatible with ubuntu 24 these dependencies require ubuntu 22

caveman99, 2024-10-29:
you can always compile yourself, it#s really not difficult.

closed 2024-10-29 by caveman99

thebentern, 2024-10-30:
> 
> OK, but how? "Building Instructions" concern the firmware only, and in the **src/** folder there are cpp and h files, but no Makefile, no cmakefile,... :-(

https://meshtastic.org/docs/software/linux-native/

zhuzhe1983, 2024-11-05:
I also meet the same problem on my uconsole device based on rpi cm4.
I try to install the yaml-cpp0.7 from github source code and it dosen't work.

`zhuzhe@raspberrypi:~/Downloads/yaml-cpp/build $  lsb_release -a
No LSB modules are available.
Distributor ID: Debian
Description:    Debian GNU/Linux 11 (bullseye)
Release:        11
Codename:       bullseye
zhuzhe@raspberrypi:~/Downloads/yaml-cpp/build $ cat /proc/version
Linux version 5.10.17-v8+ (cpi@cpi-X79) (aarch64-linux-gnu-gcc (Ubuntu 11.3.0-1ubuntu1~22.04) 11.3.0, GNU ld (GNU Binutils for Ubuntu) 2.38) #5 SMP PREEMPT Thu Jun 29 13:36:01 CST 2023
zhuzhe@raspberrypi:~/Downloads/yaml-cpp/build $ hostnamectl
   Static hostname: raspberrypi
         Icon name: computer
        Machine ID: 3d22b274f2544131abcbd688d285d047
           Boot ID: 54eef9742d0e42b3897cdddebe28b1fb
  Operating System: Debian GNU/Linux 11 (bullseye)
            Kernel: Linux 5.10.17-v8+
      Architecture: arm64`

`zhuzhe@raspberrypi:~/Downloads/yaml-cpp/build $ sudo dpkg -i ~/Downloads/meshtasticd_2.5.9.936260f_arm64.deb
Selecting previously unselected package meshtasticd.
(Reading database ... 134329 files and directories currently installed.)
Preparing to unpack .../meshtasticd_2.5.9.936260f_arm64.deb ...
Unpacking meshtasticd (2.5.9.936260f) ...
dpkg: dependency problems prevent configuration of meshtasticd:
 meshtasticd depends on libyaml-cpp0.7; however:
  Package libyaml-cpp0.7 is not installed.

dpkg: error processing package meshtasticd (--install):
 dependency problems - leaving unconfigured
Errors were encountered while processing:
 meshtasticd`