meshtastic/firmware #5189: [Bug]: GPS Ublox M10 not working on NRF52

macvenez, 2024-10-29:
### Category

Other

### Hardware

DIY

### Firmware Version

2.5.8.6485f03

### Description

I'm using DIY NRF5280 ProMicro with TCXO.
The GPS Module is a HGLRC M100 5883 (I am sure it's working because from ucenter 2 I can connect and send/receive commands); this module is basically a M10050-KB with onboard antenna and compass (unused).
I tried all 2.5 Meshtastic versions from 2.5.7.f77c87d (I couldn't compile 2.4 on this board because it gives out errors).
Pins are correct since I can see messages being transferred to/from device by sniffing data with a FTDI Serial adapter (I also tried swapping them with no success).

I wonder why it tries all gps devices apart from Ublox ones?
DEBUG | ??:??:?? 85 [GPS] Trying $PDTINFO (UC6580)...
DEBUG | ??:??:?? 85 [GPS] Trying $PDTINFO (UM600)...
DEBUG | ??:??:?? 86 [GPS] Trying $PCAS06,1*1A (ATGM336H)...
DEBUG | ??:??:?? 86 [GPS] Trying $PCAS06,1*1A (ATGM332D)...
DEBUG | ??:??:?? 87 [GPS] Trying $PAIR021*39 (AG3335)...
DEBUG | ??:??:?? 87 [GPS] Trying $PAIR021*39 (AG3352)...
DEBUG | ??:??:?? 88 [GPS] Trying $PQTMVERNO*58 (LC86)...
DEBUG | ??:??:?? 88 [GPS] Trying $PCAS06,0*1B (L76K)...
DEBUG | ??:??:?? 89 [GPS] Trying $PMTK605*31 (L76B)...
I can see here it tries a lot of models but not the one I'm interested in...

### Relevant log output

```Shell
...
WARN  | ??:??:?? 75 [GPS] No response for class 06 message 08
WARN  | ??:??:?? 75 [GPS] Failed to find GNSS Module (baudrate 4800)
DEBUG | ??:??:?? 77 [GPS] Probing for GPS at 57600
DEBUG | ??:??:?? 77 [GPS] Trying $PDTINFO (UC6580)...
DEBUG | ??:??:?? 78 [GPS] Trying $PDTINFO (UM600)...
DEBUG | ??:??:?? 78 [GPS] Trying $PCAS06,1*1A (ATGM336H)...
DEBUG | ??:??:?? 79 [GPS] Trying $PCAS06,1*1A (ATGM332D)...
DEBUG | ??:??:?? 79 [GPS] Trying $PAIR021*39 (AG3335)...
DEBUG | ??:??:?? 80 [GPS] Trying $PAIR021*39 (AG3352)...
DEBUG | ??:??:?? 80 [GPS] Trying $PQTMVERNO*58 (LC86)...
DEBUG | ??:??:?? 81 [GPS] Trying $PCAS06,0*1B (L76K)...
DEBUG | ??:??:?? 81 [GPS] Trying $PMTK605*31 (L76B)...
DEBUG | ??:??:?? 83 [GPS]
WARN  | ??:??:?? 83 [GPS] No response for class 06 message 08
WARN  | ??:??:?? 83 [GPS] Failed to find GNSS Module (baudrate 57600)
DEBUG | ??:??:?? 85 [GPS] Probing for GPS at 9600
DEBUG | ??:??:?? 85 [GPS] Trying $PDTINFO (UC6580)...
DEBUG | ??:??:?? 85 [GPS] Trying $PDTINFO (UM600)...
DEBUG | ??:??:?? 86 [GPS] Trying $PCAS06,1*1A (ATGM336H)...
DEBUG | ??:??:?? 86 [GPS] Trying $PCAS06,1*1A (ATGM332D)...
DEBUG | ??:??:?? 87 [GPS] Trying $PAIR021*39 (AG3335)...
DEBUG | ??:??:?? 87 [GPS] Trying $PAIR021*39 (AG3352)...
DEBUG | ??:??:?? 88 [GPS] Trying $PQTMVERNO*58 (LC86)...
DEBUG | ??:??:?? 88 [GPS] Trying $PCAS06,0*1B (L76K)...
DEBUG | ??:??:?? 89 [GPS] Trying $PMTK605*31 (L76B)...
DEBUG | ??:??:?? 90 [GPS]
WARN  | ??:??:?? 90 [GPS] No response for class 06 message 08
WARN  | ??:??:?? 90 [GPS] Failed to find GNSS Module (baudrate 9600)
WARN  | ??:??:?? 90 [GPS] Giving up on GPS probe and setting to 9600.

The output would be too long, it just probes on all baudrate for each gps type and after some minutes it gives up.
```

fifieldt, 2024-10-29:
Hi, thanks for the report.

The ublox-relevant message is:

No response for class 06 message 08 .

As you're familiar with making firnware, could you place

#define GPS_DEBUG 1

In your variant.h and upload detailed logs?

macvenez, 2024-10-29:
That's the output with GPS_DEBUG already enabled, now I'm trying on an ESP32 to check whether it makes any difference

fifieldt, 2024-10-29:
with GPS_DEBUG enabled, the GPS will output anything it receives over the serial line ... since there's nothing at all in the logs, most likely means the wrong pins are being used somehow

macvenez, 2024-10-30:
I'll try that again later, maybe that's because I did `#define GPS_DEBUG` instead of `#define GPS_DEBUG 1`?
Thanks

fifieldt, 2024-10-30:
I had a quick check of the code and it looks like just defining GPS_DEBUG would work :(

macvenez, 2024-10-31:
I tried on an ESP32 running Meshtastic 2.5.8 and it works, there must be something wrong with pinouts or maybe the hardware serial on the NRF52.

That's the only output I got from the GPS on NRF52 (I couldn't get the whole log but that was probing on 9600):
```
INFO  | ??:??:?? 1 myRegion->freqStart -> myRegion->freqEnd: 869.400024 -> 869.650024 (0.250000 MHz)
INFO  | ??:??:?? 1 numChannels: 1 x 250.000kHz
INFO  | ??:??:?? 1 channel_num: 1
INFO  | ??:??:?? 1 frequency: 869.525024
INFO  | ??:??:?? 1 Slot time: 25 msec
INFO  | ??:??:?? 1 Set radio: final power level=20
INFO  | ??:??:?? 1 RF95 init result -2
INFO  | ??:??:?? 1 Frequency set to 869.525024
INFO  | ??:??:?? 1 Bandwidth set to 250.000000
INFO  | ??:??:?? 1 Power output set to 20
WARN  | ??:??:?? 1 Failed to find RF95 radio
DEBUG | ??:??:?? 1 SX126xInterface(cs=45, irq=10, rst=9, busy=29)
DEBUG | ??:??:?? 1 SX126X_DIO3_TCXO_VOLTAGE defined, using DIO3 as TCXO reference voltage at 1.800000 V
INFO  | ??:??:?? 1 Starting meshradio init...
INFO  | ??:??:?? 1 Radio freq=869.525, config.lora.frequency_offset=0.000
INFO  | ??:??:?? 1 Set radio: region=EU_868, name=MediumFast, config=4, ch=0, power=27
INFO  | ??:??:?? 1 myRegion->freqStart -> myRegion->freqEnd: 869.400024 -> 869.650024 (0.250000 MHz)
INFO  | ??:??:?? 1 numChannels: 1 x 250.000kHz
INFO  | ??:??:?? 1 channel_num: 1
INFO  | ??:??:?? 1 frequency: 869.525024
INFO  | ??:??:?? 1 Slot time: 25 msec
INFO  | ??:??:?? 1 Set radio: final power level=22
INFO  | ??:??:?? 1 SX126x init result 0
INFO  | ??:??:?? 1 Frequency set to 869.525024
INFO  | ??:??:?? 1 Bandwidth set to 250.000000
INFO  | ??:??:?? 1 Power output set to 22
DEBUG | ??:??:?? 1 Current limit set to 140.000000
DEBUG | ??:??:?? 1 Current limit set result 0
DEBUG | ??:??:?? 1 Set DIO2 as RF switch, result: 0
DEBUG | ??:??:?? 1 Using MCU pin 17 as RXEN and pin -1 as TXEN to control RF switching
INFO  | ??:??:?? 1 Set RX gain to boosted mode; result: 0
WARN  | ??:??:?? 1 SX1262 Radio init succeeded, TCXO, Vref 1.800000 V
DEBUG | ??:??:?? 1 LoRA bitrate = 395.000000 bytes / sec
INFO  | ??:??:?? 1 PowerFSM init, USB power=1
DEBUG | ??:??:?? 1 Enter state: BOOT
DEBUG | ??:??:?? 1 [Power] Battery: usbPower=0, isCharging=0, batMv=3100, batPct=0
DEBUG | ??:??:?? 1 [GPS] Probing for GPS at 9600
DEBUG | ??:??:?? 1 [GPS] Trying $PDTINFO (UC6580)...
DEBUG | ??:??:?? 2 [GPS] Trying $PDTINFO (UM600)...
DEBUG | ??:??:?? 2 [GPS] $GNRMC,,V,,,,,,,,,,N,V*37.
DEBUG | ??:??:?? 2 [GPS] Trying $PCAS06,1*1A (ATGM336H)...
DEBUG | ??:??:?? 2 [GPS] $GNTXT,01,01,01,PCAS inv format*23.
DEBUG | ??:??:?? 2 [GPS] $GNTXT,01,01,01,PCAS inv format*23..$GNRMC,,V,,,,,,,,,,N,V*37.
DEBUG | ??:??:?? 3 [GPS] Trying $PCAS06,1*1A (ATGM332D)...
DEBUG | ??:??:?? 3 [GPS] Trying $PAIR021*39 (AG3335)...
DEBUG | ??:??:?? 4 [GPS] Trying $PAIR021*39 (AG3352)...
DEBUG | ??:??:?? 4 [GPS] Trying $PQTMVERNO*58 (LC86)...
DEBUG | ??:??:?? 5 [GPS] Trying $PCAS06,0*1B (L76K)...
DEBUG | ??:??:?? 5 [GPS] Trying $PMTK605*31 (L76B)...
DEBUG | ??:??:?? 7 [GPS] 
WARN  | ??:??:?? 7 [GPS] No response for class 06 message 08
WARN  | ??:??:?? 7 [GPS] Failed to find GNSS Module (baudrate 9600)

```
The GPS seems to be answering something (I rebooted the device many times and I would say I'm getting some output like 1 time out of 10).
There must be something wrong on serial receive on NRF52 as on ESP32 it works flawlessly...
I'm using a Nrf52 promini (the ones from aliexpress) with a sx1262 if that may help...

fifieldt, 2024-11-01:
Left field thought - are you specifying pins in the app or in variant.h? Pin handling from the app is different between esp and nrf

macvenez, 2024-11-01:
I am specifying them from variant.h
Btw I am receiving something so wiring must be ok.
I'll try also from the app then, if it makes any difference.
What could be the issue with software? Could it be something related to bootloader or softdevice version?

macvenez, 2024-11-01:
I tried older bootloader with no result.

By tying together RX and TX pins on the NRF52 I can see what it spits out:
```
DEBUG | ??:??:?? 1 Enter state: BOOT
DEBUG | ??:??:?? 1 [Power] Battery: usbPower=0, isCharging=0, batMv=3100, batPct=0
DEBUG | ??:??:?? 1 [GPS] Probing for GPS at 9600
DEBUG | ??:??:?? 2 [GPS] Trying $PDTINFO (UC6580)...
DEBUG | ??:??:?? 2 [GPS] 0,0,0*5E.
DEBUG | ??:??:?? 2 [GPS] 0,0,0*5E..$PDTINFO.
DEBUG | ??:??:?? 2 [GPS] Trying $PDTINFO (UM600)...
DEBUG | ??:??:?? 2 [GPS] $PDTINFO.
DEBUG | ??:??:?? 3 [GPS] Trying $PCAS06,1*1A (ATGM336H)...
DEBUG | ??:??:?? 3 [GPS] $PCAS06,1*1A.
DEBUG | ??:??:?? 3 [GPS] Trying $PCAS06,1*1A (ATGM332D)...
DEBUG | ??:??:?? 3 [GPS] $PCAS06,1*1A.
DEBUG | ??:??:?? 4 [GPS] Trying $PAIR021*39 (AG3335)...
DEBUG | ??:??:?? 4 [GPS] PAIR513*3D.
DEBUG | ??:??:?? 4 [GPS] PAIR513*3D..$PAIR021*39.
DEBUG | ??:??:?? 4 [GPS] Trying $PAIR021*39 (AG3352)...
DEBUG | ??:??:?? 4 [GPS] $PAIR021*39.
DEBUG | ??:??:?? 5 [GPS] Trying $PQTMVERNO*58 (LC86)...
DEBUG | ??:??:?? 5 [GPS] $PQTMVERNO*58.
DEBUG | ??:??:?? 5 [GPS] Trying $PCAS06,0*1B (L76K)...
DEBUG | ??:??:?? 5 [GPS] $PCAS06,0*1B.
DEBUG | ??:??:?? 6 [GPS] Trying $PMTK605*31 (L76B)...
DEBUG | ??:??:?? 6 [GPS] ,0,0,0,0,0,0,0,0,0,0,0,0,0*2E.
DEBUG | ??:??:?? 6 [GPS] ,0,0,0,0,0,0,0,0,0,0,0,0,0*2E..$PMTK605*31.
DEBUG | ??:??:?? 7 [GPS] B562060800000E30
WARN  | ??:??:?? 7 [GPS] No response for class 06 message 08
WARN  | ??:??:?? 7 [GPS] Failed to find GNSS Module (baudrate 9600)
```

Is this a desirable output?

fifieldt, 2024-11-02:
Yeah, excellent. So looks like the GPS is hooked up correctly but not responding to the Ublox "are you a Ublox" command in the way we expect.

I think the next step is to decode B562060800000E30  and work out what that is according to the Ublox specification.

fifieldt, 2024-11-02:
Here's what a Ublox-10 on NRF52 looks like when it's working:

```
DEBUG | ??:??:?? 19 [GPS] Probing for GPS at 38400

DEBUG | ??:??:?? 20 [GPS] Trying $PDTINFO (UC6580)...

DEBUG | ??:??:?? 20 [GPS] $GNTXT,01,01,01,PDTI inv format*2B.

DEBUG | ??:??:?? 20 [GPS] $GNTXT,01,01,01,PDTI inv format*2B..$GNRMC,,V,,,,,,,,,,N,V*37.

DEBUG | ??:??:?? 20 [GPS] $GNTXT,01,01,01,PDTI inv format*2B..$GNRMC,,V,,,,,,,,,,N,V*37..$GNGGA,,,,,,0,00,99.99,,,,,,*56.

DEBUG | ??:??:?? 20 [GPS] $GNTXT,01,01,01,PDTI inv format*2B..$GNRMC,,V,,,,,,,,,,N,V*37..$GNGGA,,,,,,0,00,99.99,,,,,,*56..$GNGSA,A,1,,,,,,,,,,,,,99.99,99.99,99.99,1*33.

DEBUG | ??:??:?? 20 [GPS] $GNTXT,01,01,01,PDTI inv format*2B..$GNRMC,,V,,,,,,,,,,N,V*37..$GNGGA,,,,,,0,00,99.99,,,,,,*56..$GNGSA,A,1,,,,,,,,,,,,,99.99,99.99,99.99,1*33..$GNGSA,A,1,,,,,

DEBUG | ??:??:?? 20 [GPS] $GNTXT,01,01,01,PDTI inv format*2B..$GNRMC,,V,,,,,,,,,,N,V*37..$GNGGA,,,,,,0,00,99.99,,,,,,*56..$GNGSA,A,1,,,,,,,,,,,,,99.99,99.99,99.99,1*33..$GNGSA,A,1,,,,,

DEBUG | ??:??:?? 20 [GPS] $GNTXT,01,01,01,PDTI inv format*2B..$GNRMC,,V,,,,,,,,,,N,V*37..$GNGGA,,,,,,0,00,99.99,,,,,,*56..$GNGSA,A,1,,,,,,,,,,,,,99.99,99.99,99.99,1*33..$GNGSA,A,1,,,,,

DEBUG | ??:??:?? 20 [GPS] Trying $PDTINFO (UM600)...

DEBUG | ??:??:?? 20 [GPS] $GNTXT,01,01,01,PDTI inv format*2B.

DEBUG | ??:??:?? 21 [GPS] Trying $PCAS06,1*1A (ATGM336H)...

DEBUG | ??:??:?? 21 [GPS] $GNTXT,01,01,01,PCAS inv format*23.

DEBUG | ??:??:?? 21 [GPS] $GNTXT,01,01,01,PCAS inv format*23..$GNRMC,,V,,,,,,,,,,N,V*37.

DEBUG | ??:??:?? 21 [GPS] $GNTXT,01,01,01,PCAS inv format*23..$GNRMC,,V,,,,,,,,,,N,V*37..$GNGGA,,,,,,0,00,99.99,,,,,,*56.

DEBUG | ??:??:?? 21 [GPS] $GNTXT,01,01,01,PCAS inv format*23..$GNRMC,,V,,,,,,,,,,N,V*37..$GNGGA,,,,,,0,00,99.99,,,,,,*56..$GNGSA,A,1,,,,,,,,,,,,,99.99,99.99,99.99,1*33.

DEBUG | ??:??:?? 21 [GPS] $GNTXT,01,01,01,PCAS inv format*23..$GNRMC,,V,,,,,,,,,,N,V*37..$GNGGA,,,,,,0,00,99.99,,,,,,*56..$GNGSA,A,1,,,,,,,,,,,,,99.99,99.99,99.99,1*33..$GNGSA,A,1,,,,,

DEBUG | ??:??:?? 21 [GPS] $GNTXT,01,01,01,PCAS inv format*23..$GNRMC,,V,,,,,,,,,,N,V*37..$GNGGA,,,,,,0,00,99.99,,,,,,*56..$GNGSA,A,1,,,,,,,,,,,,,99.99,99.99,99.99,1*33..$GNGSA,A,1,,,,,

DEBUG | ??:??:?? 21 [GPS] $GNTXT,01,01,01,PCAS inv format*23..$GNRMC,,V,,,,,,,,,,N,V*37..$GNGGA,,,,,,0,00,99.99,,,,,,*56..$GNGSA,A,1,,,,,,,,,,,,,99.99,99.99,99.99,1*33..$GNGSA,A,1,,,,,

DEBUG | ??:??:?? 21 [GPS] Trying $PCAS06,1*1A (ATGM332D)...

DEBUG | ??:??:?? 21 [GPS] $GNTXT,01,01,01,PCAS inv format*23.

DEBUG | ??:??:?? 22 [GPS] Trying $PAIR021*39 (AG3335)...

DEBUG | ??:??:?? 22 [GPS] 1,01,PAIR inv format*28.

DEBUG | ??:??:?? 22 [GPS] 1,01,PAIR inv format*28..$GNTXT,01,01,01,PAIR inv format*28.

DEBUG | ??:??:?? 22 [GPS] 1,01,PAIR inv format*28..$GNTXT,01,01,01,PAIR inv format*28..$GNTXT,01,01,01,PAIR inv format*28.

DEBUG | ??:??:?? 22 [GPS] 1,01,PAIR inv format*28..$GNTXT,01,01,01,PAIR inv format*28..$GNTXT,01,01,01,PAIR inv format*28..$GNTXT,01,01,01,PAIR inv format*28.

DEBUG | ??:??:?? 22 [GPS] 1,01,PAIR inv format*28..$GNTXT,01,01,01,PAIR inv format*28..$GNTXT,01,01,01,PAIR inv format*28..$GNTXT,01,01,01,PAIR inv format*28..$GNRMC,,V,,,,,,,,,,N,V*37

DEBUG | ??:??:?? 22 [GPS] 1,01,PAIR inv format*28..$GNTXT,01,01,01,PAIR inv format*28..$GNTXT,01,01,01,PAIR inv format*28..$GNTXT,01,01,01,PAIR inv format*28..$GNRMC,,V,,,,,,,,,,N,V*37

DEBUG | ??:??:?? 22 [GPS] 1,01,PAIR inv format*28..$GNTXT,01,01,01,PAIR inv format*28..$GNTXT,01,01,01,PAIR inv format*28..$GNTXT,01,01,01,PAIR inv format*28..$GNRMC,,V,,,,,,,,,,N,V*37

DEBUG | ??:??:?? 22 [GPS] 1,01,PAIR inv format*28..$GNTXT,01,01,01,PAIR inv format*28..$GNTXT,01,01,01,PAIR inv format*28..$GNTXT,01,01,01,PAIR inv format*28..$GNRMC,,V,,,,,,,,,,N,V*37

DEBUG | ??:??:?? 22 [GPS] 1,01,PAIR inv format*28..$GNTXT,01,01,01,PAIR inv format*28..$GNTXT,01,01,01,PAIR inv format*28..$GNTXT,01,01,01,PAIR inv format*28..$GNRMC,,V,,,,,,,,,,N,V*37

DEBUG | ??:??:?? 22 [GPS] 1,01,PAIR inv format*28..$GNTXT,01,01,01,PAIR inv format*28..$GNTXT,01,01,01,PAIR inv format*28..$GNTXT,01,01,01,PAIR inv format*28..$GNRMC,,V,,,,,,,,,,N,V*37

DEBUG | ??:??:?? 22 [GPS] Trying $PAIR021*39 (AG3352)...

DEBUG | ??:??:?? 22 [GPS] $GNTXT,01,01,01,PAIR inv format*28.
[... 46 lines cut ...]

INFO | ??:??:?? 25 [GPS] Got ACK for class 0A message 04 in 67 millis.

DEBUG | ??:??:?? 25 [GPS] Module Info :

DEBUG | ??:??:?? 25 [GPS] Soft version: ROM SPG 5.10 (7b202e)

DEBUG | ??:??:?? 25 [GPS] Hard version: 000A0000

DEBUG | ??:??:?? 25 [GPS] Extensions:4

DEBUG | ??:??:?? 25 [GPS]  FWVER=SPG 5.10

DEBUG | ??:??:?? 25 [GPS]  PROTVER=34.10

DEBUG | ??:??:?? 25 [GPS]  GPS;GLO;GAL;BDS

DEBUG | ??:??:?? 25 [GPS]  SBAS;QZSS

DEBUG | ??:??:?? 25 [GPS] Protocol Version:34.10

DEBUG | ??:??:?? 25 [GPS] ProtVer=34

INFO | ??:??:?? 25 [GPS] U-blox 10 detected, using 10 Module
```

macvenez, 2024-11-02:
> Yeah, excellent. So looks like the GPS is hooked up correctly but not responding to the Ublox "are you a Ublox" command in the way we expect.
> 
> I think the next step is to decode B562060800000E30 and work out what that is according to the Ublox specification.

Wait, that was the messages when RX and TX on NRF52 side were tier together so it was basically echoing itself (I was trying to understand whether it was working).
The real issue is the NRF52 doesn't get all messages the GPS module sends (at this point this must be something clock related) because the same code with same version works on an ESP32 configured the same way.
The GPS module is wired correctly as sometimes some messages are received by NRF52 and sometimes not like here:
```
DEBUG | ??:??:?? 1 [GPS] Probing for GPS at 9600
DEBUG | ??:??:?? 1 [GPS] Trying $PDTINFO (UC6580)...
DEBUG | ??:??:?? 2 [GPS] Trying $PDTINFO (UM600)...
DEBUG | ??:??:?? 2 [GPS] $GNRMC,,V,,,,,,,,,,N,V*37.
DEBUG | ??:??:?? 2 [GPS] Trying $PCAS06,1*1A (ATGM336H)... 
DEBUG | ??:??:?? 2 [GPS] $GNTXT,01,01,01,PCAS inv format*23.
DEBUG | ??:??:?? 2 [GPS] $GNTXT,01,01,01,PCAS inv format*23..$GNRMC,,V,,,,,,,,,,N,V*37.
DEBUG | ??:??:?? 3 [GPS] Trying $PCAS06,1*1A (ATGM332D)... 
DEBUG | ??:??:?? 3 [GPS] Trying $PAIR021*39 (AG3335)...
DEBUG | ??:??:?? 4 [GPS] Trying $PAIR021*39 (AG3352)...
DEBUG | ??:??:?? 4 [GPS] Trying $PQTMVERNO*58 (LC86)...
DEBUG | ??:??:?? 5 [GPS] Trying $PCAS06,0*1B (L76K)...
DEBUG | ??:??:?? 5 [GPS] Trying $PMTK605*31 (L76B)...
```
As we can see here just  2 messages are sent by the GPS module and the GNRMC is the default polling for NMEA data

fifieldt, 2024-11-02:
Can we see your variant.h to look for clues?

macvenez, 2024-11-02:
Sure:
```
#ifndef _VARIANT_PROMICRO_DIY_
#define _VARIANT_PROMICRO_DIY_

/** Master clock frequency */
#define VARIANT_MCK (64000000ul)

// #define USE_LFXO // Board uses 32khz crystal for LF
#define USE_LFRC // Board uses RC for LF

#define PROMICRO_DIY_TCXO

/*----------------------------------------------------------------------------
 *        Headers
 *----------------------------------------------------------------------------*/

#include "WVariant.h"

#ifdef __cplusplus
extern "C" {
#endif // __cplusplus

/*
NRF52 PRO MICRO PIN ASSIGNMENT

| Pin   | Function    |     | Pin      | Function     | RF95  |
| ----- | ----------- | --- | -------- | ------------ | ----- |
| Gnd   |             |     | vbat     |              |       |
| P0.06 | Serial2 RX  |     | vbat     |              |       |
| P0.08 | Serial2 TX  |     | Gnd      |              |       |
| Gnd   |             |     | reset    |              |       |
| Gnd   |             |     | ext_vcc  | *see 0.13    |       |
| P0.17 | RXEN        |     | P0.31    | BATTERY_PIN  |       |
| P0.20 | GPS_RX      |     | P0.29    | BUSY         | DIO0  |
| P0.22 | GPS_TX      |     | P0.02    | MISO         | MISO  |
| P0.24 | GPS_EN      |     | P1.15    | MOSI         | MOSI  |
| P1.00 | BUTTON_PIN  |     | P1.13    | CS           | CS    |
| P0.11 | SCL         |     | P1.11    | SCK          | SCK   |
| P1.04 | SDA         |     | P0.10    | DIO1/IRQ     | DIO1  |
| P1.06 | Free pin    |     | P0.09    | RESET        | RST   |
|       |             |     |          |              |       |
|       | Mid board   |     |          | Internal     |       |
| P1.01 | Free pin    |     | 0.15     | LED          |       |
| P1.02 | Free pin    |     | 0.13     | 3V3_EN       |       |
| P1.07 | Free pin    |     |          |              |       |
*/

// Number of pins defined in PinDescription array
#define PINS_COUNT (48)
#define NUM_DIGITAL_PINS (48)
#define NUM_ANALOG_INPUTS (1)
#define NUM_ANALOG_OUTPUTS (0)

// Pin 13 enables 3.3V periphery. If the Lora module is on this pin, then it should stay enabled at all times.
#define PIN_3V3_EN (0 + 13) // P0.13

// Analog pins
#define BATTERY_PIN (0 + 31) // P0.31 Battery ADC
#define ADC_CHANNEL ADC1_GPIO4_CHANNEL
#define ADC_RESOLUTION 14
#define BATTERY_SENSE_RESOLUTION_BITS 12
#define BATTERY_SENSE_RESOLUTION 4096.0
// Definition of milliVolt per LSB => 3.0V ADC range and 12-bit ADC resolution = 3000mV/4096
#define VBAT_MV_PER_LSB (0.73242188F)
// Voltage divider value => 1.5M + 1M voltage divider on VBAT = (1.5M / (1M + 1.5M))
#define VBAT_DIVIDER (0.6F)
// Compensation factor for the VBAT divider
#define VBAT_DIVIDER_COMP (1.73)
// Fixed calculation of milliVolt from compensation value
[... 91 lines cut ...]
| Heltec       | HT-RA62          | Yes  | Int       |                                       |
| NiceRF       | Lora1262         | yes  | Int       |                                       |
| Waveshare    | Core1262-HF      | yes  | Ext       |                                       |
| Waveshare    | LoRa Node Module | yes  | Int       |                                       |
| Seeed        | Wio-SX1262       | yes  | Int       | Sooooo cute!                          |
| AI-Thinker   | RA-02            | No   | Int       | SX1278 **433mhz band only**           |
| RF Solutions | RFM95            | No   | Int       | Untested                              |

*/

#define SX126X_DIO3_TCXO_VOLTAGE 1.8
#define TCXO_OPTIONAL     // make it so that the firmware can try both TCXO and XTAL
extern float tcxoVoltage; // make this available everywhere

#ifdef __cplusplus
}
#endif

/*----------------------------------------------------------------------------
 *        Arduino objects - C++ only
 *----------------------------------------------------------------------------*/

#endif
```
Which can be found [here](https://github.com/meshtastic/firmware/blob/master/variants/diy/nrf52_promicro_diy_tcxo/variant.h) as DIY.

macvenez, 2024-11-04:
Hi, any updates on this?

fifieldt, 2024-11-05:
Can you post the rest of the log for the time when you see the stuff eg "$GNTXT,01,01,01,PCAS inv format*23.", coming back from the GPS? It cuts off right before the Ublox testing.

fifieldt, 2024-11-05:
Also, some improvement to the probe code was merged recently. I don't think it'd make a difference for this, but if it's easy for you to work from master, might be worth a shot.

fifieldt, 2024-11-05:
Also, there are a lot of experts in the #gnss channel on meshtastic discord. Probably worth asking in there in parallel.

macvenez, 2024-11-05:
Nevermind, good news! It's working now compiling master (2.5.11.8e2a3e5) as you suggested.

That's my output:
```
DEBUG | ??:??:?? 5 State: BOOT
DEBUG | ??:??:?? 5 [Power] Battery: usbPower=0, isCharging=0, batMv=3940, batPct=75
DEBUG | ??:??:?? 5 [GPS] Probe for GPS at 9600
DEBUG | ??:??:?? 5 [GPS] Trying $PDTINFO (UC6580)...
DEBUG | ??:??:?? 5 [GPS] ,,,,,,,,,,N,V*37.
DEBUG | ??:??:?? 5 [GPS] ,,,,,,,,,,N,V*37..$GNGGA,,,,,,0,00,99.99,,,,,,*56.
DEBUG | ??:??:?? 5 [GPS] ,,,,,,,,,,N,V*37..$GNGGA,,,,,,0,00,99.99,,,,,,*56..$GNTXT,01,01,01,PDTI inv format*2B.
DEBUG | ??:??:?? 6 [GPS] Trying $PDTINFO (UM600)...
DEBUG | ??:??:?? 6 [GPS] 7.
DEBUG | ??:??:?? 6 [GPS] 7..$GNGGA,,,,,,0,00,99.99,,,,,,*56.
DEBUG | ??:??:?? 6 [GPS] 7..$GNGGA,,,,,,0,00,99.99,,,,,,*56..$GNTXT,01,01,01,PDTI inv format*2B.
DEBUG | ??:??:?? 6 [GPS] Trying $PCAS06,1*1A (ATGM336H)...
DEBUG | ??:??:?? 6 [GPS] .
DEBUG | ??:??:?? 6 [GPS] ..$GNGGA,,,,,,0,00,99.99,,,,,,*56.
DEBUG | ??:??:?? 6 [GPS] ..$GNGGA,,,,,,0,00,99.99,,,,,,*56..$GNTXT,01,01,01,PCAS inv format*23.
DEBUG | ??:??:?? 7 [GPS] ..$GNGGA,,,,,,0,00,99.99,,,,,,*56..$GNTXT,01,01,01,PCAS inv format*23..$GNRMC,,V,,,,,,,,,,N,V*37.
DEBUG | ??:??:?? 7 [GPS] Trying $PCAS06,1*1A (ATGM332D)...
DEBUG | ??:??:?? 7 [GPS] NGGA,,,,,,0,00,99.99,,,,,,*56.
DEBUG | ??:??:?? 7 [GPS] NGGA,,,,,,0,00,99.99,,,,,,*56..$GNTXT,01,01,01,PCAS inv format*23.
DEBUG | ??:??:?? 7 [GPS] NGGA,,,,,,0,00,99.99,,,,,,*56..$GNTXT,01,01,01,PCAS inv format*23..$GNRMC,,V,,,,,,,,,,N,V*37.
DEBUG | ??:??:?? 7 [GPS] Trying $PAIR021*39 (AG3335)...
DEBUG | ??:??:?? 7 [GPS] T,01,01,01,PAIR inv format*28.
DEBUG | ??:??:?? 7 [GPS] T,01,01,01,PAIR inv format*28..$GNTXT,01,01,01,PAIR inv format*28.
DEBUG | ??:??:?? 7 [GPS] T,01,01,01,PAIR inv format*28..$GNTXT,01,01,01,PAIR inv format*28..$GNTXT,01,01,01,PAIR inv format*28.
DEBUG | ??:??:?? 7 [GPS] T,01,01,01,PAIR inv format*28..$GNTXT,01,01,01,PAIR inv format*28..$GNTXT,01,01,01,PAIR inv format*28..$GNTXT,01,01,01,PAIR inv format*28.
DEBUG | ??:??:?? 8 [GPS] T,01,01,01,PAIR inv format*28..$GNTXT,01,01,01,PAIR inv format*28..$GNTXT,01,01,01,PAIR inv format*28..$GNTXT,01,01,01,PAIR inv format*28..$GNRMC,,V,,,,,,,,,,
DEBUG | ??:??:?? 8 [GPS] T,01,01,01,PAIR inv format*28..$GNTXT,01,01,01,PAIR inv format*28..$GNTXT,01,01,01,PAIR inv format*28..$GNTXT,01,01,01,PAIR inv format*28..$GNRMC,,V,,,,,,,,,,
DEBUG | ??:??:?? 8 [GPS] Trying $PAIR021*39 (AG3352)...
DEBUG | ??:??:?? 8 [GPS] $GNTXT,01,01,01,PAIR inv format*28.
DEBUG | ??:??:?? 8 [GPS] $GNTXT,01,01,01,PAIR inv format*28..$GNRMC,,V,,,,,,,,,,N,V*37.
DEBUG | ??:??:?? 8 [GPS] $GNTXT,01,01,01,PAIR inv format*28..$GNRMC,,V,,,,,,,,,,N,V*37..$GNGGA,,,,,,0,00,99.99,,,,,,*56.
DEBUG | ??:??:?? 8 [GPS] Trying $PQTMVERNO*58 (LC86)...
DEBUG | ??:??:?? 8 [GPS] $GNTXT,01,01,01,PQTM inv format*3A.
DEBUG | ??:??:?? 9 [GPS] $GNTXT,01,01,01,PQTM inv format*3A..$GNRMC,,V,,,,,,,,,,N,V*37.
DEBUG | ??:??:?? 9 [GPS] $GNTXT,01,01,01,PQTM inv format*3A..$GNRMC,,V,,,,,,,,,,N,V*37..$GNGGA,,,,,,0,00,99.99,,,,,,*56.
DEBUG | ??:??:?? 9 [GPS] Trying $PCAS06,0*1B (L76K)...
DEBUG | ??:??:?? 9 [GPS] $GNTXT,01,01,01,PCAS inv format*23.
DEBUG | ??:??:?? 9 [GPS] $GNTXT,01,01,01,PCAS inv format*23..$GNRMC,,V,,,,,,,,,,N,V*37.
DEBUG | ??:??:?? 9 [GPS] $GNTXT,01,01,01,PCAS inv format*23..$GNRMC,,V,,,,,,,,,,N,V*37..$GNGGA,,,,,,0,00,99.99,,,,,,*56.
DEBUG | ??:??:?? 9 [GPS] Trying $PMTK605*31 (L76B)...
DEBUG | ??:??:?? 9 [GPS] $GNTXT,01,01,01,PMTK inv format*20.
DEBUG | ??:??:?? 9 [GPS] $GNTXT,01,01,01,PMTK inv format*20..$GNTXT,01,01,01,PMTK inv format*20.
DEBUG | ??:??:?? 10 [GPS] $GNTXT,01,01,01,PMTK inv format*20..$GNTXT,01,01,01,PMTK inv format*20..$GNRMC,,V,,,,,,,,,,N,V*37.
DEBUG | ??:??:?? 10 [GPS] $GNTXT,01,01,01,PMTK inv format*20..$GNTXT,01,01,01,PMTK inv format*20..$GNRMC,,V,,,,,,,,,,N,V*37..$GNGGA,,,,,,0,00,99.99,,,,,,*56.
DEBUG | ??:??:?? 10 [GPS] Trying $PMTK605*31 (PA1616S)...
DEBUG | ??:??:?? 10 [GPS] $GNTXT,01,01,01,PMTK inv format*20.
DEBUG | ??:??:?? 10 [GPS] $GNTXT,01,01,01,PMTK inv format*20..$GNRMC,,V,,,,,,,,,,N,V*37.
DEBUG | ??:??:?? 10 [GPS] $GNTXT,01,01,01,PMTK inv format*20..$GNRMC,,V,,,,,,,,,,N,V*37..$GNGGA,,,,,,0,00,99.99,,,,,,*56.
INFO  | ??:??:?? 10 [GPS] Got ACK for class 06 message 08 in 55ms
INFO  | ??:??:?? 11 [GPS] Got ACK for class 0A message 04 in 199ms
DEBUG | ??:??:?? 11 [GPS] Module Info : 
DEBUG | ??:??:?? 11 [GPS] Soft version: ROM SPG 5.10 (7b202e)
DEBUG | ??:??:?? 11 [GPS] Hard version: 000A0000
DEBUG | ??:??:?? 11 [GPS] Extensions:4
DEBUG | ??:??:?? 11 [GPS]   FWVER=SPG 5.10
DEBUG | ??:??:?? 11 [GPS]   PROTVER=34.10
DEBUG | ??:??:?? 11 [GPS]   GPS;GLO;GAL;BDS
DEBUG | ??:??:?? 11 [GPS]   SBAS;QZSS
DEBUG | ??:??:?? 11 [GPS] Protocol Version:34.10
DEBUG | ??:??:?? 11 [GPS] ProtVer=34
INFO  | ??:??:?? 11 [GPS] U-blox 10 detected, using 10 Module
INFO  | ??:??:?? 12 [GPS] Got ACK for class 06 message 8A in 64ms
INFO  | ??:??:?? 13 [GPS] Got ACK for class 06 message 8A in 149ms
INFO  | ??:??:?? 13 [GPS] Got ACK for class 06 message 8A in 37ms
INFO  | ??:??:?? 14 [GPS] Got ACK for class 06 message 8A in 121ms
INFO  | ??:??:?? 15 [GPS] Got ACK for class 06 message 8A in 64ms
INFO  | ??:??:?? 16 [GPS] Got ACK for class 06 message 8A in 148ms
INFO  | ??:??:?? 17 [GPS] Got ACK for class 06 message 8A in 37ms
INFO  | ??:??:?? 18 [GPS] Got ACK for class 06 message 8A in 122ms
INFO  | ??:??:?? 18 [GPS] Got ACK for class 06 message 8A in 37ms
INFO  | ??:??:?? 19 [GPS] Got ACK for class 06 message 8A in 113ms
INFO  | ??:??:?? 20 [GPS] Got ACK for class 06 message 8A in 125ms
INFO  | ??:??:?? 21 [GPS] Got ACK for class 06 message 8A in 38ms
INFO  | ??:??:?? 22 [GPS] Got ACK for class 06 message 09 in 63ms
INFO  | ??:??:?? 22 [GPS] GNSS module configuration saved!

```
I'll have a look at the latest commits to see what's changed.
Thank you a lot for the help.
Can I leave the issue open to report in case it's a intermittent issue?
Still thanks

fifieldt, 2024-11-05:
Fantastic news, glad to hear it.

thebentern, 2024-11-06:
Re-opening if it comes back up is easier. In my experience, rarely do PRs get closed outside of an intentional cleanup. 😅

closed 2024-11-06 by thebentern