meshtastic/firmware #5193: Optimise GPS probe code - one message per family

fifieldt, 2024-10-31:
In the GPS probe code (
https://github.com/meshtastic/firmware/blob/aae346aef74b72d5914a470d7f19f18bce2f0ec3/src/gps/GPS.cpp#L1137 ), we write commands on the serial line and determine which GPS we have based on the result.

GPS units in the same family sometimes use the same command, but return different results. Currently we run the command once per GPS. Instead we should run each command only once per family, record the result, and select the GNSS MODEL based on the result.

This will mean probing takes a shorter time.

absolut3ego, 2024-11-03:
can you assign this to me?

fifieldt, 2024-11-03:
Welcome to Meshtastic, we're happy to have you!

One way I thought about implementing this was to change the PROBE_SIMPLE macro.

At the end, save the command we sent as a variable (lastcmd = TOWRITE). Then, at the start, check if TOWRITE == lastcmd and perform the check there. Would require changing GPS::getACK so that it can handle returning the buffer or accepting a pointer to a buffer.

Let us know if you need any pointers!

absolut3ego, 2024-11-03:
Hi @fifieldt,

I have a few questions. First, can you tell me how the PROBE_SIMPLE macro currently works? To make sure that my modification can work correctly with your project. Second, how do you want me to solve this issue? As I initially understood, I need to add something to store the results of each GPS family, so that I can call a family many times without having to re-execute the PROBE_SIMPLE macro. But according to your suggestion, I understand it in a different way, which is to save the last command. If the next command matches it, I will get the result that was just called to return. Can you explain this issue more clearly? I need to understand clearly to be able to have the correct solution.

closed 2025-03-04 by thebentern