meshtastic/firmware #5194: De-duplicate Ambient LED management code

fifieldt, 2024-10-31:
We currently have 3 different places where we have the logic for modifying LED state of the various types of Ambient LEDs,
ExternalNotificationModule::SetExternalState
AmbientLightingThread::setLighting
AmbientLightingThread::setLightingOff

Instead, we should have a single place where this happens.

Use the notification system we have in the firmware for IPC, so that Ambient lighting offers a subscription service for LED changes and the External Notifications uses it (if available).

You can look at how it's done for notifying others about GPS position changes :-)

Blake-Latchford, 2024-11-13:
```
You can look at how it's done for notifying others about GPS position changes :-)
```
https://github.com/meshtastic/firmware/blob/e866734a25fe811bdb87253da8c8e0879be512f9/src/gps/GPS.cpp#L975

Is this what you're referring to? If I understand the proposal correctly:
1. AmbientLightingThread should own something like an `Observable<LightingConfig> lightChanged`.  
2. AmbientLightingThread::setLighting is invoked by an observer.
3. LightingConfig probably looks a lot like meshtastic_ModuleConfig_AmbientLightingConfig, but also needs to identify which light is being configured. Though maybe you're expecting that the observer is just watching for ambient lighting.
4. AmbientLightingThread::setLightingOff calls `observable.notifyObservers(&offStruct)`

fifieldt, 2025-02-21:
Hi @Blake-Latchford - I'm sorry for missing your message. A fellow contributor pointed it out to me. As I understand it, yes, basically. ExternalNotificationModule would use the Observable.

SQ6RDP, 2025-03-04:
I would like to ask for the possibility of using a frequency outside the wifi band, in the 2340MHz part of the radio amateur radio for licenced users. I dont know where address this idea, this is the first topic about 2400MHz