meshtastic/firmware #5205: [Feature Request]: Refactor getMacAddr function

alexbegoon, 2024-10-31:
### Platform

ESP32

### Description

Hi,

I would like to propose a refactor of the `getMacAddr` function to retrieve the MAC address in MAC-48 format for compatibility with IEEE 802.15.4.

Specifically, I suggest changing [this function](https://github.com/meshtastic/firmware/blob/master/src/platform/esp32/main-esp32.cpp#L52-L55) from:
```
void getMacAddr(uint8_t *dmac)
{
    assert(esp_efuse_mac_get_default(dmac) == ESP_OK);
}
```
to 
```
void getMacAddr(uint8_t *dmac)
{
    assert(esp_base_mac_addr_get(dmac) == ESP_OK);
}
```

This change will help eliminate node ID conflicts by ensuring that the MAC address used is always in the MAC-48 format, which is required for devices utilizing IEEE 802.15.4.

### Example
I have two ESP32-C6 targets, and their MAC addresses lead to identical NodeIDs:

1st device:

MAC: 11:22:33:44:55:66:77:88
BASE MAC: 11:22:33:66:77:88
MAC_EXT: 44:55
2nd device:

MAC: 11:22:33:44:55:66:77:89
BASE MAC: 11:22:33:66:77:89
MAC_EXT: 44:55
As a result, both devices have the same ID: !33445566.

GUVWAF, 2024-10-31:
We can't change this for all ESP32 boards, as they would get new node numbers after a factory reset and thus would look like two nodes in the database. It would probably be fine to do this for ESP32-C6, but are those really your MAC addresses? They look to nice to be true, to be honest.

alexbegoon, 2024-10-31:
I can't provide the real MAC addresses from these two devices, but this illustrates the real-life problem with these targets. I currently have at least five sitting on my desk, and they all have the same node ID.

From [the esp-idf docs](https://docs.espressif.com/projects/esp-idf/en/stable/esp32c6/api-reference/system/misc_system_api.html?highlight=esp_efuse_mac_get_default#_CPPv425esp_efuse_mac_get_defaultP7uint8_t)
```
esp_efuse_mac_get_default(uint8_t *mac)

Return base MAC address which is factory-programmed by Espressif in EFUSE.

Parameters
mac -- base MAC address, length: 6 bytes/8 bytes. length: 6 bytes for MAC-48 8 bytes for EUI-64(used for IEEE 802.15.4, if CONFIG_SOC_IEEE802154_SUPPORTED=y)
```

Some manufacturers are placing `MAC_EXT` in the middle of the `BASE MAC`, and that’s the issue.

GUVWAF, 2024-10-31:
Okay, fair. I think it would be OK to change it like this for ESP32-C6 specifically.

GUVWAF, 2024-11-01:
Closed by #5208.

closed 2024-11-01 by GUVWAF