meshtastic/firmware #5210: [Bug]: firmware-tlora-v1 no longer build?

edegraaff, 2024-10-31:
### Category

Hardware Compatibility

### Hardware

T-Lora v1

### Firmware Version

firmware-esp32-2.5.8.6485f03.zip

### Description

Maybe i am wrong but in my notes from march this year is written use the firmware-tlora-v1-2.4.2.5b45303.bin firmware otherwise the screen is not working anymore. I still use 2 old the https://meshtastic.org/docs/hardware/devices/lilygo/lora/?t-lora=v1

In the notes in the download page i see this:
This board is still in production but for various reasons not recommended for new purchases or for unattended installations. Firmware support is phased out. If in doubt, choose the Lora V2.1-1.6 or Lora T3S3 board.

Question:
I know the T3S3 firmware did not work with the display in march 2024. What would be the appropiate approach? Could i safely upgrade to firmware-tlora-v2-1-1_6-2.5.8.6485f03.bin

Thanks

garthvh, 2024-11-03:
The v1 tlora is no longer suppor

closed 2024-11-03 by garthvh

edegraaff, 2024-11-03:
I  was affraid of this answer, but clear, and saved me a lot of time. I will go for the newer hardware. 
Thanks for the great project, 
Regards
Eelco

C4, 2024-11-17:
Is there any chance someone else is maintaining a fork for the display to work on these? I have a bunch of these that I just found and would like to get them working on the mesh. Is it just a pin mapping for the display that's needed?

edegraaff, 2024-11-17:
I have tested in the past the other builds and they do work, but indeed, the display won't work, the why i have no clue. In fact, i understand the development goes on, on the another hand, the tlora-v1 is ok, it does the job, especcially on locations where they are used as a router. I havend any facts and figures to drop the v1 and schred it. They stay usable to show people the possibillitys of meshtastic. Even for that i want to spread the wordt by giving demos or loan people my V1 to experiment.

garthvh, 2024-11-17:
It has an old radio and esp32 chip that has been difficult to keep working, a fork is not really necessary, but it has been broken for quite a while which is why it is no longer officially supported, I don't know that the pins on the screen changed, something in the esp32 toolchain seems to have caused issues.

edegraaff, 2024-11-17:
Thanks for your response, i will step over to Heltec the upcomming weeks. I am quite curious what new features it will bring me :-)

C4, 2024-11-18:
I did indeed get this working with the Lora V2.1-1.6 and seems to be working well with 4 or 5 devices I have. If I were to try to figure out the pinout for the screen again would you consider adding it back or at least some config file? I realize this is deprecated but at least getting some way for folks to still use these would be very much appreciated.

C4, 2024-11-18:
Just to follow up here incase anyone else is trying to get one of these older devices working. I had to compile the firmware from source following [this guide](https://meshtastic.org/docs/development/firmware/build/). I just selected tlora-v1 environment and built and uploaded. Everything worked from there and now I have 2.5.14.de76caca running with display on my device.

edegraaff, 2024-11-18:
@C4  great you have been able to compile. In that case, why could not het model be added at the nightly build list? If the code is broke the team could always deside to change support of this model again into " no longer supported"

C4, 2024-11-20:
Not sure. Sounds like they don't want to support it officially any longer. If you want my firmware build let me know and I can post it.

edegraaff, 2024-11-20:
Would be nice, then i can flash the old devices and share them with some newbee's

C4, 2024-11-20:
https://github.com/C4/Meshtastic-TLoraV1-Build
You can flash using the web flasher.

RobertSasak, 2026-03-08:
@sysmanalex Thank your for a great news. Does the display works for you on Lora v1?

I forgot to mention that I used web flasher which do not offer V1 among it's options. I tried again now and web flash can detect the V1 board as V2-1-1_6. After flashing latest stable versjon 2.7.15.567b8ea the device works but display is black.

RobertSasak, 2026-03-22:
I had to add LED_BUILDIN into platformio.ini to make it build and made a PR https://github.com/meshtastic/firmware/pull/9973

I can confirm that display works!