meshtastic/firmware #5217: Feature Request: Nodeinfo depending on movement

cracky22, 2024-11-01:
Send node info or information about the current device to all devices that have a motion or acceleration sensor automatically more often when it is moving (For example, for the T1000-E). Can be activated optionally via app

cracky22, 2024-11-01:
Sorry that it appears twice now... it's probably a bug in the Android app from GitHub...

b8b8, 2024-11-01:
Node info is already automatically requested and exchanged by a node if it hears ANY packet from a node it doesn't know. Simply having your node on in the area and blasting out smart position or telemetry will trigger an automatic node info exchange 🙂

cracky22, 2024-11-01:
> Node info is already automatically requested and exchanged by a node if it hears ANY packet from a node it doesn't know. Simply having your node on in the area and blasting out smart position or telemetry will trigger an automatic node info exchange 🙂

But if smart position time is too high and you are only in the area for a short time, this could be practical. In addition, functions are always useful - whether you use them is another matter

cracky22, 2024-11-01:
Think about it if, for example, you are hiking in the mountains and want to see all nearby devices - that would be the perfect solution for this. The acceleration sensor (assuming the board has one) detects the movement and sends node info more frequently. not too often to avoid spamming the mesh

cracky22, 2024-11-01:
> Node info is already automatically requested and exchanged by a node if it hears ANY packet from a node it doesn't know. Simply having your node on in the area and blasting out smart position or telemetry will trigger an automatic node info exchange 🙂

Or based on GPS if, for example, 500 meters have been covered.

b8b8, 2024-11-01:
You should decrease your smart position time then. Also not much point to transmitting your location and receiving node info if you aren't staying in the area. But I think the plan with the accel is to update smart position based on movement anyway, then powering up the the GPS to send a new position anyway.

GUVWAF, 2024-11-02:
I agree that smart position should suffice for this. The accelerometer will also detect movement when you stand up from sitting, or when you lay down your backpack, etc.

closed 2024-11-02 by garthvh