meshtastic/firmware #5220: [Feature Request]: Enable micro SD usage for Lilygo Tlora v2.1-1.6

agieocean, 2024-11-01:
### Platform

ESP32

### Description

It would be useful if there was a way to access the micro sd on the Lilygo Tlora v2.1-1.6 as a file system in order to serve more files.

My use case is to serve a largeish script to pyodide and use it as a small mobile library for a few people

agieocean, 2024-11-02:
I'm trying it myself and setting the pins:
SPI_MOSI              27
SPI_SCK                05
SPI_MISO              19
SDCARD_CS         13

Yields me a token error [0]     0x30
It then says "GO_IDLE_STATE failed
f_mount failed (3) The physical drive cannot work
crc error

The micro sd is FAT32 and is readable by my Ubuntu laptop

garthvh, 2024-11-02:
Not sure it is going to be possible to use both the Lora radio and a as card with these older esp32 chips, should be enough pins on an esp32 s3 device.

agieocean, 2024-11-02:
Shouldn't there be enough pins already since the reader is built into the board?

gjelsoe, 2024-11-02:
> I'm trying it myself and setting the pins: SPI_MOSI 27 SPI_SCK 05 SPI_MISO 19 SDCARD_CS 13
> 
> Yields me a token error [0] 0x30 It then says "GO_IDLE_STATE failed f_mount failed (3) The physical drive cannot work crc error
> 
> The micro sd is FAT32 and is readable by my Ubuntu laptop

I've used these settings for my tlora_v2_1_16 device with Store and Forward SD Edition (PINs are from datasheet) :
```
/*
 * Use SD Card for Store and Forward
 */
#define HAS_SDCARD
#define SPI_MOSI 15
#define SPI_MISO 2
#define SPI_SCK 14
#define SPI_CS 13
#define SDCARD_CS SPI_CS
``` 
And it's really **picky** about SD cards, got it working with a Kingston SDHC 16GB Class 10 and sometimes with a Kingston SDHC 8GB Class 4. If you are running into trouble with LoRa not initializing, you may need to use `SPIClass(HSPI)` to get it working.