meshtastic/firmware #5222: [Bug]: Can not stop the alert message sound

tomhanax, 2024-11-02:
### Category

Other

### Hardware

T-Deck

### Firmware Version

2.5.10.8462d65f

### Description

I use Ext Notif module to hear alert message, with these settings:
- Alert Message Buzzer
- Use I²S Pin as Buzzer
- Nag Timeout = 60

It used to work OK with previous firmware (some 2.3.x). Now it works only partially - it starts the sound but **I can not stop it**, I have to for the timeout (60 seconds). Also, sometimes, whole device freeze or restarts (maybe related to #5150).

If it is any help, I am able to try some code changes and upload, so if anybody has an idea what to try, feel free to navigate me.

zpatten, 2024-11-03:
Same issue here on `2.5.8.6485f03`.  It locks up often if wifi is enabled.  With wifi off it will only lock up when a message is received.

caveman99, 2024-11-03:
This is a known issue because the trackball re-uses some GPIOs on the t-deck. We're looking into several solutions.

zpatten, 2024-11-03:
Just to add a note and possible workaround for others that find this; as long as wifi is off and the buzzer (ext notif) is disabled I don't seem to be having random lockups or lockups after receiving messages.

gjelsoe, 2024-11-17:
Perhaps #5375 fixes this issue ?

tomhanax, 2024-11-25:
> Perhaps #5375 fixes this issue ?
Yes, it seems so. Now I can cancel the sound by clicking the trackball. Thanks!

closed 2024-11-25 by tomhanax

thumbtak2, 2025-12-27:
I know this is a late replay, but from my understanding, you have "Nag Timeout = 60". This means 60 notification reminders in a row.