meshtastic/firmware #5231: [Bug]: Station G2 (ESP32 S3) very sleepy after setting router mode

liquidraver, 2024-11-03:
### Category

Other

### Hardware

Station G2

### Firmware Version

2.5.7, 2.5.8, 2.5.9

### Description

Dear All!

My Station G2 is behaving abnormally after setting it to router.
After a while (~30-50 min) is just stops receiving on lora, and wakes up only for a little in 3 hour intervals (nodeinfo?), start working again and the stops.
It starts to work after I connect to it on wifi (wifi shouldn't be disabled after setting router mode?) and making a traceroute or something, but only for a little while.
It starts to work if I connect to it by meshsense app, but still, only for ~30-50 min.

MeshSense shows the node is working, sending out telemetry and neighbourinfo (empty, because not receiving packets), I can connect to it by wifi, but lora receive is not working.

I can recreate the behavior anytime:
- Full erase and flash with webflasher
- Connect to wifi (bluetooth stops working after connecting to wifi, another bug?)
- Enable Nodeinfo app (900 seconds, if I leave it on 0 it never sends nodeinfo, yet another?)
- Config lora to EU868, defaults
- Set device to router mode
- Node starts to sleep abnormally.
- Setting back to Client: not resolving the issue, only a full reinstall resolves it.

Attaching serial logs involving me sitting on the roof for a good half an hour, so it's not available yet (can I get logs by wifi somehow?)

liquidraver, 2024-11-03:
One more thing i forgot: node shows it's sending out telemetry and nodeinfo but my other node in the house not receiving it, so it's not really sent out

iondulgheru, 2024-11-03:
Since I have upgraded to 2.5.8 I have the same problem, but not only in router mode, also in client mode.
The device stops receiving messages but it can transmit. If I reboot the device it starts receiving again.

liquidraver, 2024-11-03:
correction: it seems 2.5.7 disables wifi correctly after changing to router mode, testing it out now, edited out from the report for now

iondulgheru, 2024-11-03:
Do you have Neighbour info enabled?
If yes, can you try to disable it and see if the problem is resolved?

liquidraver, 2024-11-04:
These are the last lines before everything stops for a long time, nothing happens on serial after that:

```
←[36m←[0m ←[34m DEBUG ←[0m| 12:39:10 905 [NeighborInfoModule] ←[34m handleReceived(LOCAL) (id=0x930421f5 fr=0x90 to=0x01, WantAck=0, HopLim=5 Ch=0x0 Portnum=71 rxtime=1730637550 priority=10)←[0m
←[36m←[0m ←[34m DEBUG ←[0m| 12:39:10 905 [NeighborInfoModule] ←[34m No modules interested in portnum=71, src=LOCAL←[0m
←[36m←[0m ←[34m DEBUG ←[0m| 12:39:10 905 [NeighborInfoModule] ←[34m Add packet record (id=0x930421f5 fr=0x90 to=0x01, WantAck=0, HopLim=5 Ch=0x0 Portnum=71 rxtime=1730637550 priority=10)←[0m
←[36m←[0m ←[34m DEBUG ←[0m| 12:39:10 905 [NeighborInfoModule] ←[34m Expanding short PSK #1←[0m
←[36m←[0m ←[34m DEBUG ←[0m| 12:39:10 905 [NeighborInfoModule] ←[34m Using AES128 key!←[0m
←[36m←[0m ←[34m DEBUG ←[0m| 12:39:10 905 [NeighborInfoModule] ←[34m enqueuing for send (id=0x930421f5 fr=0x90 to=0x01, WantAck=0, HopLim=5 Ch=0x8 encrypted rxtime=1730637550 hopStart=5 priority=10)←[0m
←[36m←[0m ←[34m DEBUG ←[0m| 12:39:10 905 [NeighborInfoModule] ←[34m txGood=34,txRelay=29,rxGood=66,rxBad=25←[0m
←[36m←[0m ←[34m DEBUG ←[0m| 12:39:10 905 [NeighborInfoModule] ←[34m Using channel 0 (hash 0x8)←[0m
←[36m←[0m ←[34m DEBUG ←[0m| 12:39:10 905 [NeighborInfoModule] ←[34m Expanding short PSK #1←[0m
←[36m←[0m ←[34m DEBUG ←[0m| 12:39:10 905 [NeighborInfoModule] ←[34m Using AES128 key!←[0m
←[36m←[0m ←[34m DEBUG ←[0m| 12:39:10 905 [NeighborInfoModule] ←[34m decoded message (id=0x930421f5 fr=0x90 to=0x01, WantAck=0, HopLim=5 Ch=0x0 Portnum=71 rxtime=1730637550 hopStart=5 priority=10)←[0m
←[37m←[0m ←[32m INFO  ←[0m| 12:39:10 905 ←[32m Telling client we have new packets 41←[0m
←[36m←[0m ←[34m DEBUG ←[0m| 12:39:11 906 [RadioIf] ←[34m Starting low level send (id=0x930421f5 fr=0x90 to=0x01, WantAck=0, HopLim=5 Ch=0x8 encrypted rxtime=1730637550 hopStart=5 priority=10)←[0m
←[36m←[0m ←[34m DEBUG ←[0m| 12:39:11 906 [RadioIf] ←[34m Drop Tx packet because dest is broadcast no-lora←[0m
←[36m←[0m ←[34m DEBUG ←[0m| 12:39:11 906 [RadioIf] ←[34m Packet transmitted : 894ms←[0m
```

liquidraver, 2024-11-05:
Same as:
#5235

closed 2024-11-05 by liquidraver