meshtastic/firmware #5238: [Feature Request]: Add option to disable automatic neighbourinfo requests from node

aljaxus, 2024-11-03:
### Platform

NRF52, ESP32, RP2040

### Description

We have a small problem on our [slovenian] mesh where strategically positioned nodes (routers) have their nodedb filled up pretty fast. This means that nodes are getting removed when the limit (of 100) is reached.

This becomes a real problem when this node hears new packets (position/neighbourinfo/telemetry) from the nodes that it removed and since the node is not present in nodedb, it will sent its nodeinfo to the mesh (also requesting nodeinfo from the unknown node, resulting in two packets being sent over the mesh).

This cycle constantly repeats when you have more than 100 active nodes.

I think the responsible part of the code is in [`src/mesh/MeshService.cpp#89`](https://github.com/meshtastic/firmware/blob/d00acc5d7bd8c936e5d8088463355b08398a2012/src/mesh/MeshService.cpp#L89). A new setting would have to be added and an if statement checking that setting when attempting to automatically broadcast node information.

This change would only require a protobuf change in a single Config protobuf definition, in my opinion `DeviceConfig` would be a best fit.

thebentern, 2024-11-03:
After talking to @GUVWAF I think first order of business here is to probably remove the interrogation of unknown nodes from the Router role altogether, since they're not a traditional client. There's some other areas for optimization as well potentially

aljaxus, 2024-11-04:
> After talking to @GUVWAF I think first order of business here is to probably remove the interrogation of unknown nodes from the Router role altogether, since they're not a traditional client. There's some other areas for optimization as well potentially

Sure, but please also make it an option for normal client role devices. I have some strategically placed nodes that act as routers but have the client role.

GUVWAF, 2024-11-04:
I think we can also disable it when the NodeDB is full. In my opinion it doesn't sound like something that is worth a new setting.

GUVWAF, 2024-11-04:
I would like to keep this open and have a look at disabling the requests when the NodeDB is full.

GUVWAF, 2024-11-04:
See #5255.

PR #5242 (2024-11-04): Exclude preferred routing roles from nodeinfo interrogation behavior
Closes #5238
closed 2024-11-04 by thebentern