meshtastic/firmware #5245: [Bug]: Enabling Environment Metrics Module causes reboot loop (2 min intervals)

mtsdawson, 2024-11-04:
### Category

Hardware Compatibility

### Hardware

Rak4631

### Firmware Version

2.5.8, 2.5.9, 2.5.10

### Description

I just built two RAK units, new parts, same config for both:

WisBlock Starter Kit (116016)
- WisBlock Base (RAK19007)
- WisBlock Core (RAK4631)

Slot A: GNSS GPS Location Module (RAK12500)
Slot B: None
Slot IO: Wislo Radar Sensor (RAK12013)
Slot C: Temp/Hum Sensor (RAK1901)
Slot D: Barometric Pressure Sensor (RAK1902)

I've tested the latest 3 FW builds on both units, each time with full erase.  All settings default.  When I enable the following, both nodes go into a reboot cycle with 2 minute intervals:

Telemetry --> Environment metrics module enabled

Usually name and settings are retained between reboots.  Occassionally, the node is fully reset back to defaults.  A few times, settings and channels were lost but the names were retained.

Is there a conflict with modules?  Or is this in the firmware?

### Relevant log output

```Shell
Here's two sample log entries I pulled just before the reboot is triggered:

--------------
NODE BR03:

Packet11/3/24 11:54:44 PMfrom: 3817326673 (!e387c851)
to: 4294967295 (!ffffffff)
decoded {
  portnum: TELEMETRY_APP
  payload: "\r\223S(g\032\017\rH\341\360A\025\024\256\rB\035|\322zD"
}
id: 2998481360
rx_time: 1730696083
hop_limit: 5
priority: BACKGROUND

--------------
NODE ST03:

Packet11/4/24 12:24:06 AMfrom: 3005780109 (!b328908d)
to: 4294967295 (!ffffffff)
decoded {
  portnum: TELEMETRY_APP
  payload: "\r\\Z(g\032\026\r\244p\367A\025\244p\005B\035\320\367zDh\000u\000\000\000\000"
}
id: 3390433411
rx_time: 1730697833
hop_limit: 3
priority: BACKGROUND
```

garthvh, 2024-11-04:
Best bet is a 1906 I'm slot c, I would pull the sensor in slot d and try again, both sensors won't work at the same time anyways.

mtsdawson, 2024-11-04:
Yeah that makes sense. Haven't messed with these two older sensors in a while (these were spare parts builds). I just ordered two 1906s, that should fix it. Thanks for the confirm.

mtsdawson, 2024-11-04:
Pulling the baro sensor from Slot D fixed the reboots. Flashed back to 2.5.10 and it's stable. Feel free to close this issue and big thanks for the help. 

I looked everywhere and didn't see any mention of this conflict. Would it make sense to add a comment to the sensors page?

closed 2024-11-04 by thebentern