meshtastic/firmware #5248: [Bug]: RP2040-Lora ignoring power INA configuration

SavageU, 2024-11-04:
### Category

Other

### Hardware

Other

### Firmware Version

2.5.7.f77c87d

### Description

RP2040-LORA-HF device is ignoring request to set INA219 as main battery device.
Continuously reports as PWD status at 0.0V despite being configured to read INA.

Occurring on 2.5.7 and 2.5.8 (downgraded back to 2.5.7 so I could increase cadence of power metrics transmit while testing.)

INA is being correctly recognised at 0x40 address on boot:

```
INFO  | ??:??:?? 3 Scanning for i2c devices...
DEBUG | ??:??:?? 3 Scanning for I2C devices on port 1
DEBUG | ??:??:?? 3 I2C device found at address 0x40
DEBUG | ??:??:?? 3 Wire.available() = 2
DEBUG | ??:??:?? 3 Register MFG_UID: 0x2000
INFO  | ??:??:?? 3 INA219 sensor found at address 0x40
DEBUG | ??:??:?? 3 I2C device found at address 0x76
DEBUG | ??:??:?? 3 Wire.available() = 1
DEBUG | ??:??:?? 3 Wire.available() = 1
INFO  | ??:??:?? 3 BMP-280 sensor found at address 0x76
INFO  | ??:??:?? 3 2 I2C devices found
DEBUG | ??:??:?? 3 acc_info = 0
DEBUG | ??:??:?? 3 found i2c sensor meshtastic_TelemetrySensorType_BMP280
DEBUG | ??:??:?? 3 found i2c sensor meshtastic_TelemetrySensorType_INA219
INFO  | ??:??:?? 3 S:B:30,2.5.7.f77c87d
INFO  | ??:??:?? 3 Build timestamp: 1729118132

```
However, the telemetry/update module is still reporting:

```
INFO  | ??:??:?? 48 [DeviceTelemetryModule] (Sending): air_util_tx=0.077111, channel_utilization=15.256667, battery_level=101, voltage=0.000000, uptime=48
```

When a remote request is placed from another node the same results occur:

```
meshtastic -s com11 --dest [redacted] --request-telemetry
Connected to radio
Sending device_metrics telemetry request to [redacted] on channelIndex:0 (this could take a while)
Telemetry received:
Battery level: 101.00%
Voltage: 0.00 V
Total channel utilization: 10.90%
Transmit air utilization: 0.06%
Uptime: 5 s
```

When power stats are explicitly requested, the correct data returns:
```
meshtastic -s com11 --dest [redacted] --request-telemetry
Connected to radio
Sending device_metrics telemetry request to [redacted] on channelIndex:0 (this could take a while)
Telemetry received:
Battery level: 101.00%
Voltage: 0.00 V
Total channel utilization: 17.88%
Transmit air utilization: 0.99%
Uptime: 769 s

meshtastic -s com11 --dest [redacted] --request-telemetry power
Connected to radio
Sending power_metrics telemetry request to [redacted] on channelIndex:0 (this could take a while)
Telemetry received:
powerMetrics:
  ch3Voltage: 4.772
  ch3Current: -0.4

```
However I have noticed that there appears to be a bug in the INA read - it is marking the single channel device as returning data on channel 3, I do not know if this is relevant or a red herring:
```
INFO  | ??:??:?? 23 [PowerTelemetryModule] (Sending): ch1_voltage=0.000000, ch1_current=0.000000, ch2_voltage=0.000000, ch2_current=0.000000, ch3_voltage=4.776000, ch3_current=-0.200000
```
Power configuration is correctly set (I think?) to INA address of 64 (or 0x40):
 ```
 "power": {
    "waitBluetoothSecs": 60,
    "sdsSecs": 4294967295,
    "lsSecs": 300,
    "minWakeSecs": 10,
    "deviceBatteryInaAddress": 64,
    "isPowerSaving": false,
    "onBatteryShutdownAfterSecs": 0,
    "adcMultiplierOverride": 0.0,
    "powermonEnables": "0"
  },
```

TheMalkavien, 2024-11-11:
I'm using multiples rp2040 with ina219. In order to have the ina reading as battery state, I have to define BATTERY_PIN (I put 26 even if I do not use it) and BATTERY_SENSE_RESOLUTION_BITS 12.  in the variant header and compile.
These defines enable portions of code for battery monitoring, and the use of INA219 instead if present.

I think it should be possible to use ina219 without this trick, but I did not want to change too much code for that. To consider maybe ?

funemnx, 2024-12-05:
I can confirm that TheMalkavien's suggested changes to the variant.h for rp2040 lora resolved this issue. Thanks!

Dude5101, 2024-12-11:
Also applies to the XIAO ESP32S3, you can't use the INA219 for battery measurement because XIAO has no native ADC configured for voltage reading, therefor no PIN's are assigned.

I assigned GPIO 3 randomly and INA219 measurements now work
I also had to assign an ADC_CHANNEL for power.cpp to work without errors

@fifieldt can we add Battery Pin and ADC_CHANNEL assignments to the variant.h files for these boards.
Will also help people that want to go the voltage divider route

nis0521, 2025-02-01:
@Dude5101  is there any way you can do a small write up on how you got the Xiao to work properly? A few guys no the reddit side were wondering how to get this working.

fifieldt, 2025-03-16:
There's a chance this was fixed by https://github.com/meshtastic/firmware/pull/5271

And137, 2025-04-09:
With the newest source my analog measurements on A0 stopped working. Even when I use the -BATTERY_PIN=1 flag, there’s a constant potential around 3,25V on A0 that makes the proper measurement impossible.