meshtastic/firmware #5257: [Bug]: GPS never gets a lock on 2.5.10 T-ECHO

samo-nz, 2024-11-05:
### Category

Other

### Hardware

T-Echo

### Firmware Version

2.5.10.0fc5c9b

### Description

GPS no longer works for my T-Echo on firmware 2.5.10.0fc5c9b Alpha

GPS Detected fine but never gets a lock. Flashing 2.5.9.936260f Beta it gets a lock within seconds of first boot.

I have flashed different versions 6 times now to confirm, waiting over an hour with no lock, same location.

Also - unable to reproduce - but when I first flashed it would be around a 50% chance the device said on screen "No GPS present" after reboot.

### Relevant log output

```Shell
---Firmware 2.5.10.0fc5c9b Alpha ---

INFO | 04:46:41 2 GPS power state moving from OFF to ACTIVE
---
DEBUG | 04:46:44 5 [GPS] Probing for GPS at 9600
DEBUG | 04:46:45 6 [GPS] Trying $PDTINFO (UC6580)...
DEBUG | 04:46:45 6 [GPS] Trying $PDTINFO (UM600)...
DEBUG | 04:46:46 7 [GPS] Trying $PCAS06,1*1A (ATGM336H)...
DEBUG | 04:46:46 7 [GPS] Trying $PCAS06,1*1A (ATGM332D)...
DEBUG | 04:46:47 8 [GPS] Trying $PAIR021*39 (AG3335)...
DEBUG | 04:46:47 8 [GPS] Trying $PAIR021*39 (AG3352)...
DEBUG | 04:46:48 9 [GPS] Trying $PQTMVERNO*58 (LC86)...
DEBUG | 04:46:48 9 [GPS] Trying $PCAS06,0*1B (L76K)...
INFO | 04:46:48 9 [GPS] L76K detected, using GNSS_MODEL_MTK Module
---
WARN | 05:01:41 902 [GPS] Couldn't publish a valid location: didn't get a GPS lock in time.
DEBUG | 05:01:41 902 [GPS] Took 900s to get lock
DEBUG | 05:01:41 902 [GPS] Predicting 0s to get next lock
DEBUG | 05:01:41 902 [GPS] 119s until next search
DEBUG | 05:01:41 902 [GPS] gps_update_interval >= 0s needed to justify hardsleep
INFO | 05:01:41 902 [GPS] GPS power state moving from ACTIVE to HARDSLEEP
DEBUG | 05:01:41 902 [GPS] publishing pos@0:2, hasVal=0, Sats=0, GPSlock=0
DEBUG | 05:01:41 902 [GPS] onGPSChanged() pos@0 time=0 lat=0 lon=0 alt=0
INFO | 05:01:41 902 [GPS] updatePosition LOCAL pos@0 time=0 lat=0 lon=0 alt=0
DEBUG | 05:01:41 902 [GPS] Setting local position: lat=0 lon=0 time=0 timestamp=0
DEBUG | 05:01:41 902 [GPS] Node status update: 6 online, 12 total
---
WARN | 05:20:05 904 [GPS] Couldn't publish a valid location: didn't get a GPS lock in time.
DEBUG | 05:20:05 904 [GPS] Took 900s to get lock
DEBUG | 05:20:05 904 [GPS] Predicting 0s to get next lock
DEBUG | 05:20:05 904 [GPS] 119s until next search
DEBUG | 05:20:05 904 [GPS] gps_update_interval >= 0s needed to justify hardsleep
INFO | 05:20:05 904 [GPS] GPS power state moving from ACTIVE to HARDSLEEP
DEBUG | 05:20:05 904 [GPS] publishing pos@0:2, hasVal=0, Sats=0, GPSlock=0
DEBUG | 05:20:05 904 [GPS] onGPSChanged() pos@0 time=1730784005 lat=0 lon=0 alt=0
INFO | 05:20:05 904 [GPS] updatePosition LOCAL pos@0 time=1730784005 lat=0 lon=0 alt=0
DEBUG | 05:20:05 904 [GPS] Setting local position: lat=0 lon=0 time=1730784005 timestamp=0
DEBUG | 05:20:05 904 [GPS] Node status update: 8 online, 12 total

--Back to Firmware 2.5.9.936260f Beta, location of device not changed, First boot--

WARN | 05:32:32 14 [GPS] SOME data is TOO OLD: LOC 50, TIME 50, DATE 4294967295
DEBUG | 05:32:32 14 [GPS] publishing pos@0:2, hasVal=0, Sats=0, GPSlock=1
DEBUG | 05:32:32 14 [GPS] No GPS lock
DEBUG | 05:32:32 14 [GPS] onGPSChanged() pos@0 time=0 lat=0 lon=0 alt=0
INFO | 05:32:32 14 [GPS] updatePosition LOCAL pos@0 time=0 lat=0 lon=0 alt=0
DEBUG | 05:32:32 14 [GPS] Setting local position: lat=0 lon=0 time=0 timestamp=0
DEBUG | 05:32:32 14 [GPS] Node status update: 6 online, 11 total
DEBUG | 05:32:32 14 [GPS] NMEA GPS time 2024-11-05 05:32:34 age 61
DEBUG | 05:32:32 14 [GPS] Upgrading time to quality GPS
DEBUG | 05:32:34 14 [GPS] PCF8563_RTC setDateTime 2024-11-05 05:32:34 (1730784754)
DEBUG | 05:32:34 14 [GPS] Read RTC time from PCF8563 getDateTime as 2024-11-05 05:32:34 (1730784754)
DEBUG | 05:32:34 14 [GPS] hasValidLocation RISING EDGE
DEBUG | 05:32:34 14 [GPS] Took 8s to get lock
DEBUG | 065:32:34 14 [GPS] Predicting 0s to get next lock
DEBUG | 05:32:34 14 [GPS] 119s until next search
DEBUG | 05:32:34 14 [GPS] gps_update_interval >= 0s needed to justify hardsleep
INFO | 05:32:34 14 [GPS] GPS power state moving from ACTIVE to HARDSLEEP
DEBUG | 05:32:34 14 [GPS] publishing pos@6729adf2:2, hasVal=1, Sats=11, GPSlock=1
```

samo-nz, 2024-11-05:
Not sure if related to this but on 2.5.9 I'm seeing this in the log output, don't think I've seen it before on other boards, the device has a pos lock and appears to be working fine. message is repeated throughout the log.

`WARN | 06:54:39 1364 [GPS] GPS Buffer full with 1023 bytes waiting. Flushing to avoid corruption.`

radonslide, 2024-11-05:
Same effect with FW 2.5.10.0fc5c9b Alpha on a T114 too.

samo-nz, 2024-11-05:
> Same effect with FW 2.5.10.0fc5c9b Alpha on a T114 too.

Sooo yeahhh I have a few devices around and checked it out on the T114 V2 also, same result, no lock

`DEBUG | ??:??:?? 5 [GPS] Probing for GPS at 9600
DEBUG | ??:??:?? 5 [GPS] Trying $PDTINFO (UC6580)...
DEBUG | ??:??:?? 6 [GPS] Trying $PDTINFO (UM600)...
DEBUG | ??:??:?? 6 [GPS] Trying $PCAS06,1*1A (ATGM336H)...
DEBUG | ??:??:?? 7 [GPS] Trying $PCAS06,1*1A (ATGM332D)...
DEBUG | ??:??:?? 7 [GPS] Trying $PAIR021*39 (AG3335)...
DEBUG | ??:??:?? 8 [GPS] Trying $PAIR021*39 (AG3352)...
DEBUG | ??:??:?? 8 [GPS] Trying $PQTMVERNO*58 (LC86)...
DEBUG | ??:??:?? 9 [GPS] Trying $PCAS06,0*1B (L76K)...
INFO | ??:??:?? 9 [GPS] L76K detected, using GNSS_MODEL_MTK Module
---
DEBUG | ??:??:?? 30 [Router] Upgrading time to quality NTP
DEBUG | 07:43:49 30 [Router] Module 'Admin' considered          <--Time Reference
---
WARN | 07:58:22 904 [GPS] Couldn't publish a valid location: didn't get a GPS lock in time.
DEBUG | 07:58:22 904 [GPS] Took 900s to get lock
DEBUG | 07:58:22 904 [GPS] Predicting 0s to get next lock
DEBUG | 07:58:22 904 [GPS] 119s until next search
DEBUG | 07:58:22 904 [GPS] gps_update_interval >= 0s needed to justify hardsleep
INFO | 07:58:22 904 [GPS] GPS power state moving from ACTIVE to HARDSLEEP
DEBUG | 07:58:22 904 [GPS] publishing pos@0:2, hasVal=0, Sats=0, GPSlock=0
DEBUG | 07:58:22 904 [GPS] onGPSChanged() pos@0 time=1730793502 lat=0 lon=0 alt=0
INFO | 07:58:22 904 [GPS] updatePosition LOCAL pos@0 time=1730793502 lat=0 lon=0 alt=0
DEBUG | 07:58:22 904 [GPS] Setting local position: lat=0 lon=0 time=1730793502 timestamp=0
DEBUG | 07:58:22 904 [GPS] Node status update: 6 online, 14 total`

fifieldt, 2024-11-05:
I believe this is fixed in https://github.com/meshtastic/firmware/commit/1bec23b6bbfc178c8cab8bb0c0682ff0fee40ac5 , but it looks like it missed the 2.5.10 release :(

samo-nz, 2024-11-05:
> I believe this is fixed in [1bec23b](https://github.com/meshtastic/firmware/commit/1bec23b6bbfc178c8cab8bb0c0682ff0fee40ac5) , but it looks like it missed the 2.5.10 release :(

That was simply reporting no GPS present as far as I could see? this is full blown no GPS lock ever acquired

fifieldt, 2024-11-05:
Yeah, Basically there's an unexpected fix in that patch (the return false). It looks like without it the setup code didn't run.

samo-nz, 2024-11-05:
I see, you may wanna pull that alpha as it appears to affect a lot of devices? I eagerly await the next release, cheers for the heads up =)

closed 2024-11-05 by samo-nz

thebentern, 2024-11-05:
> I see, you may wanna pull that alpha as it appears to affect a lot of devices? I eagerly await the next release, cheers for the heads up =)

Since it's just a temporary display bug, and the device still has GPS, we'll just push the fix in the next release. :-)

fifieldt, 2024-11-05:
@thebentern , it's not a temporary display bug. I think the code in 2.5.10 doesn't run the GPS setup steps.

samo-nz, 2024-11-05:
> > I see, you may wanna pull that alpha as it appears to affect a lot of devices? I eagerly await the next release, cheers for the heads up =)
> 
> Since it's just a temporary display bug, and the device still has GPS, we'll just push the fix in the next release. :-)

No, GPS location does not work on this version at all. tested on both T114 and T-Echo

thebentern, 2024-11-05:
Well we have a new alpha release with the bug fix live now 😉