meshtastic/firmware #5261: [Bug]: RTC not applying time to fixed position packets

b8b8, 2024-11-05:
### Category

Other

### Hardware

Not Applicable

### Firmware Version

2.5.9

### Description

I have multiple RAK nodes with RTC's.
When the device is set with a fixed position (over bluetooth) the correct time is maintained on all future position packets that are sent. After rebooting (remotely) to update settings, the RTC still has a valid time but all future fixed position packets have their time set to zero.

Valid time properly found after fresh boot from RTC
DEBUG | ??:??:?? 1 Saved TZ: PST8PDT,M3.2.0,M11.1.0
DEBUG | ??:??:?? 1 Set Timezone to PST8PDT,M3.2.0,M11.1.0
DEBUG | ??:??:?? 1 **Read RTC time from RV3028 getTime as 2024-11-05 22:07:46** (1730844466)
DEBUG | 14:07:46 1 NeighborInfoModule is disabled
DEBUG | 14:07:46 1 Rescanning for I2C keyboard

Time is then stripped off the position packet when it is requested from another node.
INFO | 14:09:16 92 [Router] **Stripping time 0 from position send**
INFO | 14:09:16 92 [Router] Position reply: time=0 lat=49.00000 lon=-123.00000
DEBUG | 14:09:16 92 [Router] Partially randomized packet id 1561338937
INFO | 14:09:16 92 [Router] Asked module 'position' to send a response

b8b8, 2024-11-13:
Please reopen, this is not fixed and i lack the permission to open it myself.

thebentern, 2024-11-13:
Done. I'll try to dig an RTC module out. I think I have one of the WisBlock ones. If not, I'll order one and try to work with one of the external modules

caveman99, 2024-11-13:
> Done. I'll try to dig an RTC module out. I think I have one of the WisBlock ones. If not, I'll order one and try to work with one of the external modules

T-Echo has one too

jp-bennett, 2024-11-16:
Another solution to this issue, is on incoming position packets, when time is missing or 0, replace with current time.

thebentern, 2024-11-21:
Closed with #5396

PR #5262 (2024-11-06): Add exception for RTC to not strip time from position
Closes #5261
closed 2024-11-21 by thebentern