meshtastic/firmware #5265: [Bug]: Web client not bundled with 2.5.9.936260f Beta (stable) release served via Web Flasher

evoncken, 2024-11-06:
### Category

Other

### Hardware

Heltec V3

### Firmware Version

2.5.9.936260f

### Description

Device was running 2.5.4, performed Full Erase and Flash to latest stable release, 2.5.9.936260f Beta.
After flash, web interface (HTTP and HTTPS) is running, but not functional; it shows an error message indicating the web client content is missing:

-----
Web server is running.

The content you are looking for can't be found. Please see: FAQ.

admin
-----

### Relevant log output

```Shell
[HTTPS:I] New connection. Socket FID=52
[HTTPS:I] Request: GET / (FID=52)
[ 63909][E][vfs_api.cpp:105] open(): /littlefs/static/index.html does not exist, no permits for creation
[ 63919][E][vfs_api.cpp:105] open(): /littlefs/static/index.html.gz does not exist, no permits for creation
[ 63930][E][vfs_api.cpp:105] open(): /littlefs/static/index.html.gz does not exist, no permits for creation
WARN  | ??:??:?? 63 [WebServerThread] File not available - /static/index.html.gz
[HTTPS:I] Request: GET /favicon.ico (FID=52)
[ 64249][E][vfs_api.cpp:105] open(): /littlefs/static/favicon.ico does not exist, no permits for creation
[ 64259][E][vfs_api.cpp:105] open(): /littlefs/static/favicon.ico.gz does not exist, no permits for creation
[ 64271][E][vfs_api.cpp:105] open(): /littlefs/static/index.html.gz does not exist, no permits for creation
WARN  | ??:??:?? 64 [WebServerThread] File not available - /static/index.html.gz
[HTTPS:I] Request: GET / (FID=52)
[ 67028][E][vfs_api.cpp:105] open(): /littlefs/static/index.html does not exist, no permits for creation
[ 67039][E][vfs_api.cpp:105] open(): /littlefs/static/index.html.gz does not exist, no permits for creation
[ 67050][E][vfs_api.cpp:105] open(): /littlefs/static/index.html.gz does not exist, no permits for creation
WARN  | ??:??:?? 67 [WebServerThread] File not available - /static/index.html.gz
[HTTPS:I] Request: GET /favicon.ico (FID=52)
[ 67118][E][vfs_api.cpp:105] open(): /littlefs/static/favicon.ico does not exist, no permits for creation
[ 67128][E][vfs_api.cpp:105] open(): /littlefs/static/favicon.ico.gz does not exist, no permits for creation
[ 67140][E][vfs_api.cpp:105] open(): /littlefs/static/index.html.gz does not exist, no permits for creation
WARN  | ??:??:?? 67 [WebServerThread] File not available - /static/index.html.gz
[HTTPS:I] New connection. Socket FID=53
[HTTPS:E] SSL_accept failed. Aborting handshake. FID=53
[HTTPS:I] Connection closed. Socket FID=53
[HTTPS:I] New connection. Socket FID=53
[HTTPS:E] SSL_accept failed. Aborting handshake. FID=53
[HTTPS:I] Connection closed. Socket FID=53
[HTTPS:I] New connection. Socket FID=53
[HTTPS:I] Request: GET / (FID=53)
[ 77890][E][vfs_api.cpp:105] open(): /littlefs/static/index.html does not exist, no permits for creation
[ 77901][E][vfs_api.cpp:105] open(): /littlefs/static/index.html.gz does not exist, no permits for creation
[ 77912][E][vfs_api.cpp:105] open(): /littlefs/static/index.html.gz does not exist, no permits for creation
WARN  | ??:??:?? 77 [WebServerThread] File not available - /static/index.html.gz
[HTTPS:I] Request: GET /favicon.ico (FID=53)
[ 78091][E][vfs_api.cpp:105] open(): /littlefs/static/favicon.ico does not exist, no permits for creation
[ 78102][E][vfs_api.cpp:105] open(): /littlefs/static/favicon.ico.gz does not exist, no permits for creation
[ 78113][E][vfs_api.cpp:105] open(): /littlefs/static/index.html.gz does not exist, no permits for creation
WARN  | ??:??:?? 78 [WebServerThread] File not available - /static/index.html.gz
DEBUG | ??:??:?? 81 [Power] Battery: usbPower=0, isCharging=0, batMv=-1, batPct=0
[HTTPS:I] Connection timeout. FID=52
[HTTPS:I] Connection closed. Socket FID=52
```

evoncken, 2024-11-06:
Edit: tried "Full Erase and Install" followed by normal install - this did not help either

evoncken, 2024-11-06:
Note: tried with 2.5.10 alpha and 2.5.11 alpha as well (Full Erase and Install -> NOK, Install -> NOK). Could there be an issue with the Web Flasher?

thebentern, 2024-11-06:
I have confirmed this is an issue with the web-flasher. Investigating what's going on:
https://discord.com/channels/867578229534359593/871553714782081024/1303721074009182268