meshtastic/firmware #5268: [Feature Request]: nRF52832 support

Kezii, 2024-11-06:
### Platform

NRF52

### Description

I see that nRF52832 support was removed from this repo in the past

I find that unfortunate, since there are a lot of devices with nRF52832 that can be used

I see that, compiling nRF52840 target and disabling some features, the code can still fit inside an nRF52832

why is it impossible to have the nRF52832 supported?

https://github.com/meshtastic/firmware/issues/3921#issuecomment-2352897199

garthvh, 2024-11-06:
As it says, not enough resources.

closed 2024-11-06 by garthvh

Kezii, 2024-11-07:
have you even read my message? lol

fifieldt, 2024-11-07:
Flash. nRF52832 just doesn't have enough.

[image]

thebentern, 2024-11-07:
Right. We removed support for these because it just wasn't enough resources for Meshtastic and the NRF52840 is only marginally more expensive from a BoM perspective.