meshtastic/firmware #5297: [Feature Request]: More robust muting, expand ignore_incoming array

mdesmedt, 2024-11-10:
### Platform

Cross-Platform

### Description

I'm hoping Meshtastic firmware, and the clients, can support better muting (ignoring) for user-specified nodes. It is a configuration feature I would like to be able to use.

I am aware ignore_incoming exists, and it is exposed in the Android client, however the iOS client has previously decided against exposing this per https://github.com/meshtastic/Meshtastic-Apple/issues/387 because it needs a reboot and maxes out at 3 entries. Addressing these limitations to make it a more usable end-user feature in addition to it being testing functionality would be appreciated.

garthvh, 2024-11-10:
It does not reboot anymore, you are welcome to sponsor the feature on iOS, the array is probably not going to be expanded as you can't have arrays of dynamic size in protobufs.

mdesmedt, 2024-11-12:
I have, thanks to a suggestion, opted for an alternative approach where `NodeInfo` gets an `is_ignored` flag just like the `is_favorite` flag. This flag is checked in `Router::perhapsHandleReceived` and if set the packet is dropped.

I have tested my changes with two NRF52 devices and it works.

I am very new to open source but I will try to submit pull requests for the changes.

thebentern, 2024-11-12:
> I am very new to open source but I will try to submit pull requests for the changes.

Well, it's a phenomenal first change. 😄 

Are you on the Meshtastic discord server? I can get your contributor flair added if you send me your handle.

mdesmedt, 2024-11-12:
> > I am very new to open source but I will try to submit pull requests for the changes.
> 
> Well, it's a phenomenal first change. 😄
> 
> Are you on the Meshtastic discord server? I can get your contributor flair added if you send me your handle.

Thanks! I am anji0841 on Discord.

mdesmedt, 2024-11-12:
This is implemented in the firmware with commit 2ec3958

closed 2024-11-12 by mdesmedt