meshtastic/firmware #5306: [Feature Request]: Add ".local" addresses to isPrivateIpAddress()

pdxlocations, 2024-11-10:
### Platform

Cross-Platform

### Description

[isPrivateIpAddress](https://github.com/meshtastic/firmware/blob/ad214ea42a8c484f1ff90cd76dd58c89408e33fc/src/mqtt/MQTT.cpp#L707) checks for local IP addresses but does not include a check for .local addresses.

fifieldt, 2024-11-10:
https://datatracker.ietf.org/doc/html/rfc6762

fifieldt, 2024-11-16:
Just checking, @pdxlocations - do you use this domain for mqtt or know of people that do, or is this more of a request to make the function "complete" ?

pdxlocations, 2024-11-16:
I had been using, for example: raspberrypi.local to connect to my broker.  I've since changed to using the IP address to take advantage of the iPrivateIpAddress function. 

The feature request is not dire but as you said to make the function "complete".  If we want to honor "All" local addresses adding .local may save a user some frustration.  Alternatively, or in addition to, I'd be happy to write something up for the docs listing the allowable formats.

fifieldt, 2024-12-15:
I think let's go with the doc solution?

esev, 2024-12-19:
I've run across a similar issue as well. I have MQTT in a docker container that I refer to with a host name (not a `.local`) that resolves to a RFC 1918 private IP address. It would be nice if the code could check the IP address after mDNS/DNS resolution. But I didn't see any straightforward way to do this.

For now, I've just hacked my local firmware with this one-liner:

```c++
isMqttServerAddressPrivate = true; // isPrivateIpAddress(moduleConfig.mqtt.address);
```

fifieldt, 2024-12-20:
Yup, I don't think we're going to be able to handle DNS names for this :( Marking this as shut.

PR #5627 (2024-12-21): Check if MQTT remote IP is private
Consider a remote MQTT server as private if its host name resolves to a private IP.

This doesn't currently work on portduino, as the [WifiClient](https://github.com/meshtastic/WiFi/blob/master/src/WiFiClient.h) is missing support for the remoteIP method.

fixes #5306
closed 2024-12-20 by fifieldt