meshtastic/firmware #5307: [Bug]: HEAP memory leak in recent beta

LowVoltagePirate, 2024-11-10:
### Category

WiFi

### Hardware

T-Beam

### Firmware Version

2.5.11

### Description

I'm observing a memory leak with my t-beam with mqtt and wifi enabled. After approx 6-12 hours there's no heap memory left, WiFi disconnects and never reconnects, after a power cycle all is good again for those hours. It was fine with 2.4.5

karamo, 2024-11-11:
FW 2.5.12 new compiled with options` -DDEBUG_HEAP` and `-DDEBUG_HEAP_MQTT`
multiple Heap-Leaks ...
top 2 are T-Beam, bottom left T-Deck, bottom right TLora T3_V1.6.1
[image]

GUVWAF, 2024-11-11:
I found one place where a memory leak would occur when using MQTT. That one should at least be fixed with #5311.

closed 2024-11-11 by caveman99

karamo, 2024-11-11:
> I found one place where a memory leak would occur when using MQTT. That one should at least be fixed with #5311.

I will test it ...

karamo, 2024-11-11:
the first one is gone, but there are more Heap-Leaks:
see #5316 and #5317

derekpurdy, 2024-11-21:
what version will this be released in?

fifieldt, 2024-11-21:
> what version will this be released in?

Probably 2.5.14

GUVWAF, 2024-11-21:
It is already in 2.5.13.

PR #5311 (2024-11-11): Fix memory leak in MQTT
Found a potential source for a memory leak. This should at least partially fix #5307.