meshtastic/firmware #5309: [Bug]: remote administration - only 1st pub key works

xinrag, 2024-11-11:
### Category

Other

### Hardware

Rak4631

### Firmware Version

2.5.12.aa184e6

### Description

Remote rak4631 working as repeater. Three admin pub keys are stored there.

If I try to connect with NO matching key a true error messages whas shown with invalid pub key. Good.

Connecting with a matching pub key works (only) if it is the 1st one stored in the remote rak device.

If the pub key is matching for the the 2nd or 2rd entry the connection never works... still try to connect and nothing is happening.

xinrag, 2024-11-11:
2nd pub key is working here if remote device is a esp32. Interesting!

pagex, 2024-11-27:
Can confirm only first key is accepted for remote admin, tested on RP2040 second and third keys produce:

INFO | ??:??:?? 257 [Router] Received Admin from=0x6c6965e8, id=0x8f5ccd97, portnum=6, payloadlen=13
INFO | ??:??:?? 257 [Router] Ignore unauthorized admin payload 3

But even first key while in logs on target device says low level send completed, client device still sits there with 0% progress bar

fifieldt, 2024-11-27:
Thanks for confirming @pagex

jp-bennett, 2024-11-27:
> Can confirm only first key is accepted for remote admin, tested on RP2040 second and third keys produce:
> 
> INFO | ??:??:?? 257 [Router] Received Admin from=0x6c6965e8, id=0x8f5ccd97, portnum=6, payloadlen=13 INFO | ??:??:?? 257 [Router] Ignore unauthorized admin payload 3

That's interesting. That code path actually indicates the admin request isn't sent as a public-key encrypted message at all: https://github.com/meshtastic/firmware/blob/master/src/modules/AdminModule.cpp#L103C19-L103C52

I suspect this actually means there's a problem with the key exchange. You can test this with a regular DM, and see if that is delivered successfully, with the log indicating it came via PKC.

closed 2024-12-17 by garthvh