meshtastic/firmware #5317: [Bug]: one more Heap-Leak @Traceroute

karamo, 2024-11-11:
### Category

Other

### Hardware

Not Applicable

### Firmware Version

2.5.12 + fix

### Description

ther is one more Heap-Leak @Traceroute (red signs)
[image]

**Scenario:**
FW 2.5.12 with fix PR https://github.com/meshtastic/firmware/pull/5311
3 T-Beam local direct with noch connection to other nodes.

GUVWAF, 2024-11-12:
Is this when sending or receiving traceroutes? Also, might it be it also happens on other packet types like position requests or just text messages?

karamo, 2024-11-12:
I made Traceroute with MeshSense.
Other actions not tested.

karamo, 2024-11-12:
At starting Meshsense and connecting via IP to that node, there is a down-peak followd by a jump up of the Heap.
[image]

Log of Meshsense:
[MeshSense_Start_IP_4Nodes.zip](https://github.com/user-attachments/files/17715584/MeshSense_Start_IP_4Nodes.zip)

Scenario: 4 T-Beam local direkt active

karamo, 2024-11-12:
At he 1st DM there is a peak down, followed by up an no permanent Heap-down.
[image]
Log: [MEshSense_SendMsg.zip](https://github.com/user-attachments/files/17715820/MEshSense_SendMsg.zip)

karamo, 2024-11-12:
Sending Traceroute every 30s (FW limit), Heap permanently damaged:
[image]
Log: 
[MeshSense_Send-Traceroute.zip](https://github.com/user-attachments/files/17716076/MeshSense_Send-Traceroute.zip)

karamo, 2024-11-12:
It seems to me that **TraceRoute** is interrupted if another packet is received in the meantime.

karamo, 2024-11-12:
after ≈10 mins Heap partially recovered
[image]

GUVWAF, 2024-11-12:
@karamo Many thanks for all the testing. 

Can you try if #5328 fixes it?

karamo, 2024-11-12:
> Can you try if #5328 fixes it?

Yes, ist running ... wait 1 to 3 hours ...

karamo, 2024-11-12:
ok, it looks good.
Reboot 2.5.12 + Fix1 + Fix2
Significant points:
picture 1:
19:36:49 - 52428 B - Reboot an heap down with first slope to
19:56:31 - 38424 B - begin 2nd slope to jump up at
20:42:56 - 35884 B ^^
21:17:01 - 40220 B - start of MeshSense (red 1 in 2nd picture)
picture 2:
27:17:41 - 47724 B - maximum following slow down
21:34:43 - 45044 B - start 1st Traceroute every ≈30s no significant Heap-Leak (red 2 in 2nd picture)
[image]
[image]

closed 2024-11-13 by thebentern