meshtastic/firmware #5321: [Feature Request]: Add default GPS pin settings for Heltec devices

KevinKW87, 2024-11-12:
### Platform

ESP32

### Description

I got a request from a customer to add a GPS module to a Heltec WSL v3. I checked the variant.h file to see if there were any default pin setting for the TX/RX and there weren't any. 

Recommend adding a default pin selection for Heltec ESP32 devices (like what the RAK4631 has) so when customers do a fresh firmware installation, their GPS works right away. This is a bit critical for embedded boards inside enclosures.

For lack of any other pin selection, recommend using pins 45/46 as shown in this tutorial: https://adrelien.com/blog/how-to-add-gps-to-heltec-v3/

fifieldt, 2024-11-12:
Seems sane and in-line with what we do for other similar devices.

caveman99, 2024-11-12:
The RAK4631 has those pins because the baseboards have a designated routing for RX and TX pins to the slot where the GPS will slide in. Same for the Heltec T114 with their designated GPS connector. Any other board with those defines has an onboard GPS chip. 

We removed the default pins on ALL other boards by policy, those were clogging up GPIO's for other use cases. You can define the pins by setting them in the position module. This is much more flexible than parking on some fixed GPIOs that might be needed elsewhere.

closed 2024-11-12 by caveman99