meshtastic/firmware #5327: [Bug]: T-Echo RGB Matrix LED wrong/incomplete pinning.

Lma-Mike-Zulu, 2024-11-12:
### Category

Hardware Compatibility

### Hardware

T-Echo

### Firmware Version

2.5.11.8e2a3e5

### Description

Pinout for Matrix LED on T-Echo is wrong(incomplete)

By reviewing the vendors scematic [2021/6/23] ( LED color Labeling is wrong but LED and other Port Pin Label should be fine) and Datasheet for RGB Matrix LED FM-B2020RGBA-HG and checking code with real behaviour on my device I found the correct Port Pinning for the LED should be.

RED P1.03 (32 + 3)
GREEN P1.01 (32 + 1)
BLUE P0.14 (0 + 14)

I made a code proposal on Pull Request #5326

fifieldt, 2024-11-13:
From the pull request: "Your assertions are correct. Unfortunately, changing to new (correct) LED scheme broke the previous iterations of T-Echos. :-/"

So we may need to look at a new variant or some extra logic to handle the different generations of T-Echo.

Lma-Mike-Zulu, 2024-11-14:
Thanks for the feedback. 

Hopefully there will be a solution in the future to get full functionality for the Matrix LED.
Please dont hasitate to contact me if you need any feedback regarding the 11/2024 T-Eco hardware I own.

closed 2025-01-21 by github-actions[bot]