meshtastic/firmware #5333: [Feature Request]: Appropriate 900MHz band frequency for Indonesia?

ndoo, 2024-11-13:
### Platform

Cross-Platform

### Description

Dear devs,

There's 2 possibly relevant current sections of Indonesian law (https://jdih.kominfo.go.id/produk_hukum/view/id/862/t/peraturan+menteri+komunikasi+dan+informatika+nomor+2+tahun+2023) regarding 900 MHz band in Indonesia for IoT:

1)

[image]

2)

[image]

"Sesuai dengan Standar Teknis yang berlaku" is translated by Google to "In accordance with applicable Technical Standards".

For the life of me, I can't find any reference to this in any current regulation in the search page of [JDIH Kemkominfo - Search](https://jdih.kominfo.go.id/produk_hukum/pencarian).

What's an appropriate direction to take to determine the appropriate Tx power for ID_923? It looks like the Meshtastic community in Indonesia is currently using SG_923.

Would adding ID_923 at 26 dBm or 20 dBm (which?), 250 kHz b/w, 1% duty cycle be a good idea?

garthvh, 2024-11-13:
A 1% duty cycle won't work for meshtastic, you might contact the lorawan alliance and see if they have thoughts.

ndoo, 2024-11-13:
> A 1% duty cycle won't work for meshtastic, you might contact the lorawan alliance and see if they have thoughts.

Good point. Actually, the route to go would be the SRD allocation at 920 - 923 MHz, so I'll continue looking for anything that might provide the allowed EIRP for it.

ndoo, 2024-11-13:
Looks like this is the most recent document that specifies EIRP for 920 - 923 MHz band - at <= 500 mW (27 dBm):

[Regulation of the Minister of Communication and Information Technology Number 35 of 2015 concerning Technical Requirements for Short-Range Telecommunication Tools and Equipment (Regulation of the Minister of Communication and Information Technology Number 35 of 2015 concerning Technical Requirements for Short-Range Telecommunication Tools and Equipment)](https://jdih.kominfo.go.id/produk_hukum/view/id/510/t/peraturan+menteri+komunikasi+dan+informatika+nomor+35+tahun+2015+tanggal+31+desember+2015)

Applicable section:

[image]

In summary:

| Year | Document |                                     920 - 923 MHz EIRP/BW/Duty                                     |         LPWAN 920 - 923 MHz EIRP/BW/Duty        | URL                                                                                                                                             |
|:----:|:--------:|:--------------------------------------------------------------------------------------------------:|:-----------------------------------------------:|-------------------------------------------------------------------------------------------------------------------------------------------------|
| 2015 |    35    |                                   500 mW/unspecified/unspecified                                   |                   unspecified                   | [Link](https://jdih.kominfo.go.id/produk_hukum/view/id/510/t/peraturan+menteri+komunikasi+dan+informatika+nomor+35+tahun+2015+tanggal+31+desember+2015) |
| 2019 |     1    | unspecified ("Sesuai dengan persyaratan teknisyang ditetapkan dengan Peraturan Direktur Jenderal") | GW: 400 mW/250 kHz/1% Client: 100 mW/250 kHz/1% | [Link](https://jdih.kominfo.go.id/produk_hukum/view/id/676/t/peraturan+menteri+komunikasi+dan+informatika+nomor+1+tahun+2019+tanggal+24+april+2019)     |
| 2023 |     2    |                      unspecified ("Sesuai dengan Standar Teknis yang berlaku")                     | GW: 400 mW/250 kHz/1% Client: 100 mW/250 kHz/1% | [Link](https://jdih.kominfo.go.id/produk_hukum/view/id/862/t/peraturan+menteri+komunikasi+dan+informatika+nomor+2+tahun+2023)                           |

How about adding ID_923 based on the 2015 regulation? It's superseded by the 2019 and 2023 regulation, which do not provide SRD limits for the band.

ndoo, 2024-11-13:
PRs above submitted based on 2015 SRD limits

closed 2024-11-13 by garthvh

chalei, 2025-05-21:
hello @ndoo thank you for doing this request. I was about to do the same thing, because meshtastic community is growing in Indonesia at the moment and yes we are still using SG_923 band.

Hoping there would be Indonesia band soon

garthvh, 2025-05-22:
> [@garthvh](https://github.com/garthvh) because KR_923 doesn't specify a tx power limit, it gets limited to 17dBm by firmware which is not the intended behavior for Indonesia. Shall I resubmit PRs?

1% duty cycle is the blocker.

ndoo, 2025-05-25:
Hi @chalei I think you will have to write to Komdigi to convince them to increase the 1% duty cycle.

According to the latest document [Perubahan Atas Peraturan Menteri Komunikasi dan Informatika Nomor 2 Tahun 2023 Tentang Penggunaan Spektrum Frekuensi Radio Berdasarkan Izin Kelas Detail Produk Dokumen Lainnya Abstrak](https://jdih.komdigi.go.id/produk_hukum/unduh/id/949/t/peraturan+menteri+komunikasi+dan+digital+nomor+2+tahun+2025)

[image]

It's still not possible to use Meshtastic.

I suggest one of the following if you are writing to them:

- Increase duty cycle of "Gateway/Base Station" to ≥10%
- Add a line after the 2 LPWAN category for Mesh/general SRD with ≥10% duty cycle or no duty cycle limit.