meshtastic/firmware #5358: [Bug]: Panic when setting user info

bjpetit, 2024-11-15:
### Category

BLE

### Hardware

T-Beam

### Firmware Version

2.5.13.295278b

### Description

When setting user Long Name and Short Name an unhandled exception occurs.
The firmware install was done with a full erase. When I attempt to set the name back through the iOS app an exception is seen.

### Relevant log output

```Shell
INFO | 02:38:02 237 [Router] Received Admin from=0x0, id=0x895e1252, portnum=6, payloadlen=15

INFO | 02:38:02 237 [Router] Handle admin payload 32

INFO | 02:38:02 237 [Router] Client set owner

DEBUG | 02:38:02 237 [Router] old user Meshtastic 88ec/88ec, channel=0

DEBUG | 02:38:02 237 [Router] Incoming Pubkey: b0 2e 7a 16 61 20 4e 1e c4 95 a7 ab e8 3f 5c cc 04 16 cd be 7f e7 ba b7 78 49 c3 0a 1a c9 1e 2c

INFO | 02:38:02 237 [Router] Public Key set for node, not updating!

DEBUG | 02:38:02 237 [Router] Saved Pubkey: b0 2e 7a 16 61 20 4e 1e c4 95 a7 ab e8 3f 5c cc 04 16 cd be 7f e7 ba b7 78 49 c3 0a 1a c9 1e 2c

DEBUG | 02:38:02 237 [Router] Update changed=1 user WGNode/WN, channel=0

DEBUG | 02:38:02 237 [Router] Node status update: 4 online, 4 total

INFO | 02:38:02 237 [Router] Save /prefs/db.proto

Guru Meditation Error: Core 1 panic'ed (IntegerDivideByZero). Exception was unhandled.

Core 1 register dump:

PC : 0x4018f6bd PS : 0x00060c30 A0 : 0x8018f7ec A1 : 0x3ffd7900

A2 : 0x3ffbb79c A3 : 0x00000000 A4 : 0x00060a20 A5 : 0x00060a23

A6 : 0x00060a23 A7 : 0x00000001 A8 : 0x00000113 A9 : 0x3ffd78e0

A10 : 0x00000001 A11 : 0x3ffbbc4c A12 : 0x000000ff A13 : 0x00039fe2

A14 : 0x007c0478 A15 : 0x003fffff SAR : 0x00000019 EXCCAUSE: 0x00000006

EXCVADDR: 0x00000000 LBEG : 0x40093e14 LEND : 0x40093e2a LCOUNT : 0xffffffff

Backtrace: 0x4018f6ba:0x3ffd7900 0x4018f7e9:0x3ffd7930 0x4018f927:0x3ffd7970 0x4018fd31:0x3ffd79e0 0x401917dd:0x3ffd7a10 0x4018d00e:0x3ffd7a30 0x40161672:0x3ffd7a50 0x40094781:0x3ffd7a70 0x4009321e:0x3ffd7a90 0x40093068:0x3ffd7ab0 0x400930e1:0x3ffd7ae0 0x400e1447:0x3ffd7b00 0x400efd2d:0x3ffd7b20 0x40152f67:0x3ffd7b50 0x400d8711:0x3ffd7ba0 0x401388a5:0x3ffd7c10 0x4013896d:0x3ffd7c30 0x4013ab3b:0x3ffd7c50 0x4013ad6f:0x3ffd7c70 0x400dbbd8:0x3ffd7d90 0x4012a699:0x3ffd7db0 0x4012957a:0x3ffd80d0 0x40135214:0x3ffd82f0 0x4013556a:0x3ffd8350 0x4013cdc9:0x3ffd8380 0x4014fd5a:0x3ffd83c0 0x4013d771:0x3ffd83e0 0x400e7ac1:0x3ffd8420

ELF file SHA256: ea1dda66c7964b83

E (28525) esp_core_dump_flash: Core dump flash config is corrupted! CRC=0x7bd5c66f instead of 0x0

Rebooting...

ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)

configsip: 0, SPIWP:0xee

clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00

mode:DIO, clock div:2

load:0x3fff0030,len:1184

load:0x40078000,len:13232

load:0x40080400,len:3028

entry 0x400805e4

E (863) esp_core_dump_flash: No core dump��ѥѥ���found!

E (863) esp_core_dump_flash: No core dump partition found!

[ 11][D][esp32-hal-cpu.c:244] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz

[ 474][I][esp32-hal-psram.c:96] psramInit(): PSRAM enabled

��@INFO | ??:??:?? 0 

//\ E S H T /\ S T / C

INFO | ??:??:?? 0 Booted, wake cause 0 (boot count 1), reset_reason=reset

DEBUG | ??:??:?? 0 Filesystem files (1048576/1048576 Bytes):

DEBUG | ??:??:?? 0  /prefs/channels.proto (57 Bytes)
```

bjpetit, 2024-11-15:
Hmmm. Did an additional factory reset and now it has changed the name without the exception. So maybe the original install wasn't fully clean.

bjpetit, 2024-11-15:
Looking closer. I'm seeing some panics coming in from other node updates. Looking like a dup of #5354 / #5356

memo-567, 2024-11-15:
The same problems here with a Heltec V3. Unfortunately, I don’t have any log files.

todd-herbert, 2024-11-15:
Thanks for grabbing that full set of info!
I'm away from home myself, but just to give anyone else a head start on looking into it, here's the decoded exception:
```
    0x4018f6bd: lfs_alloc at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/components/esp_littlefs/src/littlefs/lfs.c:689
    0x4018f6ba: lfs_alloc at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/components/esp_littlefs/src/littlefs/lfs.c:689
    0x4018f7e9: lfs_file_relocate at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/components/esp_littlefs/src/littlefs/lfs.c:3228
    0x4018f927: lfs_file_outline at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/components/esp_littlefs/src/littlefs/lfs.c:3298
    0x4018f927: lfs_file_flushedwrite at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/components/esp_littlefs/src/littlefs/lfs.c:3529
    0x4018fd31: lfs_file_write_ at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/components/esp_littlefs/src/littlefs/lfs.c:3639
    0x401917dd: lfs_file_write at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/components/esp_littlefs/src/littlefs/lfs.c:6134
    0x4018d00e: vfs_littlefs_write at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/components/esp_littlefs/src/esp_littlefs.c:1509
    0x40161672: esp_vfs_write at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/vfs/vfs.c:431
    0x40094781: __swrite at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/newlib/newlib/libc/stdio/stdio.c:94
    0x4009321e: __sflush_r at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/newlib/newlib/libc/stdio/fflush.c:224
    0x40093068: _fclose_r at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/newlib/newlib/libc/stdio/fclose.c:94
    0x400930e1: fclose at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/newlib/newlib/libc/stdio/fclose.c:125
 ```

fifieldt, 2024-11-16:
We've recalled firmware version 2.5.13 due to this issue. Please downgrade to 2.5.12 and wait for a new version :)

closed 2024-11-16 by fifieldt