meshtastic/firmware #5359: RAK 11310 (rp2040 chip) won't set lora region

legobrick2x4, 2024-11-15:
### Category

Hardware Compatibility, Serial, Other

### Hardware

Rak11310

### Firmware Version

2.5.11.8e2a3e5

### Description

Not supper tech person but I'll try to explain it. 

Flashed firmware fine, plugged into android to configure (because no Bluetooth) and the name and region appears to configure as others do but this one is grayed out for the name and drop-down doesn't work. Settings>radio config doesn't work either. It's as if they are text and not actual links/buttons. 

Tried older beta firmwares, same results. Most recent alpha (2.5.13.295278b Alpha), same. 

Contacted RAK and they sent me a file to confirm the Lora chip is working and it does talk to my RAK 4631 just fine so the hardware is fine, somehow in firmware this doesn't work. 

also tried to set it via serial monitor on my steam deck (windows OTG image) and the web client will say it sent the data to set the region but unplugging and plugging into either steam deck or Android proves it didn't save to the device. 

Unsure what could prove useful that I can give you because I don't fully understand what I'm doing so if you need something that could help, I'll get it but please understand I'm not a talented programmer or anything.

vitaminrad, 2024-11-16:
I'm no expert but perhaps using the Erase Flash option from the Meshtastic flasher website?  It's important to export your config first so that it can be re-applied.

jakern, 2024-11-17:
I just had this same issue with a RAK4630 and erasing the flash was the answer. Thanks a bunch

fifieldt, 2024-11-18:
Thanks for the report, due to issues with the (previous) 2.5.13, recommend erasing flash and refreshing. Really appreciate your support testing the alphas and sorry you ran into this issue.

closed 2024-11-18 by fifieldt