meshtastic/firmware #5361: [Bug]: BMP280 does not get initialized on RAK11300

martinwozenilek, 2024-11-15:
### Category

Other

### Hardware

Rak11310

### Firmware Version

2.5.11.8e2a3e5

### Description

I've added a BMP280 to the RAK11300 on the RAK19003 Mini Base Board. BMP280 is recognized but does initialize. 

### Relevant log output

```Shell
INFO  | ??:??:?? 2 Scan for i2c devices...
DEBUG | ??:??:?? 2 Scan for I2C devices on port 1
DEBUG | ??:??:?? 2 I2C device found at address 0x76
DEBUG | ??:??:?? 2 Wire.available() = 0
DEBUG | ??:??:?? 2 Wire.available() = 0
INFO  | ??:??:?? 2 BMP-280 sensor found at address 0x76
INFO  | ??:??:?? 2 1 I2C devices found
...
INFO  | ??:??:?? 13 [EnvironmentTelemetry] Environment Telemetry: init
INFO  | ??:??:?? 13 [EnvironmentTelemetry] Init sensor: BMP280
WARN  | ??:??:?? 13 [EnvironmentTelemetry] Could not connect to detected BMP280 sensor. Remove from nodeTelemetrySensorsMap.
```

wkomm1, 2024-11-29:
I am seeing the same problem as you. I have a RAK1906 (Environment Sensor BOSCH BME680) with a RAK11310 (Raspberry Pi RP2040 Core Module for LoRaWAN with LoRa SX1262) on a RAK19007 (WisBlock Base Board 2nd Gen). The module is in slot B and the environment module was enabled with the Python CLI (meshtastic --set telemetry.environment_measurement_enabled true). The version is firmware-rak11310-2.5.11.8e2a3e5.

To rule out a hardware problem I loaded a Raspberry Pi Pico CircuitPython image to the RAK11310 and put together a basic program to test the RAK1906. As can be seen from the screenshot the hardware works OK. I also tested a generic SSD1306 OLED display using tutorial code from Adafruit learn and that also worked fine.

[image]

wkomm1, 2024-11-29:
I started poking into this a little more and I think the entry point to the problem is here: https://github.com/meshtastic/firmware/blob/8e2a3e5728bc997449c7507be76249b7a90166a6/src/detect/ScanI2CTwoWire.cpp#L264

The first registerValue that is looking at address 0xD0 (chip id) is getting returned is 0x00 so its dropping into the default case.

I hooked up a logic analyzer to BOTH I2C ports on the RAK IO expanded wisblock and I see some strange behavior... I see the boot scan but when my I2C device ACK the RP2040 probes the device again on the other I2C port!

[image]
[image]
[image]

fifieldt, 2024-12-23:
Confirming that if "0xD0 (chip id) is getting returned is 0x00" a BME280 will be detected as a BMP280 and not work. Do you know why the chip ID is not working correctly? Does it work correctly on your CircuitPython image?

https://www.bosch-sensortec.com/media/boschsensortec/downloads/datasheets/bst-bme280-ds002.pdf

[image]

Dylan111111, 2025-01-15:
I am encountering the same issue with the RAKwireless RAK19007 WisBlock Base Board (2nd Gen), RAK11310 (RP2040), and RAK1906 (BME680) combination. I even reverted to firmware version 2.3.9.f06c56a, but the problem persists.

fifieldt, 2025-03-16:
Appears to be fixed by 2.6

closed 2025-03-16 by fifieldt

Dylan111111, 2025-03-17:
Can confirm that this is fixed with 2.6. 
RAK11310 (RP2040) and RAK1906 (BME680) combination is working! I had the almost same issue with the RAKwireless RAK1902 (LPS22HB) and RAK11310 (RP2040) and that combination is also working again. :)