meshtastic/firmware #5362: [Feature Request]: Morse code addition to units ONLY with LED or Speaker

corsairius, 2024-11-15:
### Tell us your idea.

For sending or receiving messages, it would be interesting and maybe helpful at large, to send and receive Morse code using the T1000-E or something similar.  Probably for of a HAM skill, but would be interesting in a SHTF scenario so that smaller units could provide comms as well.  Those units, however, would need a more durable switch than the 1000E provides, as it is not a very responsive button. Maybe even buffer it for replay, and use Morse as a kind of menu control into the message sending and receiving.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

garthvh, 2024-11-15:
Duplicate of https://github.com/meshtastic/firmware/issues/1849

closed 2024-11-15 by garthvh

mashu, 2025-04-02:
How about connecting externally abpaddle?
Like one of these Putikeeg cw-morse-pocket-paddle?

We do connect that thru Arduino or USB vband dongle to practice over internet. Over Meshtastic network would be indeed even cooler.

garthvh, 2025-04-03:
The serial module and an arduino would be the perfect way to do this.

mashu, 2025-04-03:
There are USB C dongles on AliExpress that interpret left and right connection as either dot or dash and act as HID keyboard. But Aruduino can do that as well I do that with Leonardo. However many.lora boards have gpio broken out. The difficult part is transmitting Morse dot and dash over Meshtastic network and perhaps not led but buzzer sound from a node. Most high speed hams can't read flashes but they hear letters as sounds.