meshtastic/firmware #5373: [Feature Request]: Separate Mesh Limits and MQTT Limits

Technologyman00, 2024-11-17:
### Platform

Cross-Platform

### Description

With the addition of #5086 I realized that it enforces a minimum update on not only the mesh telemetry but also MQTT. This is probably fine for a vast majority of users, but for a weather station that reports to a private MQTT server it seveverly limits the resolution as many weather sharing services prefer faster than 30 minute intervals for reporting. 

I would like that the time limit is not enforced on private MQTT servers.

garthvh, 2024-11-17:
Most private servers use the default channel without any traffic management so they are even more fragile that the public server.

closed 2024-11-17 by garthvh

Charly-2, 2024-11-23:
> ...If you want faster weather station updates use a private channel and not the default key.

Hi! 
How do I use a private channel for telemetry? In the doc I found that telemetry is always sent over the default channel. Is it enough to give it another name and use a private PSK? Can I then set the telemetry-intervals to values lower 1800?
BTW: The [docs](https://meshtastic.org/docs/configuration/module/telemetry/) don't mention the new minimum limit of 1800.